roboticslab-uc3m/kinematics-dynamics #13: Teo-main compilation error

rsantos88, 2015-11-16:
Compilation error with teo-main code:

[  8%] Built target KdlSolver
[ 41%] Built target FakeControlboard
[ 44%] Built target TeoYarp
[ 47%] Built target launchTeoYarp
[ 55%] Built target dialogueManager1
[ 70%] Built target teoCartesianServer
[ 79%] Built target teoGravityCompensator
[ 82%] Building CXX object programs/teoSim/CMakeFiles/teoSim.dir/TeoSim.cpp.o
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/local/include/openrave-0.9/openrave/openrave.h:72,
                 from /usr/local/include/openrave-0.9/openrave-core.h:22,
                 from /home/teo/git/teo-main/programs/teoSim/TeoSim.hpp:13,
                 from /home/teo/git/teo-main/programs/teoSim/TeoSim.cpp:3:
/usr/include/boost/smart_ptr/shared_ptr.hpp: In instantiation of ‘boost::shared_ptr<X> boost::dynamic_pointer_cast(const boost::shared_ptr<U>&) [with T = OpenRAVE::SensorBase::CameraGeomData; U = const OpenRAVE::SensorBase::SensorGeometry]’:
/home/teo/git/teo-main/programs/teoSim/TeoSim.cpp:137:190:   required from here
/usr/include/boost/smart_ptr/shared_ptr.hpp:810:5: error: cannot dynamic_cast ‘0u’ (of type ‘const class OpenRAVE::SensorBase::SensorGeometry*’) to type ‘class OpenRAVE::SensorBase::CameraGeomData*’ (conversion casts away constness)
     (void) dynamic_cast< T* >( static_cast< U* >( 0 ) );
     ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:814:41: error: cannot dynamic_cast ‘(& r)->boost::shared_ptr<T>::get<const OpenRAVE::SensorBase::SensorGeometry>()’ (of type ‘boost::shared_ptr<const OpenRAVE::SensorBase::SensorGeometry>::element_type* {aka const class OpenRAVE::SensorBase::SensorGeometry*}’) to type ‘E* {aka class OpenRAVE::SensorBase::CameraGeomData*}’ (conversion casts away constness)
     E * p = dynamic_cast< E* >( r.get() );
                                         ^
programs/teoSim/CMakeFiles/teoSim.dir/build.make:100: recipe for target 'programs/teoSim/CMakeFiles/teoSim.dir/TeoSim.cpp.o' failed
make[2]: *** [programs/teoSim/CMakeFiles/teoSim.dir/TeoSim.cpp.o] Error 1
CMakeFiles/Makefile2:616: recipe for target 'programs/teoSim/CMakeFiles/teoSim.dir/all' failed
make[1]: *** [programs/teoSim/CMakeFiles/teoSim.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

rsantos88, 2015-11-26:
The compilation problem appears when "TeoSim.cpp" (/home/teo/git/teo-main/programs/teoSim/TeoSim.cpp) calls "dynamic_pointer_cast" function located at "/usr/include/boost/smart_ptr/shared_ptr.hpp".
The critical line that calls this function is the line number 137 of TeoSlim.cpp. This line creates the object "pcamerageomdata" that is used in the lines: 139 and 140.
If you comment these lines, the program compiles perfectly:

// Show the camera image in a separate window
// pcamerasensorbase->Configure(SensorBase::CC_RenderDataOn);
// Get some camera parameter info
//------> boost::shared_ptr<SensorBase::CameraGeomData> pcamerageomdata = boost::dynamic_pointer_cast<SensorBase::CameraGeomData>(psensorbase->GetSensorGeometry(SensorBase::ST_Camera));
// printf("Camera width: %d, height: %d.\n",pcamerageomdata->width,pcamerageomdata->height);
//------>vectorOfCameraWidth.push_back(pcamerageomdata->width);
//------>vectorOfCameraHeight.push_back(pcamerageomdata->height);

jgvictores, 2015-12-14:
Fixed at 3960f83d5a32445f4025bce69caf347d20736602 (hope it to be b/w compatible).

closed 2015-12-14 by jgvictores