roboticslab-uc3m/kinematics-dynamics #14: [idea] Refractor CMake files

jgvictores, 2015-11-28:
A great number of new CMake features would make our CMake files cleaner and better.

PeterBowman, 2016-11-25:
I've noticed that most CMake files still set a minimum required version of 2.6.x, which should be raised to benefit from newer features. This could be discussed at a global (roboticslab) level since teo-main is not the only project affected.

jgvictores, 2016-11-25:
Yes, I'd propose:
- Match the CMake version supported by YARP
- Document this on whatever comes out of https://github.com/roboticslab-uc3m/QA/issues/4
- Update cmake files of https://github.com/roboticslab-uc3m/project-generator

PeterBowman, 2017-03-07:
Related: roboticslab-uc3m/QA#7.

PeterBowman, 2017-07-18:
Done at 6f90a96. Minimum required CMake version has been bumped to 2.8.11 to allow making use of the handy [`target_include_directories()`](https://cmake.org/cmake/help/v3.0/command/target_include_directories.html) command.

closed 2017-07-18 by PeterBowman

PeterBowman, 2017-11-29:
> Minimum required CMake version has been bumped to 2.8.11...

Bumped to 3.1.3, see ca84df0.