roboticslab-uc3m/kinematics-dynamics #15: [tests] Fix KdlSolverInvDyn2 and KdlSolverInvDyn3

jgvictores, 2015-11-29:
KdlSolverInvDyn2 and KdlSolverInvDyn3 dynamics tests fail, possibly since H0 and HN were added. Please review and fix.

rsantos88, 2016-01-15:
The value returned by the function: "int CartToJnt(const JntArray &q, const JntArray &q_dot, const JntArray &q_dotdot, const Wrenches& f_ext,JntArray &torques);" in the line 145 of ICartesianSolverImpl.cpp is '-1'.
This function has been defined in KDL Library, on "chainidsolver_recursive_newton_euler.cpp" file and this result means:
//! Failed to converge
        E_NO_CONVERGE      = -1,

jgvictores, 2016-02-11:
Thanks. With commit 81eb017afc89ac7486353c43bd60bcc0a1fb78ce we now have explicit prints of this error and similar ones.
Bug not fixed yet.

jgvictores, 2016-02-20:
It turns out that chainidsolver_recursive_newton_euler.cpp returns a -1 in a very different scenario:
https://github.com/orocos/orocos_kinematics_dynamics/blob/707ce2c58a1af628696b3c3828de98bf30f63366/orocos_kdl/src/chainidsolver_recursive_newton_euler.cpp#L37-L38

jgvictores, 2016-02-20:
Put issue upstream: https://github.com/orocos/orocos_kinematics_dynamics/issues/78

jgvictores, 2016-06-17:
So... from https://github.com/orocos/orocos_kinematics_dynamics/blob/707ce2c58a1af628696b3c3828de98bf30f63366/orocos_kdl/src/chainidsolver_recursive_newton_euler.cpp#L37-L38
We see that the error is probably due to our H0 and HN. Actually, commenting out https://github.com/roboticslab-uc3m/teo-main/blob/e898d0f6e146126ceceb120b8c563d0fc3aa0cf7/libraries/TeoYarp/KdlSolver/DeviceDriverImpl.cpp#L42 and https://github.com/roboticslab-uc3m/teo-main/blob/e898d0f6e146126ceceb120b8c563d0fc3aa0cf7/libraries/TeoYarp/KdlSolver/DeviceDriverImpl.cpp#L133 allows Dyn tests to pass.

jgvictores, 2017-04-09:
In fact is fixed with simple patch: https://github.com/jgvictores/orocos_kinematics_dynamics/commit/d20c9d11e6a0c04c63a6b684a1351eed055b588b (lines are a bit different in current KDL, but patch could be updated).

jgvictores, 2017-04-10:
Fixed at 5043d8c74804f2bf95fa5320efca2cf553bd2a0e by using a combination of KDL `chain.getNrOfJoints()` and `chain.getNrOfSegments()` (note that `segments` > `joints` if there are `Joint::None`).

commit 5043d8c7 (2017-04-10): Merge branch 'fix-92-tests' into develop, fix #15
closed 2017-04-10 by jgvictores