roboticslab-uc3m/kinematics-dynamics #28: Unify robot models

jgvictores, 2016-04-14:
Issue inherited from wiki, where work has been done (resulting in [[0]](http://robots.uc3m.es/index.php/TEO_Diagrams)). The following is a list of models to potentially be merged or linked somehow.
- Mass/Inertia XLS (official based on mechanical schematics): [[1]](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.xls) and PDF [[2]](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.pdf).
- Kinematics(DH)+Mass/Inertia used by [[3]](https://github.com/roboticslab-uc3m/teo-main/tree/master/libraries/TeoYarp KdlSolver): [[4]](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics).
- OpenRAVE 3D and Kinematics used by [[5]](https://github.com/roboticslab-uc3m/teo-main/tree/master/programs/teoSim teoSim): [[6]](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/openrave/teo), teoLowres version: [[7]](https://github.com/roboticslab-uc3m/teo-main/tree/develop/share/openrave/teoLowres).
- Gazebo 3D and Kinematics+Mass/Inertia used by [[8]](https://github.com/roboticslab-uc3m/teo_batc): [[9]](https://github.com/roboticslab-uc3m/teo_batc/tree/master/teo_description)

jgvictores, 2016-04-18:
@RaulFdzbis Can you check Gazebo 3D and Kinematics [teo_description](https://github.com/roboticslab-uc3m/teo_batc/tree/master/teo_description) follows [current TEO Diagrams](http://robots.uc3m.es/index.php/TEO_Diagrams)?

RaulFdzbis, 2016-04-19:
**Note:** The teo_description package have been moved to [teo-ros-simulation](https://github.com/roboticslab-uc3m/teo-ros-simulation), also there is a model of TEO to work with a **stand-alone** setup of Gazebo in [teo-gazebo](https://github.com/roboticslab-uc3m/teo-gazebo).

Modifications of models in accordance with this issue were done in [a6199bc](https://github.com/roboticslab-uc3m/teo-ros-simulation/commit/a6199bc965c44919686355d27d876d9e002fa77c) for the **teo-ros-simulation** package and [7718f22](https://github.com/roboticslab-uc3m/teo-gazebo/commit/7718f22c90b09dc30c5b4b4b55e1aa86e3c85566) for the **teo-gazebo** repository.

In addition the joint limitations of both models have been also updated in accordance to the last version of [teo_main](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics).

jgvictores, 2016-04-20:
@RaulFdzbis Thanks! Marked Gazebo part as done!

jgvictores, 2016-04-21:
teoSim openrave was last missing, done at https://github.com/jgvictores/teo-main/commit/e518e2c43aaeabb2350ad07e04c442169c2d9962.

closed 2016-04-21 by jgvictores