roboticslab-uc3m/kinematics-dynamics #29: Unify robot model dynamics

jgvictores, 2016-04-20:
Issue inherited from wiki, where work has been done (resulting in [current TEO Diagrams](http://robots.uc3m.es/index.php/TEO_Diagrams)). The following is a list of models to potentially be merged or linked somehow.

- [ ] Kinematics(DH)+Mass/Inertia used by [KdlSolver](https://github.com/roboticslab-uc3m/teo-main/tree/master/libraries/TeoYarp/KdlSolver): [[3]](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics).
- [ ] OpenRAVE 3D and Kinematics used by [teoSim](https://github.com/roboticslab-uc3m/teo-main/tree/master/programs/teoSim): [[4]](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/openrave/teo), teoLowres version: [[5]](https://github.com/roboticslab-uc3m/teo-main/tree/develop/share/openrave/teoLowres).
- [ ] Gazebo 3D and Kinematics+Mass/Inertia used by [teo_batc](https://github.com/roboticslab-uc3m/teo_batc): [[6]](https://github.com/roboticslab-uc3m/teo_batc/tree/master/teo_description)

For future reference, links to Mass/Inertia (official based on mechanical schematics): [.XLS](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.xls) and [.PDF](http://robots.uc3m.es/svn/TEOrepo/modelling/dynamics/Inercias_090518.pdf).

lpinel, 2016-05-06:
Added to [http://robots.uc3m.es/index.php/TEO_Diagrams](url) temporary Force-Torque Sensors Diagram wich includes sensors axes, location and connections to DAQ card.

PeterBowman, 2017-08-16:
This issue was moved to roboticslab-uc3m/teo-main#16

closed 2022-05-22 by PeterBowman