roboticslab-uc3m/kinematics-dynamics #31: Unify robot model joint limits

jgvictores, 2016-04-21:
Should be in wiki, where work has been done (resulting in [current TEO Diagrams](http://robots.uc3m.es/index.php/TEO_Diagrams)). The following is a list of models to potentially be merged or linked somehow.

- [ ] Real joint limits are in [.ini of launch files in teo-body](https://github.com/roboticslab-uc3m/teo-body/tree/master/share), should be the same across files.
- [ ] Kinematics used by [KdlSolver](https://github.com/roboticslab-uc3m/teo-main/tree/master/libraries/TeoYarp/KdlSolver): ([.ini files that contain DH parameters](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics)). Perhaps could be removed from here?
- [ ] Kinematics used by [teo_batc](https://github.com/roboticslab-uc3m/teo_batc): ([Gazebo model](https://github.com/roboticslab-uc3m/teo-gazebo)), ([ROS Gazebo model](https://github.com/roboticslab-uc3m/teo-ros-simulation)).
- [ ] Kinematics used by [teoSim](https://github.com/roboticslab-uc3m/teo-main/tree/master/programs/teoSim): ([Openrave model](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/openrave/teo)).

RaulFdzbis, 2016-05-03:
[Gazebo model](https://github.com/roboticslab-uc3m/teo-gazebo) and [Ros Gazebo model](https://github.com/roboticslab-uc3m/teo-ros-simulation), were already updated in [7718f22](https://github.com/roboticslab-uc3m/teo-gazebo/commit/7718f22c90b09daludo.c30c5b4b4b55e1aa86e3c85566) and [a6199bc](https://github.com/roboticslab-uc3m/teo-ros-simulation/commit/a6199bc965c44919686355d27d876d9e002fa77c) with the joint limits used by the KdlSolver in [teo_main](https://github.com/roboticslab-uc3m/teo-main/tree/master/share/kinematics).

PeterBowman, 2017-08-16:
This issue was moved to roboticslab-uc3m/teo-main#15

closed 2022-05-22 by PeterBowman