roboticslab-uc3m/kinematics-dynamics #32: Lista de objetos que hay que llevar para RoboCity

rsantos88, 2016-05-24:
Lista de objetos que hay que llevar:

1. Teo
2. Grúa o soporte para sostener a Teo
3. Fuentes de alimentación de Teo (no separar, ambas van conectadas en paralelo) con cables de alimentación.
4. Router para conectarse a Teo con cables ethernet: “head” y “main” y otro por si a caso.
5. PC “Teo-main” con teclado, ratón, monitor y cable de alimentación.
6. Monitor extra con cable VGA.
7. Teclado extra y ratón inalámbrico
8. Micrófono bluetooth
9. Botella de bici roja (se necesita llena de agua para la demo)
10. Opcional: Televisión LG grande con cable de alimentación, cable HDMI y mando.
11. Regletas (x2)

Se admiten sugerencias si falta alguna cosa más por llevar @jgvictores @smcdiaz @jmgarciah @lpinel

RaulFdzbis, 2016-05-24:
Si se lleva la grua nueva estaría bien llevar el cargador de la misma, por si acaso.

David-Estevez, 2016-05-24:
Lo mismo habría que llevar herramientas para posibles ñapas que haya que hacer en directo. :wrench:

jgvictores, 2016-05-24:
@rsantos88 he actualizado tu lista con el cargador que decía @RaulFdzbis

rsantos88, 2016-05-24:
@jmgarciah también opina que deberíamos llevar material para hacer reparaciones rápidas: bridas, llaves allen, destornilladores...

David-Estevez, 2016-05-24:
Buuuuh!! Que @jmgarciah comente él por aquí, que le echamos de menos!! :cry:

jgvictores, 2016-05-24:
@jmgarciah Me incluyo en el listado de los que te extrañan vía GitHub. ¿Podrias por favor ampliar la lista que empezó MSc @David-Estevez?

smcdiaz, 2016-05-24:
Se llevará una caja con herramientas pero a ver si somos capaces de no romper nada!! (cosa difícil....)

smcdiaz, 2016-05-24:
Tranquis. Mañana preparo yo un kit básico de supervivencia robótica con muchas bridas, cinta aislante y, por supuesto, 'superglu'

lpinel, 2016-05-24:
y la seta de emergencia ....

jgvictores, 2016-06-07:
Conferenced passed, closing issue,

commit bd9e39a4 (2016-06-17): testKdlSolver force angleRepr axisAngle, and use in KdlSolverInvKin1 and KdlSolverInvKin2, fix #32
closed 2016-06-07 by jgvictores