roboticslab-uc3m/kinematics-dynamics #35: Acquire and install perforated hardboard for tools

jgvictores, 2016-06-13:
[A board with holes in it will not protect you from leprosy, but will keep your hammers in place.](http://www.core77.com/posts/15718/The-basics-of-tool-organization-systems-Part-1-Pegboard)

smcdiaz, 2016-06-13:
We have a laboratory, not a workshop. Order in the laboratory doesn't depend on a board with holes but on people (I am the first one)

David-Estevez, 2016-06-13:
I agree with @jgvictores, each tool requires a fixed, visible place where it belongs, so it is easier to find them and put them back.

As current methods such as "throwing them at the cart" or "putting them on top of the cart" do not work, I think it might be good to consider suitable alternatives. If no one follows the current method it might be the method the one to be blamed and not the people.

smcdiaz, 2016-06-13:
It is a good idea if the people put the tools back to the board but, in my experience, it doesn't happend. There was a method before: 'leave the tools inside the toolbox' but didn't work.

Do you think that the change to: 'leave the tool on the board' will work?

jgvictores, 2016-06-13:
> Do you think that the change to: 'leave the tool on the board' will work?
Yes.

smcdiaz, 2016-06-13:
I don't mind. The answer is no to the proposed system. Look for another.

jgvictores, 2016-06-14:
*Walls and ceilings are actually recommended places (...), since in most cases these spaces are  underutilized [1].*

[1] A. J. Huete, J. G. Victores, S. Martinez, A. Gimenez and C. Balaguer, "Personal Autonomy Rehabilitation in Home Environments by a Portable Assistive Robot," in IEEE Transactions on Systems, Man, and Cybernetics, Part C (Applications and Reviews), vol. 42, no. 4, pp. 561-570, July 2012.
doi: [10.1109/TSMCC.2011.2159201](ieeexplore.ieee.org/xpls/abs_all.jsp?arnumber=5959995)

smcdiaz, 2016-06-14:
Ok. Use the ceilling

jmgarciah, 2016-06-14:
Why do you need a harboard , if you never uses the tools for teo ?? Pedro already reached agreement with Antonio and David for the use of tools. We were all agreed and there has been no problem.

smcdiaz, 2016-06-14:
Hardboard is not a good solution for the laboratory. Other suggestions are welcomed and can be discussed in the next TEO meeting

jgvictores, 2016-06-14:
@jmgarciah There also seems to be an agreement on the _ceilling_.
1. We constantly and consistently use tools for TEO and activities related to TEO.
2. We constantly and consistently need the table for experiments (see #36).

smcdiaz, 2016-06-14:
Insisto. Esto se discutirá en la próxima reunión de TEO.

jgvictores, 2016-06-14:
Por tanto, open issue.

jgvictores, 2016-12-07:
Moved to https://github.com/jgvictores/laboratorios/issues/8

closed 2016-12-07 by jgvictores