roboticslab-uc3m/kinematics-dynamics #40: Error in CMake and Make to compile Teo-main software. (warning yarpdeprecated).

serna92, 2016-06-17:
Hello,
While I'm following the tutorial to install the Teo-main software I get an error when I compile with Make:

[make_output.txt](https://github.com/roboticslab-uc3m/teo-main/files/320895/make_output.txt)

Taking a look to the cmake command output I see that it throws a warning related with Yarp:

CMake Warning at /usr/local/share/yarp/cmake/YarpDeprecatedWarning.cmake:7 (message):
  YARP_ADD_PLUGIN_LIBRARY_EXECUTABLE is deprecated.  Use
  YARP_ADD_PLUGIN_YARPDEV_EXECUTABLE instead.
Call Stack (most recent call first):
  /usr/local/share/yarp/cmake/YarpPlugin.cmake:413 (yarp_deprecated_warning)
  libraries/TeoYarp/CMakeLists.txt:25 (yarp_add_plugin_library_executable)

[cmake_output.txt](https://github.com/roboticslab-uc3m/teo-main/files/320896/cmake_output.txt)

I reinstalled Yarp again following the documentation but I got the same error.

¿How can I solve it?

David-Estevez, 2016-06-17:
After taking a look at the output of the make command, it seems to me that it is a problem with KDL  and not with YARP. I don't know what is exactly the problem, but I hope that at least knowing that the problem is KDL can direct your troubleshooting to the actual cause. :smile: 

The warning you obtained from YARP is just a warning, not an error, and indeed you cannot do anything to make it disappear, as the warning comes from the Teo-main software, not your YARP version.

The warning

serna92, 2016-06-17:
I compiled again KDL following the documentation but I got the same error while compiling Teo-main.

jgvictores, 2016-06-17:
Your ```error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘strError’``` shouldn't be happening with the [current KDL master](https://github.com/orocos/orocos_kinematics_dynamics/blob/a8bc786bb269c0275496cc45d8479715a5fe93d6/orocos_kdl/src/solveri.hpp).

Did you follow [the KDL install guide from teo-main](http://robots.uc3m.es/dox-teo-main/teo_install_kdl_on_ubuntu_14_10.html) or a different one?

serna92, 2016-06-17:
Yes, I followed the one from teo-main and just in case I repeated the steps again removing the cloned repository before.

jgvictores, 2016-06-27:
Please do the following, and copy+paste the output here:
```bash
sudo updatedb
locate solveri.hpp
locate chain.hpp
cd ~/repos/orocos_kinematics_dynamics
git log
```

serna92, 2016-06-27:
miguel@miguel-HP-15-Notebook-PC:~$ sudo updatedb
[sudo] password for miguel: 
miguel@miguel-HP-15-Notebook-PC:~$ locate solveri.hpp
/home/miguel/repos/orocos_kinematics_dynamics/orocos_kdl/src/solveri.hpp
/opt/ros/jade/include/kdl/solveri.hpp
/usr/local/include/kdl/solveri.hpp
miguel@miguel-HP-15-Notebook-PC:~$ locate chain.hpp
/home/miguel/repos/orocos_kinematics_dynamics/orocos_kdl/src/chain.hpp
/opt/ros/jade/include/kdl/chain.hpp
/usr/include/boost/container/detail/multiallocation_chain.hpp
/usr/include/boost/iostreams/chain.hpp
/usr/local/include/kdl/chain.hpp
miguel@miguel-HP-15-Notebook-PC:~$ cd ~/repos/orocos_kinematics_dynamics
miguel@miguel-HP-15-Notebook-PC:~/repos/orocos_kinematics_dynamics$ git log
commit a8bc786bb269c0275496cc45d8479715a5fe93d6
Merge: dfceb87 beee9e5
Author: Ruben Smits <ruben.smits@intermodalics.eu>
Date:   Tue Apr 5 12:09:28 2016 +0200

    Merge pull request #70 from mikepurvis/relocatable
    
    Make orocos_kdl installspace relocatable.

commit dfceb87d5c48940a07d98a5b75fcf1db8b9e146f
Merge: 707ce2c f7b2b8c
Author: Ruben Smits <ruben.smits@intermodalics.eu>
Date:   Tue Apr 5 11:55:31 2016 +0200

    Merge pull request #66 from traversaro/patch-4
    
    Fix config when Eigen_INCLUDE_DIR includes a space

commit 707ce2c58a1af628696b3c3828de98bf30f63366
Merge: 912f44c c9db5e4
Author: Ruben Smits <ruben.smits@intermodalics.eu>
Date:   Mon Feb 15 15:25:56 2016 +0100

:

jgvictores, 2016-06-27:
Please do the following, and copy+paste the output here:
```diff  /opt/ros/jade/include/kdl/solveri.hpp  /usr/local/include/kdl/solveri.hpp```

serna92, 2016-06-27:
miguel@miguel-HP-15-Notebook-PC:~$ diff /opt/ros/jade/include/kdl/solveri.hpp /usr/local/include/kdl/solveri.hpp
116a117,118
> 		else if (E_UNDEFINED == error) return "Undefined value";
> 		else if (E_DEGRADED == error) return "Converged but degraded solution";

jgvictores, 2016-06-27:
Please do the following, and copy+paste the output here:
```
locate chainidsolver_recursive_newton_euler.hpp
locate chainidsolver.hpp
```

serna92, 2016-06-27:
miguel@miguel-HP-15-Notebook-PC:~$ locate chainidsolver_recursive_newton_euler.hpp
/home/miguel/repos/orocos_kinematics_dynamics/orocos_kdl/src/chainidsolver_recursive_newton_euler.hpp
/opt/ros/jade/include/kdl/chainidsolver_recursive_newton_euler.hpp
/usr/local/include/kdl/chainidsolver_recursive_newton_euler.hpp
miguel@miguel-HP-15-Notebook-PC:~$ locate chainidsolver.hpp
/home/miguel/repos/orocos_kinematics_dynamics/orocos_kdl/src/chainidsolver.hpp
/opt/ros/jade/include/kdl/chainidsolver.hpp
/usr/local/include/kdl/chainidsolver.hpp

jgvictores, 2016-06-27:
Please do the following, and copy+paste the output here:
```
diff  /opt/ros/jade/include/kdl/chainidsolver_recursive_newton_euler.hpp  /usr/local/include/kdl/chainidsolver_recursive_newton_euler.hpp
diff  /opt/ros/jade/include/kdl/chainidsolver.hpp  /usr/local/include/kdl/chainidsolver.hpp
```

serna92, 2016-06-27:
miguel@miguel-HP-15-Notebook-PC:~$ diff  /opt/ros/jade/include/kdl/chainidsolver_recursive_newton_euler.hpp  /usr/local/include/kdl/chainidsolver_recursive_newton_euler.hpp
miguel@miguel-HP-15-Notebook-PC:~$ diff  /opt/ros/jade/include/kdl/chainidsolver.hpp  /usr/local/include/kdl/chainidsolver.hpp
27a28
> #include "solveri.hpp"
39c40
< 	class ChainIdSolver
---
> 	class ChainIdSolver : public KDL::SolverI

jgvictores, 2016-06-27:
So, apparently, the ROS Jade version of KDL is from before ChainIdSolver became a SolverI (which implements `strError()`), and this `/opt/ros/jade/include/kdl/chainidsolver.hpp` is shadowing `/usr/local/include/kdl/chainidsolver.hpp`.

Looking for a quick solution (to avoid doing anything ugly like modifying an installed header, or anything complex like understanding what is happening in CMake)... Are you using ROS for anything? Would you have any problem with uninstalling it?

PS: The ugly way would involve modifying `/opt/ros/jade/include/kdl/chainidsolver.hpp` line `class ChainIdSolver` to `class ChainIdSolver : public KDL::SolverI`. Maybe you could do that as a test, and then think if you are satisfied with that hack.

serna92, 2016-06-27:
I'm not using ROS for something important now. How can I uninstall it?

jgvictores, 2016-06-27:
According to [this](http://answers.ros.org/question/57213/how-i-completely-remove-all-ros-from-my-system/), you should be able to simply `sudo apt-get remove ros-*`.

I'd do a `sudo updatedb`, `locate chainidsolver.hpp` to look and make sure the ROS version has been removed. Once you are sure, proceede to re-compiling teo-main.

serna92, 2016-06-27:
It worked :), thank you very much and sorry about the disturbances.

jgvictores, 2016-06-27:
Great. :)
Just to know before closing the issue, what did you do? Uninstall ROS Jade, or modify the installed header?

serna92, 2016-06-27:
Uninstall ROS Jade

jgvictores, 2016-06-27:
Ok, thanks! Closing issue.

closed 2016-06-27 by jgvictores

jgvictores, 2017-02-15:
A quick note in case ROS is shadowing KDL installed from source to `/usr/local/share`:
```
cmake .. -Dorocos_kdl_DIR=/usr/local/share/orocos_kdl
```

jgvictores, 2017-04-14:
A better approach is to clean environment variables (found with `set | grep $ROS_DISTRO`), perhaps not all required:
```
export CMAKE_PREFIX_PATH=
export CPATH=
export LD_LIBRARY_PATH=
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
export PKG_CONFIG_PATH=
```

jgvictores, 2017-04-22:
Yet another valid method for compilation commented by @PeterBowman is to temporarily comment out the `source` line that ROS installation guides indicate, such as the following `~/.bashrc` line for indigo:
```
source /opt/ros/indigo/setup.bash
```
Before compiling from a new terminal.

jgvictores, 2017-11-27:
Similar to https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/40#issuecomment-294209890 is `unset CPATH`.