roboticslab-uc3m/kinematics-dynamics #41: [KdlSolver] Refractor vectorToFrame() and frameToVector()

jgvictores, 2016-06-27:
Though computational cost is not so much of an issue nowadays, the current string comparison implementation of KdlSolver vectorToFrame() and frameToVector() could be rendered more efficient in many ways. Since we're in C++, I'd be thinking in polymorphism, but even a simple pointer to a function would be better than the current implementation.

PeterBowman, 2017-08-16:
Possibly superseded by #113.

PeterBowman, 2017-08-23:
Closing as invalid due to #114. Logic for selection of angle representations has been transferred to [KinematicRepresentation.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/0ff88b2110b1bba691457a2f67136410e9c34c3e/libraries/KinematicRepresentationLib/KinematicRepresentation.cpp) and it's no longer based on string comparisons; rather, we've moved to using enums.

closed 2022-05-22 by PeterBowman