roboticslab-uc3m/kinematics-dynamics #42: [Yarp module manager] Problem with computers status detection

rsantos88, 2016-06-28:
Se ha detectado que en determinadas ocasiones el Yarp module manager no detecta correctamente los ordenadores de Teo, marcando estos como Status: unavailable. En cambio, la conexión con estos es correcta si accedemos por ssh. La única solución hasta el momento es reiniciar el ordenador en cuestión y refrescar el Yarp module manager hasta que lo detecta correctamente.

jgvictores, 2016-06-28:
**Respuesta rápida:** En [TEO Network troubleshooting](http://robots.uc3m.es/index.php/TEO_Network_troubleshooting) tienes una solución más rápida que reiniciar el ordenador en cuestión.

**Explicación más densa:** El Yarp module manager ([yarpmanager](http://www.yarp.it/yarpmanager.html)) no detecta los PCs como tal, sino instancias de servidores yarprun. Las instancias de yarprun en cada PC se manejan con daemontools, como está documentado en [TEO Network information](http://robots.uc3m.es/index.php/TEO_Network_information) (la documentación de yarprun está [aquí](http://www.yarp.it/yarprun.html), pero no creo que sea muy necesario para esto). Lo que hay que hacer es mejorar el servicio de daemontools de yarprun para que realmente detecte el yarp server y de ahí lance el yarprun.

jgvictores, 2016-06-28:
He creado #43 referente al problema descrito en la explicación más densa. Ok cerrar esta issue?

rsantos88, 2016-06-28:
Ok, la cierro. 
Gracias.

closed 2016-06-28 by rsantos88