roboticslab-uc3m/kinematics-dynamics #46: [BasicCartesianControl] needs improvements in movl

jgvictores, 2016-07-06:
[BasicCartesianControl] needs improvements in movl.
- [ ] Trajectories in simulation need gain and shouldn't need any. Trajectories on the real robot need more gain and are rather faulty.
- [ ]Should be able to set velocity instead of duration.

PeterBowman, 2017-10-16:
These improvements could be handled by a new controller interface, see https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/105#issuecomment-334736223:

> I was thinking about a new interface (inheritable by BasicCartesianControl and the like) for managing controller configuration during runtime. In fact, it would be nice to have the ability to select a different gain, tweak the maximum joint velocity and trajectory time (along the lines of #46), change the trajectory generator, enable RPC/streaming mode, or even choose the reference frame for input coordinates. Do you think it's feasible?

PeterBowman, 2017-10-28:
We are going to dismiss the proposal quoted in my previous comment and implement new methods directly in the controller interface. Until that, this issue is blocked by #121.

jgvictores, 2018-01-19:
Related: #135

PeterBowman, 2018-06-28:
It's almost two years now and nothing has moved here. I guess this issue originated when the first implementation for BasicCartesianControl became available and still required testing. Conclusion:

* The first subtask may be deemed invalid (either unclear or solved). I've been working with this controller on a real robot and didn't notice any particular issues with the controller gain. Also, I can set it to zero and still work on the simulator.

* The second subtask has been moved to https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/157.

closed 2022-05-22 by PeterBowman