roboticslab-uc3m/kinematics-dynamics #50: Migrate install instructions from [dox] to [md]

jgvictores, 2016-07-22:
Migrate install instructions from [dox] to [md].

rsantos88, 2016-07-26:
I will add in this issue links that there broken in doxygen documentation: 
* [Simulation and Basic Control: Now what can I do? ] (http://robots.uc3m.es/dox-teo-main/teo_post_install.html)
    * testRaveBot (http://robots.uc3m.es/dox-teo-main/group__testRaveBot.html#testRaveBot_interfacing)
    * ravebotCompare.png
    * other programs. (http://robots.uc3m.es/dox-teo-main/programs.html)

rsantos88, 2016-07-26:
Migrate is completed in branch " fix50installDoxToMd ": 
[Simulation and Basic Control: Installation from Source Code (Ubuntu 14.10)] (https://github.com/roboticslab-uc3m/teo-main/blob/fix50installDoxToMd/doc/teo_install_on_ubuntu_14_10.md)

rsantos88, 2016-07-26:
Pull request #54 finished!

closed 2016-07-26 by rsantos88