roboticslab-uc3m/kinematics-dynamics #53: Integrate GaitTrajectory with BasicCartesianControl

jgvictores, 2016-07-23:
Integrate [GaitTrajectory](https://github.com/roboticslab-uc3m/teo-main/blob/develop/libraries/TrajectoryLib) with [BasicCartesianControl](https://github.com/roboticslab-uc3m/teo-main/tree/develop/libraries/TeoYarp/BasicCartesianControl).

jgvictores, 2016-10-04:
Not sure if it should actually be not with `BasicCartesianControl`, but with `BasicTwoLimbCartesianControl` off the current `twolimbcc` branch, see:
https://github.com/roboticslab-uc3m/teo-main/tree/b42dfe0d5c8b78825b5931b93fe1fa11c3112f3f/libraries/TeoYarp/BasicTwoLimbCartesianControl

munozyanez, 2016-10-14:
I will need directions to integrate. I know GaitTrajectory, but not BasicCartesianControl.
A staring point, like the controller basics and the functions involved should be great.

jgvictores, 2016-10-14:
Oh, I was planning on doing it myself. It was kind of a reminder to myself on who's involved, who I can consult if I have doubts. Sorry for causing ambiguities with the assignment!

munozyanez, 2016-10-14:
Oh, ok, nvm. Anyway, ask for help if needed :-)

jgvictores, 2017-12-03:
**Note to self:** This issue in fact blocking https://github.com/orgs/roboticslab-uc3m/projects/3.

jgvictores, 2018-01-14:
Blocked by #134

PeterBowman, 2018-06-28:
> Note to self: This issue in fact blocking https://github.com/orgs/roboticslab-uc3m/projects/3.

That project is dead. I'm removing the *blocking* label, please tell which tasks are actually blocked before reinstating it. Also, since overall efforts in gait development are dead, too, it makes little sense to me to claim that such task is still regarded as high-priority (since 2016!).

Anyway, regarding the "new" *BasicTwoLimbCartesianControl* device: https://github.com/roboticslab-uc3m/teo-bimanipulation/issues/3#issuecomment-395489706.

jgvictores, 2018-06-28:
Disclaimer: sorry for the abuse of `epic`.

> I'm removing the *blocking* label, please tell which tasks are actually blocked before reinstating it.

I'd say https://github.com/roboticslab-uc3m/teo-main/issues/12 (`epic`) is blocked by this issue.

> Also, since overall efforts in gait development are dead, too, it makes little sense to me to claim that such task is still regarded as high-priority (since 2016!).

1. We have seen advances from @munozyanez and @AlvaroMartinezR in the last TEO meeting.
1. In each TEO meeting it is still considered a high-priority task. I know it's kind of `epic`, but I'd like to preserve it to remind myself of it's relevance (e.g. for big-boss), even though I think it should be technically blocked by kind of `epic` #134.

PeterBowman, 2018-06-28:
For "gait development" I wanted to stress on whatever concerns this issue only, which is focused on integrating gait mechanisms into our controller(s). It is in fact referenced in https://github.com/roboticslab-uc3m/teo-main/issues/12. I am not attending TEO meetings, perhaps code-related follow-ups could be shared as comments in any of these issues?

I'm restoring the labels, sorry for my rush.

PeterBowman, 2020-03-16:
@jgvictores can we please take another look at this?

* This issue is almost four years old. We might have seen "advances" on directions of rotations and such, but actually none on the software behind gait control since https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/9984674de0817a291100b809ae3c0e5a886c32f6 (September '17). Don't beat a dead horse.
* People involved in it is no longer working on gait/UC3M (for long).
* Those actually involved in gait (like me) are not developing new solutions in this repo. Actually, kinematics-dynamics doesn't seem the best place for it. Are we ever going to integrate GaitTrajectory with CartesianControl? **I think not** (March '20).
* Last gait-control-related stuff has been nuked at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/d20f135a299fdc898dcead3aa16f84e62a88606f (hence no more BasicTwoLimbCartesianControl). This effectively reverts the original development as stated in the issue description, and renders unchecked items invalid.
* Whatever you aimed for completion at https://github.com/roboticslab-uc3m/teo-main/issues/12, it is hard to conclude on your scarce comments. Anyway, given the outcome of this issue, I really think you should revisit that one (and focus on TEO).

+1 for closing as invalid/due to inactivity.

jgvictores, 2020-03-16:
I totally agree. `/me` closing issue due to inactivity.

closed 2022-05-22 by PeterBowman