roboticslab-uc3m/kinematics-dynamics #55: [Errors]  Yarp Manager and eSpeak with new Mbrola Voices

rsantos88, 2016-09-08:
A tener en cuenta los siguientes problemas que tienen lugar en el Yarp Module Manager (en concreto, cuando se trata de arrancar el Follow-me):
* El nuevo paquete del habla, ubicado en "teoBase_App",arranca correctamente la primera vez. Si se decide parar dicho módulo, al intentar arrancarlo de nuevo, mostrará el error:
`[ERR] (teoBase_App) cannot run /sbin/runuser on /head : cannot run /sbin/runuser on /head`
La única solución para volver a arrancarlo es reiniciando el PC de la cabeza.
* Si se consigue arrancar con éxito el módulo del habla y se prueba en teoTools_App (escribiendo por teclado), funciona correctamente. Si arrancamos todos los módulos para hacer funcionar la Demo de Follow-me y establecemos las conexiones oportunas, no funcionará el habla que corresponde a la Demo (Teo no pronunciará las frases correspondientes al Follow-me).

jgvictores, 2016-09-08:
1. Si te fijas en las líneas 48 y 54 de https://github.com/roboticslab-uc3m/teo-main/commit/51927f59f164420dceac83cd8a8d647d129c53d0) para el nuevo sistema de habla consecuencia de https://github.com/roboticslab-uc3m/follow-me/issues/11, hay un pipe (símbolo `|`) entre dos procesos. Esto es, se están lanzando dos procesos como si fuesen una sola en el `yarpmanager`. Resulta lógico que `yarpmanager` no gestione esto bien, probablemente al cerrar solamente cierre uno o el otro, quedando uno de los procesos abiertos causando conflicto al intentar abrir otra vez. Sospechaba que algo así podría ocurrir con esta solución propuesta en esa issue, pero era una solución rápida y quería ver si daba problemas en la práctica (supuestamente teoBase no lo estamos abriendo/cerrando).
La única solución que se me ocurre es hacer un wrapper en condiciones de lo que estemos utilizando: el equivalente o ampliación de `yarpdev --device speech` (o el obsoleto `iSpeak`) para espeak/mbrola. Al tratarse de un solo proceso, y bajo el supuesto de que esté bien implementado (responda bien a señales de sistema operativo), `yarpmanager` lo deberá gestionar bien.
2. Únicamente se me ocurre que se esté utilizando una versión de follow-me que no envíe los comandos en el que se lo espera `yarpdev --device speech` (por ejemplo, que siga enviando en formato `iSpeak`). En cuanto se solucione (1), nos aseguraremos que la parte de follow-me (que actúa de cliente o publicador, según) respete el protocolo definitivo.

jgvictores, 2016-10-13:
Created https://github.com/roboticslab-uc3m/teo-head/issues/38 to work on issue upstream.

jgvictores, 2016-10-13:
Original source: https://github.com/roboticslab-uc3m/follow-me/issues/11

jgvictores, 2016-11-11:
Closed because https://github.com/roboticslab-uc3m/teo-head/issues/38 is closed.

closed 2016-11-11 by jgvictores