roboticslab-uc3m/kinematics-dynamics #62: [teoBase app] and [teoSim app] use absolute paths for kinematics

jgvictores, 2016-10-12:
[teoBase app] and [teoSim app] use absolute paths for kinematics. Should use relative, e.g. with ResourceFinder.

PeterBowman, 2017-10-16:
The initialization part was handled at 33c4b42 on the `KdlSolver` plugin. Now, we may consider updating relevant XMLs at teo-configuration-files:

* [teoBase.xml](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/e52590665a54cfbf33863f6e6264bdcc1df154a1/share/teoBase/scripts/teoBase.xml)
* [teoSimBase.xml](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/e52590665a54cfbf33863f6e6264bdcc1df154a1/share/teoSimBase/scripts/teoSimBase.xml)

@jgvictores could you take a look at this (move/close)?

jgvictores, 2017-10-22:
This issue was moved to roboticslab-uc3m/teo-configuration-files#3

closed 2017-10-22 by jgvictores