roboticslab-uc3m/kinematics-dynamics #63: Need functional collision avoidance module

jgvictores, 2016-10-14:
Started work on C++ teoCollisionAvoidance, but may be more practical to do in Python.

RaulFdzbis, 2016-10-19:
Some advances have been made in this area. Some problems and solutions have been found working in this issue. The followings are:
**Good things**

- **The initial [idea]( http://openrave-users-list.185357.n3.nabble.com/Feature-Announcement-Automated-Convex-Decomposition-for-Collision-Meshes-td186409.html) behind ConvexDecompositions was to be used for collision checks, just as we wanna do**. So thats good, i guess.

- If we want to set the cdmodel parameters, the generate(**kwargs) function should be used instead the autogenerate function used in the [example](http://openrave.org/docs/0.8.2/openravepy/examples.checkconvexdecomposition/).

- The checkconvexdecomposition [example](http://openrave.org/docs/0.8.2/openravepy/examples.checkconvexdecomposition/) works fine with the TEO model. We had some problems first, but after introducing some padding value the example worked properly. Here is a [picture](http://i.imgur.com/G6M6nEK.png) of the results. This is a HUGE advance since this means that the **TEO cdmodel is properly generated.**

**Not so good things**
-  **Some of the [examples](http://openrave.org/docs/0.8.2/examples/) in OpenRave do not work properly** using ubuntu 14.04. So far this problems are only presented in the examples using robots different than the barrettwam robot used [here](http://openrave.org/docs/0.8.2/openravepy/examples.checkconvexdecomposition/). Some similar [issue](http://openrave-users-list.185357.n3.nabble.com/Errors-when-i-execute-python-examples-td4027742.html) have already been reported, however no solution have been found yet. Since this is not critital for the resolution of this issue, **this is not a priority.**

-  When the convex decomposition model (cdmdel) is generated, this cdmodel is saved in the openrave database as a function of it's geometry. This means that if some padding is done in the cdmodel, the geometry changes and therefore its name. This is an already reported [behaviour](http://openrave-users-list.185357.n3.nabble.com/Re-Hand-Jacobian-td3506786.html) in OpenRave. **At the end this implies that a new model have to be generated everytime we want to load a padded model.** To solve this problem, the temporal solution is to (since the "load" function always load the not padded cdmodel) to manually rename the padded model to have the same as the not padded one. 

- **Checkselfcollision() does not work in the experiments done with the TEO cdmodel**. These experiments consisted in using a large padding value to force the cdmodel to have some selfcollision without the need of moving the robot. [Here](http://i.imgur.com/ECPloW8.png) is a picture example of this. Different padding values have been tested. However no selfcollision was detected. SelfCollision function was called in two different ways, directly using the robot variable loaded in openrave in two scenarios (with and without cdmodel.setrobot()), and using the robot variable referenced in the cdmodel class this means cdmodel.robot.checkselfcollision(), without success. No collision were detected in any of the cases. As a note there is an [open issue ](https://github.com/rdiankov/openrave/issues/118) in openrave, with the goal of disable selfcollision using cdmodels. However is an OPEN issue, so it should not be implemented. Further study should be done in this direction.

- I am not yet sure of how to use and how [setrobot()](http://openrave.org/docs/0.8.2/_modules/openravepy/databases/convexdecomposition/#ConvexDecompositionModel.setrobot) function works. Maybe this is behind why checkselfcollision does not work.

- When a cdmodel.show() is executed in the code, the execution of the rest of the program stops. Is this normal?

_At this moment the direction of the work done in this issue is to be able to perform a checkselfcollision test using the cdmodel. However this is not as critital as could seem. The reason is because there are many other ways to do a padding in the robot model. This could be done manually changing the xml file of the robot (rude way of doing things), or creating some new trimesh and assign it to the robot collision mesh (something similar as what is done in CD). However right now the best option still is to use the convexdecomposition module. So some more work should be done in this direction._

RaulFdzbis, 2016-10-20:
After some experiments the conclusion is that the **setrobot()** function is in charge of assign the collisionmesh generated with the ConvexDecomposition module to the xml robot. 

This means that in our tests the collisionmesh is changed to the convexdecomposition model. However, the selfcollision stills not detecting the selfcollision that should appear after this padding. Some study about how the selfcollision function works should be done at this moment.

RaulFdzbis, 2016-10-21:
**Update 2:**
CheckSelfcollision() was successfully tested in the new experiments. Openrave is able to detect collision using the cdmodel as the collision mesh. To test this a new experiment was performed. In this experiment instead of using high padding values, the robot was initially [moved](https://github.com/roboticslab-uc3m/teo-main/blob/8426dc81cd7a5b28e6b2736ba9eac4033935e4d6/python/teoCollisionAvoidance/test/test.py#L16-L21) to a collision position. Then a [checkselfcollision test](https://github.com/roboticslab-uc3m/teo-main/blob/8426dc81cd7a5b28e6b2736ba9eac4033935e4d6/python/teoCollisionAvoidance/test/test.py#L54-L58) is performed.

The results of these experiments show how in some positions the robot didnt detect any self collision using the detailed model (without ConvexDecomposition). However, using a padded cdmodel in the same position the robot detected a collision. This means that the padded cdmodel is successfully generated and used for collision  checks in openrave.

jgvictores, 2016-11-10:
Now at: https://github.com/roboticslab-uc3m/openrave-yarp-plugins

jgvictores, 2016-11-14:
Good against simulator at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/8704a332ad4dfd3e044f4b8ffd64d731c59c69c4

jgvictores, 2016-12-03:
Tested and working with simulator and real robot, now only issue is to "Improve implementation and documentation for teo vs teoSim" (https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/13). Closing issue.

jgvictores, 2016-12-03:
Reopening just to reflect state on https://github.com/roboticslab-uc3m/teo-main/projects/1 which is not at global scope of organization.

jgvictores, 2017-02-12:
Finished at around https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/4afbb7cf2e0e81a8c4d537cc7f8c923d208aa69f.

closed 2017-02-12 by jgvictores

jgvictores, 2018-03-14:
This issue was moved to roboticslab-uc3m/openrave-yarp-plugins#84