roboticslab-uc3m/kinematics-dynamics #68: [asr] teoBase needs a reduced dictionary to work

jgvictores, 2016-10-14:
[asr] teoBase needs a reduced dictionary to work. We've been having issues since https://github.com/roboticslab-uc3m/teo-head/issues/32

rsantos88, 2016-11-04:
I think, this issue is concluded with the new version of speechRecognition (see this closed [issue](https://github.com/roboticslab-uc3m/teo-head/issues/42)) and with separated dictionaries (see [this](https://github.com/roboticslab-uc3m/teo-head/pull/47))
Please, close it if you think is that true :)

jgvictores, 2016-11-05:
Okay, code looks good.
I'd prefer to close this issue once we're done with https://github.com/roboticslab-uc3m/teo-head/issues/46 as this is a teoBase/teo-main issue that arose with these updates.

rsantos88, 2016-11-07:
it's ready [:D](https://github.com/roboticslab-uc3m/teo-head/issues/46#issuecomment-258812920) !!

jgvictores, 2016-11-11:
Cool!

closed 2016-11-11 by jgvictores