roboticslab-uc3m/kinematics-dynamics #70: [OpenRave] Need to update documentation!!

rsantos88, 2016-11-23:
Al instalar openrave, y realizar "cmake .." nos encontramos con que nos pide diferentes dependecias.
Faltaría por añadir:
* sudo apt-get install python-dev
* sudo apt-get install libboost-all-dev
... y finalmente me pide instalar FCL.
¿Hace falta instalarlo o se puede desactivar algún modulo desde ccmake?

jgvictores, 2016-11-23:
Enlace?

PeterBowman, 2016-11-23:
Los siguientes tutoriales nos dieron buenos resultados con la compilación de OpenRAVE para su uso con asibot-main, tanto en Ubuntu Trusty como en Xenial:

* http://fsuarez6.github.io/blog/openrave-trusty/
* http://fsuarez6.github.io/blog/workstation-setup-xenial/ (óbviese la parte de ROS)

Si no recuerdo mal, también instalábamos FCL en Trusty aunque en el tutorial correspondiente no se indique (ver los pasos a seguir en el segundo enlace). Nótese que, adicionalmente, tuvimos que desactivar la opción `OPENRAVE_PLUGIN_MOBYRAVE`, posiblemente también `OPT_FLANN` y `OPT_QTOSG_VIEWER` (no recuerdo si venían desactivadas por defecto).

jgvictores, 2016-11-23:
PD: para desactivar FCL, basta con `cmake .. -DOPT_FCL_COLLISION=OFF`

jgvictores, 2016-12-03:
Closing this issue, as `libfcl-dev` is documented in the new location (https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install_openrave.md) and the others where already documented. Thanks to all the participants!

closed 2016-12-03 by jgvictores