roboticslab-uc3m/kinematics-dynamics #79: Error de compilación en teo-main: "no se puede encontrar -lTeoYarp"

rsantos88, 2017-02-19:
Intentando realizar una instalación limpia de teo-main, me encuentro que al compilar me da el siguiente error:
```bash
[  8%] Built target Gait
[  9%] Built target KdlVectorConverterLib
Scanning dependencies of target TrajectoryLib
[ 11%] Building CXX object libraries/TrajectoryLib/CMakeFiles/TrajectoryLib.dir/LineTrajectory.cpp.o
[ 13%] Building CXX object libraries/TrajectoryLib/CMakeFiles/TrajectoryLib.dir/GaitTrajectory.cpp.o
Linking CXX shared library ../../lib/libTrajectoryLib.so
[ 13%] Built target TrajectoryLib
[ 34%] Built target FakeControlboard
[ 57%] Built target FakeControlboardOR
[ 62%] Built target KdlSolver
Scanning dependencies of target BasicCartesianControl
[ 63%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/yarp_plugin_BasicCartesianControl.cpp.o
[ 65%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/DeviceDriverImpl.cpp.o
[ 67%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/ICartesianControlImpl.cpp.o
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::movj(const std::vector<double>&)’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:98:46: warning: ‘virtual bool yarp::dev::IPositionControl::setPositionMode()’ is deprecated (declared at /usr/local/include/yarp/dev/IPositionControl.h:192) [-Wdeprecated-declarations]
     if ( ! iPositionControl->setPositionMode() )
                                              ^
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::movl(const std::vector<double>&)’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:176:39: warning: ‘virtual bool yarp::dev::IVelocityControl::setVelocityMode()’ is deprecated (declared at /usr/local/include/yarp/dev/IVelocityControl.h:137) [-Wdeprecated-declarations]
     iVelocityControl->setVelocityMode();
                                       ^
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::movv(const std::vector<double>&)’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:201:39: warning: ‘virtual bool yarp::dev::IVelocityControl::setVelocityMode()’ is deprecated (declared at /usr/local/include/yarp/dev/IVelocityControl.h:137) [-Wdeprecated-declarations]
     iVelocityControl->setVelocityMode();
                                       ^
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::gcmp()’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:211:35: warning: ‘virtual bool yarp::dev::ITorqueControl::setTorqueMode()’ is deprecated (declared at /usr/local/include/yarp/dev/ITorqueControl.h:60) [-Wdeprecated-declarations]
     iTorqueControl->setTorqueMode();
                                   ^
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::forc(const std::vector<double>&)’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:224:35: warning: ‘virtual bool yarp::dev::ITorqueControl::setTorqueMode()’ is deprecated (declared at /usr/local/include/yarp/dev/ITorqueControl.h:60) [-Wdeprecated-declarations]
     iTorqueControl->setTorqueMode();
                                   ^
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp: In member function ‘virtual bool teo::BasicCartesianControl::stopControl()’:
/home/aion/repos/rsantos88/teo-main/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp:233:39: warning: ‘virtual bool yarp::dev::IPositionControl::setPositionMode()’ is deprecated (declared at /usr/local/include/yarp/dev/IPositionControl.h:192) [-Wdeprecated-declarations]
     iPositionControl->setPositionMode();
                                       ^
[ 68%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/RateThreadImpl.cpp.o
Linking CXX shared module ../../../lib/yarp/BasicCartesianControl.so
[ 68%] Built target BasicCartesianControl
[ 73%] Built target CartesianControlServer
[ 78%] Built target CartesianControlClient
Linking CXX executable ../../bin/teoSim
/usr/bin/ld: no se puede encontrar -lTeoYarp
collect2: error: ld returned 1 exit status
make[2]: *** [bin/teoSim] Error 1
make[1]: *** [programs/teoSim/CMakeFiles/teoSim.dir/all] Error 2
make: *** [all] Error 2
```

jgvictores, 2017-02-19:
Ok, debe estar arreglado en 3856a845a7415c1097032ff30e5ada57eabb100f (branch `issue-79-lTeoYarp`).

Please confirm! :smile:

rsantos88, 2017-02-23:
Perfecto!! Ya está arreglado. Gracias @jgvictores

closed 2017-02-23 by rsantos88