roboticslab-uc3m/kinematics-dynamics #84: Cleaning big file history

jgvictores, 2017-03-05:
Running into:
https://jorin.me/clean-git-history/

Log
```
+ f7b1730...1e89782 color-debug -> color-debug (forced update)
 + 9be2824...5f0228f develop -> develop (forced update)
 + b9e41cc...5d6b30f master -> master (forced update)
 + 7284bd3...9a16c1b v1.0.0 -> v1.0.0 (forced update)
 ! [remote rejected] refs/pull/18/head -> refs/pull/18/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/19/head -> refs/pull/19/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/20/head -> refs/pull/20/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/21/head -> refs/pull/21/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/23/head -> refs/pull/23/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/30/head -> refs/pull/30/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/47/head -> refs/pull/47/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/48/head -> refs/pull/48/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/5/head -> refs/pull/5/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/51/head -> refs/pull/51/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/54/head -> refs/pull/54/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/61/head -> refs/pull/61/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/69/head -> refs/pull/69/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/7/head -> refs/pull/7/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/75/head -> refs/pull/75/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/8/head -> refs/pull/8/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/81/head -> refs/pull/81/head (deny updating a hidden ref)
 ! [remote rejected] refs/pull/83/head -> refs/pull/83/head (deny updating a hidden ref)
error: failed to push some refs to 'https://github.com/roboticslab-uc3m/kinematics-dynamics.git'
```

PeterBowman, 2017-03-06:
I followed a mixture of instructions from http://naleid.com/blog/2012/01/17/finding-and-purging-big-files-from-git-history and https://help.github.com/articles/removing-sensitive-data-from-a-repository/ and it seems to work (the overall size decreased in \~200 MB after having deleted 10\~15 files from history).

The next attachment lists all files with their associated SHA and size in bytes. @jgvictores could you please indicate which ones (maybe on a per-directory basis) should be kept?

```
d890c1ec7ca90122a3c8ecff00df98b5b31d8d36 79759575 share/teoSim/models/teo/rightArm/rightArm5.wrl
62e01ab4d311d666b8d2648ad267c99c83bd78aa 11647523 share/teoSim/models/teo/rightArm/rightArm6.wrl
566ed848f1cba06777b246fc3816cfb1958603ff 11626530 share/openrave/teoDepreciated/leftArm/leftArm6.wrl
3fa8d8fc46c2f1ff9d6b70db41dd78adcb494d8c 11626306 share/openrave/teoDepreciated/rightArm/rightArm6.wrl
504139a77d12483f3fd2789cf02b9313d8953d00 9967136 share/openrave/teo/teo.dae
8fb886cc26f203069f6be5e6f69970e01851b885 8309656 share/openrave/teoDepreciated/teo.robot.collision.pp
87b2ec65af8ded906b4d273e69dfe62e80fdb81e 5127900 superseded/share/teoSim/models/teoWhite/wrls/chest/chest_white.wrl
fa45facdb8f82e922e9e6f8aa2a291737a7fcf17 4333187 superseded/share/teoSim/models/teoWhite/wrls/chest/chest.wrl
8b481111c70c6b24e05e12da265158bf4c70aeec 3880309 superseded/share/teoSim/models/teoWhite/wrls/left_arm/wrist.wrl
cbe16b2a6ff5284a5557a11ceac30f69802c46b0 3486836 superseded/share/teoSim/models/teoWhite/wrls/hip/hip.wrl
086e51052795830a08ef0314718ef2b5393d6f80 2870693 doc/P091.063.CANopen.iPOS.UM.pdf
61e61c0e75d7bb02736f796f20f1273ebfd87998 2745599 share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast0x10000048.Transform6D.4_5_6_7_8_9.cpp
c4dce043484390689ee0b99803b1a88330451635 2310017 superseded/share/teoSim/models/teoWhite/wrls/left_arm/forearm.wrl
19693ff09c4aff304e2322c5d49616a8f3338565 1661983 superseded/share/teoSim/models/teoWhite/wrls/right_arm/forearm.wrl
2f6010f7a56a74246786f4d61a27f737bd0cf366 1525258 superseded/share/teoSim/models/teoWhite/wrls/left_leg/thigh.wrl
dfc2487c50b98d9f201cf777e57900eff9a8df91 1143199 superseded/share/teoSim/models/teoWhite/wrls/right_arm/arm.wrl
cc70b1b20d3cabb1dcc4c7de6b668e7e8472c0d1 1143126 superseded/share/teoSim/models/teoWhite/wrls/left_arm/arm.wrl
66817e4947156993ece44d5a1a750004288612b8 1132585 superseded/share/teoSim/models/teoWhite/wrls/head/xtion.wrl
f607bb71a354c9153776ee4832e3d229e21904a9 1121972 superseded/share/teoSim/models/teoWhite/wrls/right_arm/wrist.wrl
78c616624d9b29cbe103e7a956e71d71350bce45 1112704 superseded/share/teoSim/models/teoWhite/wrls/left_leg/foot.wrl
9525c33a1478df23220567d02e49da18e2832f4b 1066671 superseded/share/teoSim/models/teoWhite/wrls/left_leg/leg.wrl
355c5f856cbed86ce6ba3e2294c0596606bc06e2 1064712 share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast0x10000048.Transform6D.x86_64.4_5_6_7_8_9.so
4f3804f703c5e672f06efdf6357995a559eeab02 830563 test/gtest-1.7.0/fused-src/gtest/gtest.h
53ec2de232349365ff20ada744ca77c23db0b9e2 765905 superseded/share/teoSim/models/teoWhite/wrls/right_leg/ankle.wrl
def1f2ec92005f5b9e859aa16aa8042f63ab7e3d 735014 superseded/share/teoSim/models/teoWhite/wrls/left_arm/prewrist.wrl
918a717e5679d1344bc551389f9e350aab5a291e 731514 superseded/share/teoSim/models/teoWhite/wrls/right_arm/prewrist.wrl
9195534695b887166ab6cc7ca164e36784fcd7b8 692639 superseded/share/teoSim/models/teoWhite/wrls/right_leg/foot.wrl
cd266620cf124440c409b7257d9077f10b337c1f 617252 superseded/share/teoSim/models/teoWhite/wrls/left_leg/hip_yaw.wrl
487a7436abc7e25e33aa0fb83be0bdb84dfcccdc 605462 superseded/share/teoSim/models/teoWhite/wrls/right_leg/leg.wrl
f05efb78fe3401aea62abe78b1d3a4063cce1fd6 599471 superseded/share/teoSim/models/teoWhite/wrls/right_arm/shoulder_roll.wrl
d87efd084f882ec512464f38de1eda4e77661e6b 579024 superseded/share/teoSim/models/teoWhite/wrls/left_arm/shoulder_pitch.wrl
b691e2c032a9c8d1c7aac0549e09fd84710e8c70 577348 superseded/share/teoSim/models/teoWhite/wrls/right_leg/thigh.wrl
582a9a05f5d74634b328520f779e0de4f393ca8c 564464 test/gtest-1.7.0/configure
4c8b0dca34e94a12d8b3dbc5ff90b4613b2360fa 433246 doc/hcanpci_linux_v1535en_edi1878.pdf
b38c5e360dad00c35da450dc46fe2b2b4dada333 419021 doc/fig/src/DH.svg
828104cfde21845bcd4632d75a4f63ce9fca60fb 287004 test/gtest-1.7.0/m4/libtool.m4
c2852d856135458112c39b5f156145980860253a 283680 test/gtest-1.7.0/build-aux/ltmain.sh
fb650b168c33b1666495f12103646dd799f5915b 253128 superseded/share/teoSim/models/teoWhite/wrls/left_arm/shoulder_roll.wrl
bd8578be7589548ebe78dc33bfcc019c7be6ccef 249873 superseded/share/teoSim/models/teoWhite/wrls/chest/backbone.wrl
8adffb7f74b413e4745ceb7a042fa419af414c25 248728 superseded/share/teoSim/models/teoWhite/wrls/right_leg/hip_roll.wrl
0cab07d1564bbf028331068626e708799a5b1513 239177 test/gtest-1.7.0/test/gtest_unittest.cc
7899ad8fbec094183c729d8891d51500e2095b9c 226071 superseded/share/teoSim/models/teoWhite/wrls/left_leg/ankle.wrl
740984407a9791a55bcd6532dfac0eb83db39a5a 224574 superseded/share/teoSim/models/teoWhite/wrls/right_arm/shoulder_pitch.wrl
d5319c131c557d9c9c02e568116d48b226d38411 222092 doc/fig/DH.png
9f5f1c51c58aacf842c6080b9c635b022ee13dcc 215801 share/openrave/teo/models/3.1^torso_simple_links.wrl
4d107d6c3bb07f4dbcbd25bfbfc7c899eb975100 198457 share/openrave/teo/models/5.5^pierna_izq_links.wrl
[... cut ...]
52b38618a074ca422c9db2fddeba09d88b4ea60a 69 libraries/ColorDebug/example/build/CMakeFiles/TargetDirectories.txt
e3b887e66e24eefb448aa2cc7d8e04c52b1ff41e 50 .gitignore
9642fe9a47dd2de81b2399842b633d1392c0126b 30 .gitignore
0faa51a4602ccf330954a31a120f9dafe09729aa 23 superseded/share/kdlcontroller/conf/kdlcontroller.ini
781c7de277467a8195d562b5d08164806763e9bb 22 libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/progress.make
d00491fd7e5bb6fa28c517a0bb32b8b506539d4d 2 libraries/ColorDebug/example/build/CMakeFiles/progress.marks
```

jgvictores, 2017-03-06:
Keep:
- ROOT such as README and .gitignore
- doc
- scripts
- cmake
- programs
- libraries ((except `build`))
- src ((maybe not; everything good has been ported to src/libraries and src/programs))

Which essentially means removing:
- share
- test/gtest
- superseded
- libraries/ColorDebug/example/build

jgvictores, 2017-03-06:
Updated comment above. Essentially we want to get rid of `share/openrave` and `superseded`, and probably `gtest` an `build`while we're at it. :smile:

PeterBowman, 2017-03-06:
OK, thanks! This is the final list of files to be deleted from history:
```
libraries/ColorDebug/example/build/CMakeCache.txt
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CMakeCXXCompiler.cmake
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CMakeDetermineCompilerABI_CXX.bin
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CMakeSystem.cmake
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CompilerIdC/CMakeCCompilerId.c
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CompilerIdCXX/a.out
libraries/ColorDebug/example/build/CMakeFiles/2.8.12.2/CompilerIdCXX/CMakeCXXCompilerId.cpp
libraries/ColorDebug/example/build/CMakeFiles/cmake.check_cache
libraries/ColorDebug/example/build/CMakeFiles/CMakeDirectoryInformation.cmake
libraries/ColorDebug/example/build/CMakeFiles/CMakeOutput.log
libraries/ColorDebug/example/build/CMakeFiles/Makefile.cmake
libraries/ColorDebug/example/build/CMakeFiles/progress.marks
libraries/ColorDebug/example/build/CMakeFiles/TargetDirectories.txt
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/build.make
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/cmake_clean.cmake
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/CXX.includecache
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/DependInfo.cmake
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/depend.internal
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/depend.make
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/flags.make
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/link.txt
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/progress.make
libraries/ColorDebug/example/build/CMakeFiles/testColorDebug.dir/testColorDebug.cpp.o
libraries/ColorDebug/example/build/cmake_install.cmake
libraries/ColorDebug/example/build/Makefile
share/openrave/CMakeLists.txt
share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast0x10000048.Transform6D.4_5_6_7_8_9.cpp
share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast0x10000048.Transform6D.4_5_6_7_8_9.pp
share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast0x10000048.Transform6D.x86_64.4_5_6_7_8_9.so
share/openrave/teoDepreciated/leftArm/kinematics.dc6725ca8537841fafa1f56193170b61/ikfast.h
share/openrave/teoDepreciated/leftArm/leftArm6.wrl
share/openrave/teoDepreciated/rightArm/rightArm6.wrl
share/openrave/teoDepreciated/teo.robot.collision.pp
share/openrave/teoDepreciated/teo.robot.xml
share/openrave/teo/head/head.kinbody.xml
share/openrave/teoLowres/models/TEO_by_links.JPG
share/openrave/teoLowres/teo_in1file.xml
share/openrave/teoLowres/teo_leftArm.kinbody.xml
share/openrave/teoLowres/teo.robot.xml
share/openrave/teo/models/0.0^cabeza_links.wrl
share/openrave/teo/models/0.1^cabeza_links.wrl
share/openrave/teo/models/1.0^brazo_derecho_links.wrl
share/openrave/teo/models/1.1^brazo_derecho_links.wrl
share/openrave/teo/models/1.2^brazo_derecho_links.wrl
share/openrave/teo/models/1.3^brazo_derecho_links.wrl
share/openrave/teo/models/1.4^brazo_derecho_links.wrl
share/openrave/teo/models/1.5^brazo_derecho_links.wrl
share/openrave/teo/models/2.0^brazo_izquierdo_links.wrl
share/openrave/teo/models/2.1^brazo_izquierdo_links.wrl
share/openrave/teo/models/2.2^brazo_izquierdo_links.wrl
share/openrave/teo/models/3.0^torso_simple_links.wrl
share/openrave/teo/models/3.1^torso_simple_links.wrl
share/openrave/teo/models/4.3^pierna_der_links.wrl
share/openrave/teo/models/4.4^pierna_der_links.wrl
share/openrave/teo/models/5.0^pierna_izq_links.wrl
share/openrave/teo/models/5.1^pierna_izq_links.wrl
share/openrave/teo/models/5.2^pierna_izq_links.wrl
share/openrave/teo/models/5.4^pierna_izq_links.wrl
share/openrave/teo/models/5.5^pierna_izq_links.wrl
share/openrave/teo/models/cintura_links.wrl
share/openrave/teo/rightArm/rightArm.kinbody.xml
share/openrave/teo/teo.dae
share/openrave/teo/teo_head.kinbody.xml
share/openrave/teo/teo_trunk.kinbody.xml
superseded/share/kdlcontroller/conf/kdlcontroller.ini
superseded/share/teoCartesianServer/conf/teoCartesianServer.ini
superseded/share/teoCartesianServer/conf/teoCartesianServerMonolithic.ini
superseded/share/teoCartesianServer/scripts/teoCartesianServer.xml.template
[... 116 lines cut ...]
test/gtest-1.7.0/test/gtest_pred_impl_unittest.cc
test/gtest-1.7.0/test/gtest_premature_exit_test.cc
test/gtest-1.7.0/test/gtest-printers_test.cc
test/gtest-1.7.0/test/gtest_repeat_test.cc
test/gtest-1.7.0/test/gtest_shuffle_test.py
test/gtest-1.7.0/test/gtest_stress_test.cc
test/gtest-1.7.0/test/gtest-test-part_test.cc
test/gtest-1.7.0/test/gtest_test_utils.py
test/gtest-1.7.0/test/gtest_throw_on_failure_ex_test.cc
test/gtest-1.7.0/test/gtest-tuple_test.cc
test/gtest-1.7.0/test/gtest-typed-test_test.cc
test/gtest-1.7.0/test/gtest-unittest-api_test.cc
test/gtest-1.7.0/test/gtest_unittest.cc
test/gtest-1.7.0/test/gtest_xml_output_unittest_.cc
test/gtest-1.7.0/test/gtest_xml_output_unittest.py
test/gtest-1.7.0/test/gtest_xml_test_utils.py
test/gtest-1.7.0/xcode/Config/DebugProject.xcconfig
test/gtest-1.7.0/xcode/Config/General.xcconfig
test/gtest-1.7.0/xcode/Config/ReleaseProject.xcconfig
test/gtest-1.7.0/xcode/Config/StaticLibraryTarget.xcconfig
test/gtest-1.7.0/xcode/gtest.xcodeproj/project.pbxproj
test/gtest-1.7.0/xcode/Resources/Info.plist
test/gtest-1.7.0/xcode/Scripts/runtests.sh
test/gtest-1.7.0/xcode/Scripts/versiongenerate.py
```

jgvictores, 2017-03-06:
LGTM!!

PeterBowman, 2017-03-06:
This is taking a bit too long, I'll report any progress in a few hours.

PeterBowman, 2017-03-06:
It didn't seem to work last night, but after deleting `share/openrave/` from the working tree, the overall size of the source directory dropped to 74,5 MB (was ~1.8 GB). I'll run the script one more time to make sure it's purged once and for all, might take up to 2 hours.

@jgvictores are you OK with force-pushing this to GitHub? I'm afraid the tag (v1.0.0) was lost in the process, although it could be reinstated with the correct commit reference. Apart from that, I had to merge #83 into develop again since my local copy wasn't updated before BFG rewrote the entire history.

jgvictores, 2017-03-06:
No problem losing v1.0.0.

1.8 Gb to < 80 Mb sounds great!!

PeterBowman, 2017-03-06:
For future reference, these are the steps I followed to purge a set of files from Git history:

1. Retrieve a list of SHAs associated to every object (needed later on):
    ```bash
    git rev-list --objects --all | sort -k 2 > allfileshas.txt
    ```

2. Get the latest SHA for committed files and sort them by size:
    ```bash
    git gc && git verify-pack -v .git/objects/pack/pack-*.idx | egrep "^\w+ blob\W+[0-9]+ [0-9]+ [0-9]+$" | sort -k 3 -n -r > bigobjects.txt
    ```

3. Associate previous list entries with actual file names:
    ```bash
    for SHA in `cut -f 1 -d\  < bigobjects.txt`; do
    echo $(grep $SHA bigobjects.txt) $(grep $SHA allfileshas.txt) | awk '{print $1,$3,$7}' >> bigtosmall.txt
    done;
    ```

4. Apply some regex magic on `bigtosmall.txt` to filter out the path+filename column, sort it for convenience and remove the entries that you want to keep in history.

5. Make sure those files are no longer present in the working tree at **any** branch tip (that is, checkout branch by branch, remove unwanted files and commit the new state of the repo).

6. Remove files from history. Replace `DELETE_THIS` with the file path (relative to the root directory of the source tree):
    ```bash
    git filter-branch --force --prune-empty --index-filter 'git rm -rf --cached --ignore-unmatch DELETE_THIS' --prune-empty --tag-name-filter cat -- --all
    ```
    For convenience, you may want to save this code in `process.sh`:
    ```bash
    #!/bin/bash
    while IFS='' read -r line || [[ -n "$line" ]]; do
        echo "Text read from file: $line"
        git filter-branch --force --prune-empty --index-filter 'git rm -rf --cached --ignore-unmatch '"$line"'' --prune-empty --tag-name-filter cat -- --all
    done < "$1"
    ```
    And iterate through all files listed in `bigtosmall.txt` by issuing:
    ```bash
    sudo chmod +x process.sh
    ../process.sh bigtosmall.txt
    ```

7. The history should be clean now and you may force-push all branches and tags to the remote repo:
    ```bash
    git push origin --force --all
    git push origin --force --tags
    ```

8. Traces of deleted files are kept in a backup ref in your local tree. To purge them definitely, type:
    ```bash
    git for-each-ref --format='delete %(refname)' refs/original | git update-ref --stdin
    git reflog expire --expire=now --all
    git gc --prune=now # --aggressive
    ```
    See `man git gc` for details on the `--aggressive` option.

PeterBowman, 2017-03-06:
@jgvictores I realize now BFG was the better way to go despite that error. I couldn't beat the 7x MB mark while your solution reached only 22.1 MB (perform a fresh clone and check the size). Moreover, GitHub seems to store old refs that you can access at any time from the web, e.g. see f8d03d0:
* browse the repo at that point in history and find that [this file](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/f8d03d0/share/openrave/teoDepreciated/leftArm/leftArm0.wrl.REMOVED.git-id) was removed
* in the commit description, BFG inserted the previous commit SHA, namely b76a85c
* follow the same steps - the full, 43 MB model can be downloaded from [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/b76a85c7e29b6e4c3791f3524e244a0ec8c109b3/share/openrave/teoDepreciated/leftArm/leftArm0.wrl)
* upon cloning in your local machine, old refs are not downloaded and the repo size is kept small, which is exactly what we wanted

I must have overlooked some crucial step in the process since all models may still be browsed at older revisions (check my [fork](https://github.com/PeterBowman/kinematics-dynamics/network)). Anyway, I bet BFG didn't take that much time. If those `refs/pull/` errors haven't blown anything up, I'd leave the repo as it is now.

BTW care must be taken to not reintroduce deleted files with future merges. From https://help.github.com/articles/removing-sensitive-data-from-a-repository/:

> Tell your collaborators to rebase, not merge, any branches they created off of your old (tainted) repository history. One merge commit could reintroduce some or all of the tainted history that you just went to the trouble of purging.

jgvictores, 2017-03-06:
Announcement put on #85. @PeterBowman  Safe to close this issue?

PeterBowman, 2017-03-06:
> Safe to close this issue?

Yes, thanks.

closed 2017-03-06 by PeterBowman