roboticslab-uc3m/kinematics-dynamics #86: cartesian controller mismatches

munozyanez, 2017-03-07:
After integration of GaitSupportPoligon in TrajectoryLib, and testing in teoSim, I have noticed a big difference between the target joint positions and the actual (in teoSim) joint positions.
Find attached the output from  "yarpdev --device BasicTwoLimbCartesianControl" command.

Steps to reproduce:
1-(newterminal)$ yarp server
2-(newterminal)$ teoSim
3-(newterminal)$ yarpdev --device BasicTwoLimbCartesianControl
4-(newterminal)$ yarp rpc /TwoLimbCartesianControl/rpc:s
5-write "step" in the console of point 4 (the one of yarp rpc).
6-Watch the output in the console of point number 3.

The output follows this block format:
> rfX: 0,-0.0085,-0.765,> rot: -0.707106,-0.00117567,0.707106,180.135, //target cartesian right foot
> lfX: 0,0.2485,-0.765,> rot: 0,-1,0,90,  //target cartesian left foot
> rfXdot: 0,0,0,> rotdot: -1,-7.3914e-17,-1.45648e-15,2.86479, //velocity cartesian right foot
> lfXdot: 0,0,0,> rotdot: 0,0,0,0,  //velocity cartesian left foot
[STEP] [2.066506] x 0.000000 -0.008500 -0.765000 -0.707106 -0.001176 0.707106 180.134722 0.000000 0.248500 -0.765000 0.000000 -1.000000 0.000000 90.000000 xdot 0.000000 0.000000 0.000000 -1.000000 -0.000000 -0.000000 2.864789 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
[STEP] [2.066506] 0.000142 -0.000273 -0.003718 -0.953521 -0.000456 0.003304 -> -0.159662 -11.351168 7.405704 -15.843832 8.413246 65.983875 [deg/s]
rightLeg encoders: [0.00652374, -12.9386, -24.9962, 52.7143, -27.6868, 15.6016] //actual values
[STEP] [2.066506] 0.000185 -0.000376 -0.003315 0.002186 -0.000498 0.000006 -> 0.006784 0.138676 0.556850 -1.216207 0.630110 -0.263936 [deg/s]
leftLeg encoders: [-0.00667999, -12.4872, -26.6057, 56.1492, -29.5143, 12.6125]  //actual values

jgvictores, 2017-03-07:
Hi!
If you look at the terminal output of `teoSim`, there are `OpenRAVE` warnings like:
```
2017-03-07 23:06:28,967 openrave [WARN] [kinbody.cpp:1504 KinBody::SetDOFValues] dof 20 value is not in limits -3.193866e-01<-3.141593e-01
```

`OpenRAVE` begins to do pretty strange things once you're off limits.

munozyanez, 2017-03-08:
Never had this issue in the past, but let me check and back with the results.

munozyanez, 2017-03-08:
Nope, tried removing the limits in both robot descriptions (openrave and kdl) with no luck. Results are the same:
> rfX: 0,-0.0085,-0.765,> rot: -0.707106,-0.00171292,0.707106,180.196,
> lfX: 0,0.2485,-0.765,> rot: 0,-1,0,90,
> rfXdot: 0,0,0,> rotdot: -1,-7.3914e-17,-1.45648e-15,2.86479,
> lfXdot: 0,0,0,> rotdot: 0,0,0,0,
[STEP] [2.096897] x 0.000000 -0.008500 -0.765000 -0.707106 -0.001713 0.707106 180.196286 0.000000 0.248500 -0.765000 0.000000 -1.000000 0.000000 90.000000 xdot 0.000000 0.000000 0.000000 -1.000000 -0.000000 -0.000000 2.864789 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
[STEP] [2.096897] 0.000130 -0.002140 -0.004483 -0.954052 -0.000426 0.004758 -> -0.241589 -11.150425 7.484944 -16.057169 8.552755 65.813568 [deg/s]
rightLeg encoders: [0.00298635, -13.1343, -25.0526, 52.8299, -27.7456, 15.7669]
[STEP] [2.096897] 0.000228 -0.002045 -0.004076 0.002504 -0.000570 0.000006 -> 0.007951 0.328917 0.627822 -1.375833 0.714491 -0.472366 [deg/s]
leftLeg encoders: [-0.00771276, -12.6762, -26.6798, 56.3152, -29.6019, 12.8196]

jgvictores, 2017-03-08:
Is the teoSim console warning still there?

munozyanez, 2017-03-08:
Sure, the limits are
      -90,90. No out of limits message at all.
    Jorge Muñoz
munoz.yanez@gmail.com
    On 08/03/17 13:35, Juan G. Victores
      wrote:
    
    
      Is the teoSim console warning still there?
      —
          it on GitHub, or mute
          the thread.
      
        
          
          
        
        
      
slab-uc3m/kinematics-dynamics"</a>}},"updates":{"snippets":[{"icon":"PERSON","message":"@jgvictores in #86: Is the teoSim console warning still there? "}],"action":{"name":"View Issue","url":<a class="moz-txt-link-rfc2396E" href="https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/86#issuecomment-285029494">"https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/86#issuecomment-285029494"</a>}}}

jgvictores, 2017-03-08:
I'm referring to console output in yellow, where you launch teoSim. 
Have you made any new commits regarding limits?

munozyanez, 2017-03-08:
Of course, no out of limits warnings in the teoSim output console...

munozyanez, 2017-03-08:
@jgvictores , find also the dirty fix, as requested, in this branch:
https://github.com/roboticslab-uc3m/teo-openrave-models/tree/testing

jgvictores, 2017-03-08:
I'm suspicious two things are happening here:
1. For some reason, you have console output of OpenRAVE warnings deactivated.
2. The hack introduced in https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/4fc24d4b3cec3b11eee5d4682266ab203b35c2f2 as recommended in https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/82#issuecomment-284444029 was not a good one.

In consequence, please launch `teoSim` as in the following line, and tell me if the encoder value issue is still present:
```
teoSim --env /usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml
```

munozyanez, 2017-03-08:
As expected, the same behavior.
What do you mean by encoder value issue?

jgvictores, 2017-03-08:
> What do you mean by encoder value issue?

I was referring to your initial concern: `I have noticed a big difference between the target joint positions and the actual (in teoSim) joint positions`

munozyanez, 2017-03-08:
Oh, yes, same big difference.

munozyanez, 2017-03-15:
New data for the issue.
I tried to change frame (from relative to absolute)  for angular velocities provided to function

 bool teo::GaitTrajectory::getXdot(const double movementTime, std::vector<double>& xdot)

Having similar results.
IMHO,  we can discard the trajectories commanded as a source for the problem.

munozyanez, 2017-03-24:
Such a honorable quest, but I am not the best candidate for this task. I propose @jgvictores  for this. At the moment I am full involved with ZMP.

munozyanez, 2017-03-24:
Reassigned by request of @jgvictores and under the conditions that I will not spend any time in it but some testing.

PeterBowman, 2017-07-08:
Might not be the root cause, but I'm quite confident that the issue described in the commit comments at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/446a6b3#diff-c0546e6bbb0f0bbf3295a6b885f71cfe is related. Copy-pasting since they may not last forever:

> PeterBowman 2017-07-05T22:39:39Z
>
> @munozyanez @jgvictores unless I'm really wrong, `desiredXdotA` stores **7** elements corresponding to the translation + orientation (as **axis-angle**) of limb A computed at `movementTime`. Then, a per-element addition is performed inside the for-loop between this collection and a **6**-element vector of linear (3) and angular (3) velocities - `commandXdotA` ([implementation](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/446a6b34d1ad54357b7a2ac3be39188140af04c7/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp#L47-L52)). Is this correct? By contrast, [`LineTrayectory::getXdot`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/dea1670121724c00ec6e13ecb894812af31fe988/libraries/TrajectoryLib/LineTrajectory.cpp#L24-L29) fills a 6-element twist vector.

> munozyanez 2017-07-06T12:14:21Z
>
> As I see, after calling [`vectorToFrame(xd,frameXd)`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/libraries/KdlVectorConverterLib/KdlVectorConverter.cpp#L14-L41) inside [`fwdKinError`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp#L47-L71) , the input value of  [`iCartesianSolverA->fwdKinError(desiredXA,currentQA, commandXdotA)`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/libraries/TeoYarp/BasicTwoLimbCartesianControl/RateThreadImpl.cpp#L56) will be expected according to `angleRepr` variable.

> PeterBowman 2017-07-06T12:31:56Z
>
> What about the output, that is, `commandXdotA`? This is a 6-element twist vector which carries no information about the angle representation. The for-loop is correctly set to iterate over its six elements, but the operand `desiredXdotA` is my main concern. According to [`GaitTrajectory::getXdot`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/dea1670121724c00ec6e13ecb894812af31fe988/libraries/TrajectoryLib/GaitTrajectory.cpp#L102-L159), the output vector `xdot`/`desiredXdot` is aware of the axis-angle representation and stores 7 elements for each limb. How can you add these two magnitudes?

> munozyanez 2017-07-06T14:35:04Z
>
> You are right. It can only work fine if desiredXdot rotation part (last three values) define a twist.
Lets wait for @jgvictores comment, as he did most of this codes.

> jgvictores 2017-07-07T14:15:00Z
>
> I agree with @PeterBowman . In the old code we were using the **6**-element vector `fwdKinError` that is returned as can be seen in [its code](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/446a6b34d1ad54357b7a2ac3be39188140af04c7/libraries/TeoYarp/KdlSolver/ICartesianSolverImpl.cpp#L31-L55). This looks more dangerous.

> munozyanez 2017-07-07T14:21:05Z
>
> Yes, we all agree in the error, the thing is what solution fits better.
> One easy option is to convert the axis-angle to twist and keep all the same, but maybe we can find a better solution.
> As I said, this code is mostly yours, so what is your preference?

> jgvictores 2017-07-07T14:43:45Z
>
> `desiredXdotA` and in general all `Xdot` should be **6**-element vectors as in twists, right?

> jgvictores 2017-07-07T14:46:06Z
>
> Which should go all the way back to [Trajectory](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/dea1670121724c00ec6e13ecb894812af31fe988/libraries/TrajectoryLib/Trajectory.hpp#L31) and therefore [GaitTrajectory](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/dea1670121724c00ec6e13ecb894812af31fe988/libraries/TrajectoryLib/GaitTrajectory.cpp#L102-L159).

> munozyanez 2017-07-07T14:59:38Z
>
> As you want. My opinion is that I would keep both getX and getXdot using the same parameter format. Both twist, or both axis-angle.

cc @munozyanez, @jgvictores

munozyanez, 2017-07-09:
Totally agree about this two can be related.
We must remember to check this issue after fixing the other.

PeterBowman, 2017-08-22:
`getXdot` now returns a twist at ba24510 (axis-angle scaled by the rotation angle).

PeterBowman, 2017-08-25:
At some point, `GaitTrajectory::getX` should get updated to return positions in scaled axis-angle notation so that it's consistent with other methods (see #113). I'm already reflecting this usage in docs (#107).

munozyanez, 2017-08-26:
Ok, written down in todo list. For now, the function is used in different programs, so I will need to refactor a little. It will take some time.

munozyanez, 2017-09-05:
Ok, it's done.
Now there is a question. I've taken scale angle to be in [rad], but haven't found documentation on it. Is there any? If don't, it should be somewhere.
Also, what was needed in BasicTwoLimbCartesianControl to change?

PeterBowman, 2017-09-05:
> Now there is a question. I've taken scale angle to be in [rad], but haven't found documentation on it. Is there any? If don't, it should be somewhere.

It was noted here:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2c4e7712e1f46a91052ec68eb1c475ed8cc3b7ab/libraries/TrajectoryLib/Trajectory.hpp#L32

and here:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2c4e7712e1f46a91052ec68eb1c475ed8cc3b7ab/libraries/TeoYarp/ITwoLimbCartesianControl.h#L49

but I might have overlooked some other places.

> Also, what was needed in BasicTwoLimbCartesianControl to change?

You need to make sure that the whole call chain is consistent. Once `GaitTrajectory::getX` is fixed, the output vector must reflect its new size (2\*6 elements instead of 2\*7) at:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2c4e7712e1f46a91052ec68eb1c475ed8cc3b7ab/libraries/TeoYarp/BasicTwoLimbCartesianControl/RateThreadImpl.cpp#L50-L51

With that done, both KdlSolver and GaitTrajectory will correctly handle scaled axis-angle parameters and BasicCartesianControl would pass them to each other. The network wrapper (TwoLimbCartesianControlServer) should require no action, but in case you want its `stat` command to return values in other representations, you may want to follow this (slightly convoluted) path currently adopted in CartesianControlServer:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2c4e7712e1f46a91052ec68eb1c475ed8cc3b7ab/libraries/TeoYarp/CartesianControlServer/RpcResponder.cpp#L177-L180

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2c4e7712e1f46a91052ec68eb1c475ed8cc3b7ab/libraries/TeoYarp/CartesianControlServer/RpcResponder.cpp#L66-L70

munozyanez, 2017-09-05:
All right. Done.
Thanks a lot @PeterBowman for your invaluable help.

jgvictores, 2017-12-04:
> All right. Done.
Thanks a lot @PeterBowman for your invaluable help.

Mmm... looks like I'm late for the fun...
@munozyanez Does that mean we can close this issue?

munozyanez, 2017-12-04:
Don't see why not. I'm closing.

closed 2017-12-04 by munozyanez

jgvictores, 2017-12-04:
@munozyanez Thanks!!

jgvictores, 2017-12-04:
@PeterBowman Thanks!

PS to both: Assigning, as always, is to keep historic on who did the work on closed issues.