roboticslab-uc3m/kinematics-dynamics #87: Denavit Hartenberg revision for legs

munozyanez, 2017-03-09:
Recent merge (https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/83) has uploaded a revised DH parameters configuration in the following folder:
- https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/develop/share/kinematics
To be more specific, files 
- https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/share/kinematics/leftLegKinematics.ini
- https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/share/kinematics/rightLegKinematics.ini
Have changed according  DH Parameter Table published in http://robots.uc3m.es/index.php/TEO_Diagrams .

jgvictores, 2017-03-09:
@munozyanez Thanks a lot for this info. It will be very helpful in the future, and also in the present debugging issues such as #86.

munozyanez, 2017-03-09:
By the way, I would like to propose to put all this files and ( https://github.com/roboticslab-uc3m/teo-openrave-models ) under specific teo-config repo. I mean, following the idea of teo-openrave-models, put all configuration files together can be useful for code cleanliness, and reuse with other robots.

jgvictores, 2017-03-09:
> By the way, I would like to propose to put all this files and ( https://github.com/roboticslab-uc3m/teo-openrave-models ) under specific teo-config repo. I mean, following the idea of [...]

Yes, this is definitely a good idea. The only consideration is putting config files directly in teo-main, or in a new teo-config or teo-configuration-files etc, which will then be YCM'ed into teo-main. I'll be working on this tomorrow and will probably take a look at how it works directly in teo-main, without discarding the option of a new repo.

munozyanez, 2017-03-24:
I can do it, but I am very busy at the moment. I can take if this is a long term assignment.

munozyanez, 2017-03-24:
As you asked, the minimum knee angle for static gait for the robot is asin(0.1285/0.845) = 8.7 deg.
This angle sets the foot just under the root point. Angle must be a bit bigger.

PeterBowman, 2017-08-16:
This issue was moved to roboticslab-uc3m/teo-main#18

closed 2022-05-22 by PeterBowman