roboticslab-uc3m/kinematics-dynamics #89: What to do with teo-post-install.md ?

jgvictores, 2017-03-20:
What to do with [https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c14c5daaaa89f3f45e16166bc20665da233b11d4/doc/teo-post-install.md](teo-post-install.md) ?

Seems like https://github.com/roboticslab-uc3m/teo-software-manual because it can require sim and cartesian, but do we want to limit it to TEO ?

PeterBowman, 2017-10-22:
Robot related stuff could ultimately end up in its superbuild orchestrating repo (e.g. [teo-main](https://github.com/roboticslab-uc3m/teo-main/), take a look at the issues it currently hosts) - provided we go on working with YCM (roboticslab-uc3m/QA#1).

For something more generic, maybe use QA? See roboticslab-uc3m/QA#10.

PeterBowman, 2017-10-22:
Fixed link in the description. BTW a very similar .md guide at asibot-main may have gone unnoticed: [asibot-post-install.md](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5b6e049be9a59e1c11b807faefe51fee6f/doc/asibot-post-install.md).

PeterBowman, 2018-06-28:
The old *best-practices* repo has been renamed to [*developer-manual*](https://github.com/roboticslab-uc3m/developer-manual) and split into categories. Well, a RoboticsLab-oriented robot-agnostic simulator guide might fit there.

jgvictores, 2020-10-01:
First 1/3 migrated at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/305dfdf6b6f6924adfa02b336fa587831a28e420

jgvictores, 2020-10-01:
https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c14c5daaaa89f3f45e16166bc20665da233b11d4/doc/teo-post-install.md#interfacing used to point to something similar to https://sourceforge.net/p/roboticslab/code/2578/tree/asibot/main/src/modules/testRaveBot/main.cpp

jgvictores, 2020-10-01:
Last 1/3 migrated at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/ba72bb600b2008982387a243c67c404a73432584

jgvictores, 2020-10-17:
Middle 2/3 was on teoSim parameters. Only parameter documented at https://github.com/roboticslab-uc3m/teo-developer-manual/commit/007ccd05e8643f49254d12e8ae45ca0e0daa8aaf

closed 2020-10-17 by jgvictores