roboticslab-uc3m/kinematics-dynamics #91: FindGAIT.cmake is too hard-coded

jgvictores, 2017-04-09:
FindGAIT.cmake is too hard-coded, especially [this line](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/fix-90-travis/cmake/FindGAIT.cmake#L8) if it ever makes it into `develop`.

PeterBowman, 2017-04-09:
We could get rid of `FindGAIT.cmake` if Gait started using the `export(PACKAGE)` command ([docs](https://cmake.org/cmake/help/v3.0/command/export.html)). Or keep it as a fallback.

jgvictores, 2017-04-09:
Yes, I totally agree. This is yet another of those issues that are mainly intended to remember what is where. :slightly_smiling_face:

PeterBowman, 2017-07-18:
`FindGAIT.cmake` is gone for good after the last CMake code review: 6f90a96. Downstream maintainers can find an updated usage guide in README.md at https://github.com/roboticslab-uc3m/gait.

closed 2022-05-22 by PeterBowman