roboticslab-uc3m/kinematics-dynamics #93: testBasicCartesianControl ocasionally fails

jgvictores, 2017-04-14:
Can happen with `gcc`and/or `clang`. There are at several threads (at `FakeControlboard` level, at `BasicCartesianControl` level), so there may be race conditions (during #92 many fixes were made to `stop`(join) threads and `close` devices). Example:
- No output has been received in the last 10m0s: https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/222171839
- Segfault https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/222171421

PeterBowman, 2017-05-02:
These are the last lines I see on my PC upon a faulty test run:

```
[FakeControlboard] close()
pure virtual method called
temrinate called without an active exception
Aborted (core dumped)
```

Something I found on the web, not sure how/if this applies here: https://tombarta.wordpress.com/2008/07/10/gcc-pure-virtual-method-called/.

jgvictores, 2017-05-02:
> Something I found on the web, not sure how/if this applies here: https://tombarta.wordpress.com/2008/07/10/gcc-pure-virtual-method-called/.

Yes, it's definitely a situation like the above described, involving deletes (some called by `close` members) and threads. Just not extremely high priority to debug right now. :slightly_smiling_face:

PeterBowman, 2017-05-03:
Raising priority since this particular bug is spoiling (almost) every single Travis build.

PeterBowman, 2017-12-06:
I made `FakeControlboard::run` print a debug message right after entering the method's body. This is an excerpt from a successful Travis CI build:

```
...
[INFO]created device <KdlSolver>. See C++ class roboticslab::KdlSolver for documentation.
FakeControlboard not using individual initPoss, defaulting to genInitPos.
FakeControlboard not using individual jointTols, defaulting to genJointTol.
FakeControlboard not using individual maxLimits, defaulting to genMaxLimit.
FakeControlboard not using individual minLimits, defaulting to genMinLimit.
FakeControlboard not using individual refSpeeds, defaulting to genRefSpeed.
FakeControlboard not using individual encRawExposeds, defaulting to genEncRawExposed.
FakeControlboard not using individual velRawExposeds, defaulting to genVelRawExposed.
[FakeControlboard] success: threadInit()
[INFO]created device <FakeControlboard>. See C++ class roboticslab::FakeControlboard for documentation.
[info] IPositionImpl.cpp:9 getAxes(): Reporting 1 axes are present
[info] DeviceDriverImpl.cpp:64 open(): numRobotJoints: 1.
[info] DeviceDriverImpl.cpp:67 open(): numSolverJoints: 1.
[FakeControlboard] Range of axis 0 read: -180.000000 to 180.000000.
[info] DeviceDriverImpl.cpp:81 open(): Joint 0 limits: [-180.000000,180.000000]
In FakeControlboard::run()
[INFO]created device <BasicCartesianControl>. See C++ class roboticslab::BasicCartesianControl for documentation.
In FakeControlboard::run()
...
```

And now the buggy one:

```
...
[INFO]created device <KdlSolver>. See C++ class roboticslab::KdlSolver for documentation.
FakeControlboard not using individual initPoss, defaulting to genInitPos.
FakeControlboard not using individual jointTols, defaulting to genJointTol.
FakeControlboard not using individual maxLimits, defaulting to genMaxLimit.
FakeControlboard not using individual minLimits, defaulting to genMinLimit.
FakeControlboard not using individual refSpeeds, defaulting to genRefSpeed.
FakeControlboard not using individual encRawExposeds, defaulting to genEncRawExposed.
FakeControlboard not using individual velRawExposeds, defaulting to genVelRawExposed.
[FakeControlboard] success: threadInit()
In FakeControlboard::run()
[INFO]created device <FakeControlboard>. See C++ class roboticslab::FakeControlboard for documentation.
[info] IPositionImpl.cpp:9 getAxes(): Reporting 1 axes are present
[info] DeviceDriverImpl.cpp:64 open(): numRobotJoints: 1.
[info] DeviceDriverImpl.cpp:67 open(): numSolverJoints: 1.
[FakeControlboard] Range of axis 0 read: -180.000000 to 180.000000.
[info] DeviceDriverImpl.cpp:81 open(): Joint 0 limits: [-180.000000,180.000000]
[INFO]created device <BasicCartesianControl>. See C++ class roboticslab::BasicCartesianControl for documentation.
[build stalls]
```

Notice the placement of said debug message (`In FakeControlboard::run()`).

PeterBowman, 2017-12-11:
I noticed that cron builds rarely fail (maybe even not at all). I'm using a new Travis feature to launch customized builds ([blog entry](https://blog.travis-ci.com/2017-08-24-trigger-custom-build)): let's see how it behaves when forcing [cron jobs](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/314836016) (i.e. pull YARP's `devel` branch) vs [non-cron ones](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/314834865) (YARP at `master`).

PeterBowman, 2017-12-11:
Cron job completed successfully, last one that failed (and *not* because of the infamous YCM hash mismatch error) was https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/241719930 (11th June 2017, robotology/yarp@b6654ad). It doesn't make sense to me anyway, that commit was merged to YARP `master` since the last release.

PeterBowman, 2017-12-16:
Need to check this ([ref](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/e8f39911a2d69aea55b4153c2a9dbb1679336a37/libraries/YarpPlugins/FakeControlboard/DeviceDriverImpl.cpp#L118)):

```c++
// Start the RateThread
this->setRate(jmcMs);
this->start();
```

FakeControlBoard's JMC rate thread period may evaluate to whatever was stored in memory in the previous run since `jmcMs` is never initialized (neither in the constructor nor in `DeviceDriver::open`). Might be fixed after roboticslab-uc3m/openrave-yarp-plugins@9a2785e.

PeterBowman, 2017-12-16:
So far, it's working. Of course, this could be totally unrelated, so let's give it some time (a generous bunch of CI builds) and check it once again.

@jgvictores I've added the following entry to our C++ guide: roboticslab-uc3m/best-practices@bd291bf. It's something worth worrying about and already gave me some headaches at asrob-uc3m/robotDevastation@5f8aff2.

jgvictores, 2017-12-16:
Cool, thanks a lot for all that!

PeterBowman, 2017-12-29:
No errors reported since 5e98536 + roboticslab-uc3m/openrave-yarp-plugins@9a2785e, closing.

closed 2017-12-29 by PeterBowman