roboticslab-uc3m/kinematics-dynamics #94: step test failed in TwoLimbBasicCartesianControl

munozyanez, 2017-04-18:


munozyanez, 2017-04-18:
/TwoLimbCartesianControl/rpc:s is failing at step command:
```
$ yarpdev --device BasicTwoLimbCartesianControl
[info] DeviceDriverImpl.cpp:14 open(): Subdevice BasicTwoLimbCartesianControl
[debug] DeviceDriverImpl.cpp:9 open(): Config: (single_threaded 1) (device BasicTwoLimbCartesianControl) (subdevice BasicTwoLimbCartesianControl).
[info] DeviceDriverImpl.cpp:30 configureLimbA(): solver: KdlSolver [KdlSolver]
[info] DeviceDriverImpl.cpp:34 configureLimbA(): angleRepr: axisAngle [axisAngle]
[info] DeviceDriverImpl.cpp:38 configureLimbA(): kinematics: rightLegKinematics.ini [rightLegKinematics.ini]
[debug] DeviceDriverImpl.cpp:10 open(): config: (angleRepr axisAngle) (kinematics rightLegKinematics.ini) (device KdlSolver).
[info] DeviceDriverImpl.cpp:14 open(): kinematics: rightLegKinematics.ini [none.ini]
[debug] DeviceDriverImpl.cpp:24 open(): fullConfig: (angleRepr axisAngle) (kinematics rightLegKinematics.ini) (H0 (0 1 0 0 1 0 0 -0.146 0 0 -1 -0.092 0 0 0 1)) (numLinks 6) (link_4 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (min -18.0) (max 20.0)) (link_3 (offset 0.0) (D 0.0175) (A 0.3) (alpha 0.0) (min -5.0) (max 80.0)) (link_5 (offset 0.0) (D 0.0) (A 0.123005) (alpha 0.0) (min -25.0) (max 25.0)) (device KdlSolver) (link_2 (offset 0.0) (D 0.0) (A 0.33) (alpha 0.0) (min -30.0) (max 45.0)) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (min -90.0) (max 90.0)) (link_1 (offset 90.0) (D 0.0) (A 0.0) (alpha 90.0) (min -20.0) (max 20.0)).
[info] DeviceDriverImpl.cpp:28 open(): numLinks: 6 [1]
[info] DeviceDriverImpl.cpp:32 open(): angleRepr: axisAngle [axisAngle]
[info] DeviceDriverImpl.cpp:59 open(): gravity: 0.0 0.0 -9.81 [0.0 0.0 -9.81]
[info] DeviceDriverImpl.cpp:75 open(): H0:
 0.000000	 1.000000	 0.000000	 0.000000
 1.000000	 0.000000	 0.000000	-0.146000
 0.000000	 0.000000	-1.000000	-0.092000
 0.000000	 0.000000	 0.000000	 1.000000
[ 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000]
[info] DeviceDriverImpl.cpp:107 open(): Added: link_0 (offset 0.000000) (D 0.000000) (A 0.000000) (alpha 90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_1 (offset 90.000000) (D 0.000000) (A 0.000000) (alpha 90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_2 (offset 0.000000) (D 0.000000) (A 0.330000) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_3 (offset 0.000000) (D 0.017500) (A 0.300000) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_4 (offset 0.000000) (D 0.000000) (A 0.000000) (alpha -90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_5 (offset 0.000000) (D 0.000000) (A 0.123005) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:172 open(): HN:
 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000
[ 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000]
[info] DeviceDriverImpl.cpp:175 open(): Chain number of segments (post- H0 and HN): 8
[info] DeviceDriverImpl.cpp:176 open(): Chain number of joints (post- H0 and HN): 6
[INFO]created device <KdlSolver>. See C++ class teo::KdlSolver for documentation.
[info] DeviceDriverImpl.cpp:61 configureLimbA(): remote: /teo/rightLeg [/teo/rightLeg]
yarp: Port /BasicTwoLimbCartesianControl/teo/rightLeg/rpc:o active at tcp://2.2.2.210:10033
yarp: Port /BasicTwoLimbCartesianControl/teo/rightLeg/command:o active at tcp://2.2.2.210:10034
yarp: Port /BasicTwoLimbCartesianControl/teo/rightLeg/state:i active at tcp://2.2.2.210:10035
yarp: Port /BasicTwoLimbCartesianControl/teo/rightLeg/stateExt:i active at tcp://2.2.2.210:10036
yarp: Sending output from /BasicTwoLimbCartesianControl/teo/rightLeg/rpc:o to /teo/rightLeg/rpc:i using tcp
yarp: Sending output from /BasicTwoLimbCartesianControl/teo/rightLeg/command:o to /teo/rightLeg/command:i using udp
yarp: Receiving input from /teo/rightLeg/state:o to /BasicTwoLimbCartesianControl/teo/rightLeg/state:i using udp
yarp: Receiving input from /teo/rightLeg/stateExt:o to /BasicTwoLimbCartesianControl/teo/rightLeg/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[info] DeviceDriverImpl.cpp:92 configureLimbA(): numRobotJoints: 6.
[info] DeviceDriverImpl.cpp:95 configureLimbA(): numSolverLinks: 8.
[warning] DeviceDriverImpl.cpp:99 configureLimbA(): numRobotJoints(6) != numSolverLinks(8) !!!
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 0 limits: [-90.000000,90.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 1 limits: [-20.000000,20.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 2 limits: [-30.000000,45.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 3 limits: [-5.000000,80.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 4 limits: [-25.000000,25.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 5 limits: [-25.000000,25.000000]
[info] DeviceDriverImpl.cpp:134 configureLimbB(): solver: KdlSolver [KdlSolver]
[info] DeviceDriverImpl.cpp:138 configureLimbB(): angleRepr: axisAngle [axisAngle]
[info] DeviceDriverImpl.cpp:142 configureLimbB(): kinematics: leftLegKinematics.ini [leftLegKinematics.ini]
[... cut ...]
Warning! Adding waypoint [102] with default velocities. dt = 0.2
dtp: 0.2 , dtr: 0
Warning! Adding waypoint [102] with default velocities. dt = 0.2
[success] ICartesianControlImpl.cpp:56 step(): Waiting
.New trajectory segment : 0->1, Segment index: 0
> rfXdot: 0,0,0.1,> rotdot: 0,0,0,0,
New trajectory segment : 0->1, Segment index: 0
> lfXdot: 0,0,0.1,> rotdot: 0,0,0,0,
[STEP] [0.025232] x xdot 0.000000 0.000000 0.100000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.100000 0.000000 0.000000 0.000000 0.000000 
Violación de segmento
```

munozyanez, 2017-04-18:
Running on debian 9.

munozyanez, 2017-04-18:
Also happens on ubuntu 14.04.

jgvictores, 2017-04-18:
Could replicate with 2e5828cadeb6226e87f310b2be4f4d592d79eda9 on an Ubuntu 16.04.2. However, moving to father commit 70b263018319d24ed5baa7d4a0ab7f21302556bc, does not segfault.

munozyanez, 2017-04-19:
70b263018319d24ed5baa7d4a0ab7f21302556bc Not working anyway.

jgvictores, 2017-04-19:
On my pc it did. Could you paste the error here?

munozyanez, 2017-04-19:
It did not work yesterday at lab. I am trying code in teoSim. I will report asap.

munozyanez, 2017-04-19:
Nope, not working in that revision neither. Looking at the revision changes, I do not see how it could be related...
The error:
$ yarpdev --device BasicTwoLimbCartesianControl
[info] DeviceDriverImpl.cpp:14 open(): Subdevice BasicTwoLimbCartesianControl
[debug] DeviceDriverImpl.cpp:9 open(): Config: (single_threaded 1) (device BasicTwoLimbCartesianControl) (subdevice BasicTwoLimbCartesianControl).
[info] DeviceDriverImpl.cpp:30 configureLimbA(): solver: KdlSolver [KdlSolver]
[info] DeviceDriverImpl.cpp:34 configureLimbA(): angleRepr: axisAngle [axisAngle]
[info] DeviceDriverImpl.cpp:38 configureLimbA(): kinematics: rightLegKinematics.ini [rightLegKinematics.ini]
[debug] DeviceDriverImpl.cpp:10 open(): config: (angleRepr axisAngle) (kinematics rightLegKinematics.ini) (device KdlSolver).
[info] DeviceDriverImpl.cpp:14 open(): kinematics: rightLegKinematics.ini [none.ini]
[debug] DeviceDriverImpl.cpp:24 open(): fullConfig: (angleRepr axisAngle) (kinematics rightLegKinematics.ini) (H0 (0 1 0 0 1 0 0 -0.146 0 0 -1 -0.092 0 0 0 1)) (numLinks 6) (link_4 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (min -18.0) (max 20.0)) (link_3 (offset 0.0) (D 0.0175) (A 0.3) (alpha 0.0) (min -5.0) (max 80.0)) (link_5 (offset 0.0) (D 0.0) (A 0.123005) (alpha 0.0) (min -25.0) (max 25.0)) (device KdlSolver) (link_2 (offset 0.0) (D 0.0) (A 0.33) (alpha 0.0) (min -30.0) (max 45.0)) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (min -90.0) (max 90.0)) (link_1 (offset 90.0) (D 0.0) (A 0.0) (alpha 90.0) (min -20.0) (max 20.0)).
[info] DeviceDriverImpl.cpp:28 open(): numLinks: 6 [1]
[info] DeviceDriverImpl.cpp:32 open(): angleRepr: axisAngle [axisAngle]
[info] DeviceDriverImpl.cpp:59 open(): gravity: 0.0 0.0 -9.81 [0.0 0.0 -9.81]
[info] DeviceDriverImpl.cpp:75 open(): H0:
 0.000000	 1.000000	 0.000000	 0.000000
 1.000000	 0.000000	 0.000000	-0.146000
 0.000000	 0.000000	-1.000000	-0.092000
 0.000000	 0.000000	 0.000000	 1.000000
[ 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000]
[info] DeviceDriverImpl.cpp:107 open(): Added: link_0 (offset 0.000000) (D 0.000000) (A 0.000000) (alpha 90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_1 (offset 90.000000) (D 0.000000) (A 0.000000) (alpha 90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_2 (offset 0.000000) (D 0.000000) (A 0.330000) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_3 (offset 0.000000) (D 0.017500) (A 0.300000) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_4 (offset 0.000000) (D 0.000000) (A 0.000000) (alpha -90.000000)
[info] DeviceDriverImpl.cpp:107 open(): Added: link_5 (offset 0.000000) (D 0.000000) (A 0.123005) (alpha 0.000000)
[info] DeviceDriverImpl.cpp:172 open(): HN:
 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000
[ 1.000000	 0.000000	 0.000000	 0.000000
 0.000000	 1.000000	 0.000000	 0.000000
 0.000000	 0.000000	 1.000000	 0.000000
 0.000000	 0.000000	 0.000000	 1.000000]
[info] DeviceDriverImpl.cpp:175 open(): Chain number of segments (post- H0 and HN): 8
[info] DeviceDriverImpl.cpp:176 open(): Chain number of joints (post- H0 and HN): 6
[INFO]created device <KdlSolver>. See C++ class teo::KdlSolver for documentation.
[info] DeviceDriverImpl.cpp:61 configureLimbA(): remote: /teoSim/rightLeg [/teoSim/rightLeg]
yarp: Port /BasicTwoLimbCartesianControl/teoSim/rightLeg/rpc:o active at tcp://192.168.10.2:10026
yarp: Port /BasicTwoLimbCartesianControl/teoSim/rightLeg/command:o active at tcp://192.168.10.2:10027
yarp: Port /BasicTwoLimbCartesianControl/teoSim/rightLeg/state:i active at tcp://192.168.10.2:10028
yarp: Port /BasicTwoLimbCartesianControl/teoSim/rightLeg/stateExt:i active at tcp://192.168.10.2:10029
yarp: Sending output from /BasicTwoLimbCartesianControl/teoSim/rightLeg/rpc:o to /teoSim/rightLeg/rpc:i using tcp
yarp: Sending output from /BasicTwoLimbCartesianControl/teoSim/rightLeg/command:o to /teoSim/rightLeg/command:i using udp
yarp: Receiving input from /teoSim/rightLeg/state:o to /BasicTwoLimbCartesianControl/teoSim/rightLeg/state:i using udp
yarp: Receiving input from /teoSim/rightLeg/stateExt:o to /BasicTwoLimbCartesianControl/teoSim/rightLeg/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[info] DeviceDriverImpl.cpp:92 configureLimbA(): numRobotJoints: 6.
[info] DeviceDriverImpl.cpp:95 configureLimbA(): numSolverLinks: 8.
[warning] DeviceDriverImpl.cpp:99 configureLimbA(): numRobotJoints(6) != numSolverLinks(8) !!!
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 0 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 1 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 2 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 3 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 4 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:109 configureLimbA(): Joint 5 limits: [0.000000,0.000000]
[info] DeviceDriverImpl.cpp:134 configureLimbB(): solver: KdlSolver [KdlSolver]
[info] DeviceDriverImpl.cpp:138 configureLimbB(): angleRepr: axisAngle [axisAngle]
[... cut ...]
Warning! Adding waypoint [102] with default velocities. dt = 0.2
dtp: 0.2 , dtr: 0
Warning! Adding waypoint [102] with default velocities. dt = 0.2
[success] ICartesianControlImpl.cpp:56 step(): Waiting
.New trajectory segment : 0->1, Segment index: 0
> rfXdot: 0,0,0.1,> rotdot: 0,0,0,0,
New trajectory segment : 0->1, Segment index: 0
> lfXdot: 0,0,0.1,> rotdot: 0,0,0,0,
[STEP] [0.003664] x xdot 0.000000 0.000000 0.100000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.100000 0.000000 0.000000 0.000000 0.000000 
Violación de segmento

jgvictores, 2017-04-19:
My intuition was that there might be some loop somewhere that has a return value as a break condition, and something is looping more than it should since `bool` logic is somewhat inverted with respect to the new `long` representation (extract from [here](https://github.com/roboticslab-uc3m/gait/compare/03fe5bdaab682b1d0efa9fce9fad1417aa716406...08fd8221f40350028a0a8ba82ff91bc66e0c1b8b#diff-02a5c822dac3211b227f53f928b2ef67R289):
```cpp
-bool Pose::GetPosition(double &pose_x, double &pose_y, double &pose_z)
+long Pose::GetPosition(double &pose_x, double &pose_y, double &pose_z)
 {
     pose_x=x;
     pose_y=y;
     pose_z=z;
-    return true;
+    return 0;
 }
 ```
Will check on your PC just to make sure we're speaking about the same commits and have correct installations.

PeterBowman, 2017-04-19:
Did a pass through the debugger at latest `develop`. I'm reading an empty `desiredX` vector at [RateThreadImpl.cpp#35](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2e5828c/libraries/TeoYarp/BasicTwoLimbCartesianControl/RateThreadImpl.cpp#L35), which in turn causes initialization issues at [line 48](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2e5828c/libraries/TeoYarp/BasicTwoLimbCartesianControl/RateThreadImpl.cpp#L48) (`desiredX.begin()+7` exceeds `desiredX`'s range).

PeterBowman, 2017-04-19:
BTW `libGait.so` gets installed on `/usr/local/bin`, it should land on `lib/yarp` instead.

munozyanez, 2017-04-19:
That was the point!
The change in Juan's words "since bool logic is somewhat inverted" made desireX return empty vectors.
Currently fixed, I am going to commit and test step tomorrow morning.
Thanks a lot you two.

munozyanez, 2017-04-19:
Ok, I will make it land at the correct folder.

PeterBowman, 2017-04-19:
> BTW libGait.so gets installed on /usr/local/bin, it should land on lib/yarp instead.

@munozyanez My bad, I thought Gait was implemented as a YARP plugin, hence the proposed path. Anyway, it strikes me that `libGait.so`'s destination path was set to `bin` like any executable ([link](https://github.com/roboticslab-uc3m/gait/blob/56f6174/src/CMakeLists.txt#L37)). You may as well leave it as it is, although `yarpdev` might complain that it's unable to find the shared library (as it happened to me today).

munozyanez, 2017-04-19:
I is definitely a code recycle error. I would say the place for it is usr/lib, and so I will do. If there is another preference just tell.

munozyanez, 2017-04-19:
Now it installs under /usr/local/lib/ , but If there is another preference just tell.
Closing issue.

closed 2017-04-19 by munozyanez