roboticslab-uc3m/kinematics-dynamics #95: Implement tool change capability

PeterBowman, 2017-04-19:
ASIBOT's RPC server offers an additional "tool" command that could be reused here: [xRpcResponder.cpp#L145](https://github.com/roboticslab-uc3m/asibot-main/blob/c2644a2/src/libraries/CartesianServerLib/xRpcResponder.cpp#L145).

PeterBowman, 2017-05-02:
Old `CartesianBot` (from asibot-main) included support for one tool at the robot's tip. Relevant point in the commit history with previous implementation so that we may come back to this in the future: [libraries/TeoYarp/AsibotSolver@6d24a22](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/6d24a22/libraries/TeoYarp/AsibotSolver).

jgvictores, 2017-05-02:
Here are some proposals:
1. `setHN(const std::vector<double>& H)`, as a function that replaces the existing `HN`.
1. `setTool(const std::vector<double>& H)`, as a function that appends to the existing `HN`.
1. `setTool(const std::vector<double>& x)`, as a function that appends to the existing `HN`.

I think I'd go with the last one, and document that `x` is implementation-dependent.

jgvictores, 2017-05-02:
PS: Would affect [/libraries/TeoYarp/ICartesianSolver.h](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/23714689c203a9f3ccfea8dcae977fe1fdff2dbd/libraries/TeoYarp/ICartesianSolver.h) and classes which inherit from it.

jgvictores, 2017-05-03:
New proposal:
- `appendLink(const std::vector<double>& x)`, as a function that appends to the existing `HN`.

jgvictores, 2017-05-06:
Methods implemented at 0dc58e6f8561f0936c8b5f98b1ffffcfdfea7751.

Ideally, would develop some tests, but for now will be happy when using new methods at `transCoordsUsingJoints`.

PeterBowman, 2017-05-10:
What's left:

* add the tool change functionality in the `ICartesianControl` interface (vocab + method)
* handle new RPC command in `CartesianControlServer`
* implement new methods in the `BasicCartesianControl` and `CartesianControlClient` trees

PeterBowman, 2017-08-07:
Takeover! (nah, please contribute, the new branch is `fix-95-tool-change`). I've named the new `ICartesianControl` method as `tool` to keep it consistent with the current convention `VOCAB_CC_XXX` = `ICartesianControl::XXX`. It wouldn't hurt to name it `changeTool`, but neither would `relativeMove`, `velocityMove` and so on. The only black sheep here is the `stopControl` method...

PeterBowman, 2017-08-23:
> What's left: ...

Done at 563bb74.

closed 2017-08-23 by PeterBowman