roboticslab-uc3m/kinematics-dynamics #96: Extract and import ASIBOT's analytical cartesian solver from asibot-main repo

PeterBowman, 2017-04-20:
Currently implemented as a `yarp::dev::ICartesianControl` ([path](https://github.com/roboticslab-uc3m/asibot-main/tree/c2644a2/src/libraries/RlPlugins/cartesianbot)), should be stripped from any control loops and repurposed as a `teo::ICartesianSolver`.

PeterBowman, 2017-05-01:
WIP at the `asibot-cartesian-solver` branch.

PeterBowman, 2017-05-01:
Just wanted to note that there are some conversion operations left in [CartesianServerLib/xCallbackPort.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/310d5dc/src/libraries/CartesianServerLib/xCallbackPort.cpp) at which I'd like to take a look.

PeterBowman, 2017-05-02:
CartesianBot implements tool tip calculations, but we lack a proper interface. Perhaps blocked until #95 is resolved?

jgvictores, 2017-05-06:
> CartesianBot implements tool tip calculations, but we lack a proper interface. Perhaps blocked until #95 is resolved?

Yes, I agree.

PeterBowman, 2017-05-17:
The [`TrajGen.h`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/31c635d/libraries/TeoYarp/AsibotSolver/TrajGen.hpp) library I've imported along with ASIBOT's solver could be easily replaced by [KDL::VelocityProfile_Spline](http://docs.ros.org/hydro/api/orocos_kdl/html/classKDL_1_1VelocityProfile__Spline.html).

PeterBowman, 2017-08-07:
Good enough at 974de24, follow-ups include:

* #108 (unimplemented `ICartesianControl` methods)
* #109 (investigate singularities)
* #110 (enhance unit tests)
* #111 (don't restrict rotations to 180º)

I've left the [TrajGen.hpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/974de241243c09612e8ab069317c9b6b76da6a2e/libraries/TeoYarp/AsibotSolver/TrajGen.hpp) file mostly as is, but it does not intervene at all in the compilation process. It might be re-used as an alternative to current trajectory generators at `BasicCartesianControl`, which in turn would require more work than it was initially planned at here.

closed 2017-08-07 by PeterBowman