roboticslab-uc3m/kinematics-dynamics #97: [transCoordsUsingJoints] Compilation error

rsantos88, 2017-05-05:
I am trying to compile **kinematics-dynamic**  on **transCoordsUsingJoints** branch (I know that is developing at this moment...). It shows this messages:

```bash
Scanning dependencies of target BasicCartesianControl
[ 47%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/yarp_plugin_BasicCartesianControl.cpp.o
[ 50%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/DeviceDriverImpl.cpp.o
[ 52%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/ICartesianControlImpl.cpp.o
[ 54%] Building CXX object libraries/TeoYarp/BasicCartesianControl/CMakeFiles/BasicCartesianControl.dir/RateThreadImpl.cpp.o
Linking CXX shared module ../../../lib/yarp/BasicCartesianControl.so
[ 54%] Built target BasicCartesianControl
[ 60%] Built target CartesianControlServer
[ 67%] Built target CartesianControlClient
Scanning dependencies of target BasicTwoLimbCartesianControl
[ 69%] Building CXX object libraries/TeoYarp/BasicTwoLimbCartesianControl/CMakeFiles/BasicTwoLimbCartesianControl.dir/yarp_plugin_BasicTwoLimbCartesianControl.cpp.o
[ 71%] Building CXX object libraries/TeoYarp/BasicTwoLimbCartesianControl/CMakeFiles/BasicTwoLimbCartesianControl.dir/DeviceDriverImpl.cpp.o
[ 73%] Building CXX object libraries/TeoYarp/BasicTwoLimbCartesianControl/CMakeFiles/BasicTwoLimbCartesianControl.dir/ICartesianControlImpl.cpp.o
[ 76%] Building CXX object libraries/TeoYarp/BasicTwoLimbCartesianControl/CMakeFiles/BasicTwoLimbCartesianControl.dir/RateThreadImpl.cpp.o
Linking CXX shared module ../../../lib/yarp/BasicTwoLimbCartesianControl.so
[ 76%] Built target BasicTwoLimbCartesianControl
[ 82%] Built target TwoLimbCartesianControlServer
[ 84%] Building CXX object programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/PremultPorts.cpp.o
In file included from /home/teo/repos/kinematics-dynamics/programs/transCoordsUsingJoints/PremultPorts.hpp:9:0,
                 from /home/teo/repos/kinematics-dynamics/programs/transCoordsUsingJoints/PremultPorts.cpp:1:
/usr/local/include/yarp/dev/IEncoders.h:24:31: error: variable ‘YARP_dev_API yarp::dev::IEncodersRaw’ has initializer but incomplete type
 class YARP_dev_API yarp::dev::IEncodersRaw
                               ^
/usr/local/include/yarp/dev/IEncoders.h:26:1: error: expected primary-expression before ‘public’
 public:
 ^
/usr/local/include/yarp/dev/IEncoders.h:26:1: error: expected ‘}’ before ‘public’
/usr/local/include/yarp/dev/IEncoders.h:26:1: error: expected ‘,’ or ‘;’ before ‘public’
/usr/local/include/yarp/dev/IEncoders.h:37:33: error: ‘virtual’ outside class declaration
     virtual bool getAxes(int *ax) = 0;
                                 ^
/usr/local/include/yarp/dev/IEncoders.h:37:37: error: function ‘bool getAxes(int*)’ is initialized like a variable
     virtual bool getAxes(int *ax) = 0;
                                     ^
/usr/local/include/yarp/dev/IEncoders.h:44:39: error: ‘virtual’ outside class declaration
     virtual bool resetEncoderRaw(int j)=0;
                                       ^
/usr/local/include/yarp/dev/IEncoders.h:44:41: error: function ‘bool resetEncoderRaw(int)’ is initialized like a variable
     virtual bool resetEncoderRaw(int j)=0;
                                         ^
/usr/local/include/yarp/dev/IEncoders.h:50:35: error: ‘virtual’ outside class declaration
     virtual bool resetEncodersRaw()=0;
                                   ^
/usr/local/include/yarp/dev/IEncoders.h:50:37: error: function ‘bool resetEncodersRaw()’ is initialized like a variable
     virtual bool resetEncodersRaw()=0;
                                     ^
/usr/local/include/yarp/dev/IEncoders.h:58:49: error: ‘virtual’ outside class declaration
     virtual bool setEncoderRaw(int j, double val)=0;
                                                 ^
/usr/local/include/yarp/dev/IEncoders.h:58:51: error: function ‘bool setEncoderRaw(int, double)’ is initialized like a variable
     virtual bool setEncoderRaw(int j, double val)=0;
                                                   ^
/usr/local/include/yarp/dev/IEncoders.h:65:51: error: ‘virtual’ outside class declaration
     virtual bool setEncodersRaw(const double *vals)=0;
                                                   ^
/usr/local/include/yarp/dev/IEncoders.h:65:53: error: function ‘bool setEncodersRaw(const double*)’ is initialized like a variable
     virtual bool setEncodersRaw(const double *vals)=0;
                                                     ^
/usr/local/include/yarp/dev/IEncoders.h:73:48: error: ‘virtual’ outside class declaration
     virtual bool getEncoderRaw(int j, double *v)=0;
                                                ^
/usr/local/include/yarp/dev/IEncoders.h:73:50: error: function ‘bool getEncoderRaw(int, double*)’ is initialized like a variable
     virtual bool getEncoderRaw(int j, double *v)=0;
                                                  ^
/usr/local/include/yarp/dev/IEncoders.h:80:45: error: ‘virtual’ outside class declaration
     virtual bool getEncodersRaw(double *encs)=0;
                                             ^
/usr/local/include/yarp/dev/IEncoders.h:80:47: error: function ‘bool getEncodersRaw(double*)’ is initialized like a variable
     virtual bool getEncodersRaw(double *encs)=0;
                                               ^
/usr/local/include/yarp/dev/IEncoders.h:88:54: error: ‘virtual’ outside class declaration
     virtual bool getEncoderSpeedRaw(int j, double *sp)=0;
                                                      ^
/usr/local/include/yarp/dev/IEncoders.h:88:56: error: function ‘bool getEncoderSpeedRaw(int, double*)’ is initialized like a variable
     virtual bool getEncoderSpeedRaw(int j, double *sp)=0;
                                                        ^
/usr/local/include/yarp/dev/IEncoders.h:95:50: error: ‘virtual’ outside class declaration
     virtual bool getEncoderSpeedsRaw(double *spds)=0;
                                                  ^
/usr/local/include/yarp/dev/IEncoders.h:95:52: error: function ‘bool getEncoderSpeedsRaw(double*)’ is initialized like a variable
     virtual bool getEncoderSpeedsRaw(double *spds)=0;
                                                    ^
/usr/local/include/yarp/dev/IEncoders.h:102:63: error: ‘virtual’ outside class declaration
     virtual bool getEncoderAccelerationRaw(int j, double *spds)=0;
                                                               ^
/usr/local/include/yarp/dev/IEncoders.h:102:65: error: function ‘bool getEncoderAccelerationRaw(int, double*)’ is initialized like a variable
     virtual bool getEncoderAccelerationRaw(int j, double *spds)=0;
                                                                 ^
/usr/local/include/yarp/dev/IEncoders.h:109:57: error: ‘virtual’ outside class declaration
     virtual bool getEncoderAccelerationsRaw(double *accs)=0;
                                                         ^
/usr/local/include/yarp/dev/IEncoders.h:109:59: error: function ‘bool getEncoderAccelerationsRaw(double*)’ is initialized like a variable
     virtual bool getEncoderAccelerationsRaw(double *accs)=0;
                                                           ^
/usr/local/include/yarp/dev/IEncoders.h:110:1: error: expected declaration before ‘}’ token
 };
 ^
programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/build.make:54: recipe for target 'programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/PremultPorts.cpp.o' failed
make[2]: *** [programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/PremultPorts.cpp.o] Error 1
CMakeFiles/Makefile2:837: recipe for target 'programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/all' failed
make[1]: *** [programs/transCoordsUsingJoints/CMakeFiles/transCoordsUsingJoints.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

```
I think it could be helpful for future problems...

PeterBowman, 2017-05-05:
```
/usr/local/include/yarp/dev/IEncoders.h:24:31: error: variable ‘YARP_dev_API yarp::dev::IEncodersRaw’ has initializer but incomplete type
 class YARP_dev_API yarp::dev::IEncodersRaw
```

It looks like it's treating `YARP_dev_API` as the type of `yarp::dev::IEncodersRaw` (¿?). I could reproduce this error and "solve" it by adding `#include <yarp/dev/api.h>` before  `#include <yarp/dev/IEncoders.h>` in `PremultPorts.hpp`.

PeterBowman, 2017-05-05:
This is an upstream bug solved in current `master` branch at robotology/yarp@5ea5c2a. I'd just put said include and remove it once YARP 2.3.70 is released (although leaving it there forever won't hurt anyway).

jgvictores, 2017-05-06:
Applied temporary fix at 9091d6018b23008765648bc94129023dd5a59773. Ok to remove it once next YARP is released (`2.3.70` I guess).

rsantos88, 2017-05-08:
Thanks @PeterBowman @jgvictores !!

PeterBowman, 2017-05-18:
We'd like to keep track of the fix and eventually remove it at certain point (it will be no longer needed at YARP 2.3.68.1/2.3.70). @jgvictores I've perpetrated 791b0a3. What do you think?

jgvictores, 2017-05-18:
Mmmm looks like the conditional is on:
```
#include <yarp/dev/IEncoders.h>
```

Rather than the *hack* which is actually including:
```
#include <yarp/dev/api.h>
```

Right? 😅

PeterBowman, 2017-05-18:
Oops. Fixed, @jgvictores: 8cc04cd.

jgvictores, 2017-05-18:
1. Just [took an extra look](https://github.com/search?q=org%3Aroboticslab-uc3m+%22yarp%2Fdev%2FIEncoders.h%22&type=Code&utf8=%E2%9C%93) because I thought our code would use this a lot, but I see that in [the vast majority](https://github.com/search?utf8=%E2%9C%93&q=org%3Aroboticslab-uc3m+%22yarp%2Fdev%2Fall.h%22&type=Code) we just `#include <yarp/dev/all.h>`. I'm a fan of including only what is used, so for me it's okay to keep the `yarp/dev/IEncoders.h` only if it's the only header required (rather than just migrating to `yarp/dev/all.h`).

1. Would `__YARP_VERSION_N` break with a potential YARP `2.3.100` or similar?

PeterBowman, 2017-05-18:
Ad 2: no, `2.3.100` would translate into `__YARP_VERSION_N == 231000` thanks to the arithmetics performed at lines [L12-L14](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/8cc04cd/programs/transCoordsUsingJoints/PremultPorts.hpp#L12-L14).

jgvictores, 2017-05-18:
Re: Ad 2: Haha, let's try a different example: `2.10.0` and `3.0.0`.

PeterBowman, 2017-05-18:
Yeah, I know that, but this hack is not meant to last forever, certainly less than it would take to reach YARP x.10.x. Taking another look at the code, I should have shifted all version parts but the tweak number by one digit to the left. That is, `2.3.67.10` (I don't know if such version exists) is computed as `230680` with this algorithm, thus overlapping `2.3.68`.

jgvictores, 2017-05-18:
> this hack is not meant to last forever, certainly less than it would take to reach YARP x.10.x

👍

jgvictores, 2017-05-25:
Merged fix into `develop` at 7b5d67bb84130b622982358bd39fad1d9b322f94.

closed 2017-05-25 by jgvictores

PeterBowman, 2018-01-17:
I'm going to remove the hack in 18e5f5b. I think it's reasonable enough to assume that everyone using the YARP 2.3.68 release have already caught up with the tweak version 2.3.68.1 (these are often non-breaking, security updates, so don't hesitate and upgrade *now*).

For the record, the patch looked like this:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/55fe7988f56eda3de7346fabb26b0185abfb3d86/programs/transCoordsUsingJoints/PremultPorts.hpp#L12-L22

jgvictores, 2018-01-17:
Is it required for https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-yarp-2368-debian-6010 ?

PeterBowman, 2018-01-17:
Yes... Can we assume that nobody will try to launch `transCoordUsingJoints` on a Debian 6.0 with that specific YARP configuration?

jgvictores, 2018-01-17:
Mmmmm IMHO, that's not a safe assumption until https://github.com/roboticslab-uc3m/teo-main/issues/26 is solved...

PeterBowman, 2018-01-17:
OK, I didn't know you were using TEO's on-board CPUs to perform kinematics. I'm not planning to merge this anytime soon, anyway.

jgvictores, 2018-01-17:
Yes, using TEO's on-board CPUs to perform kinematics is definitely a use case.
As good news, we've unblocked https://github.com/roboticslab-uc3m/teo-main/issues/26 today, so hopefully we'll have some progress on that side soon.

PeterBowman, 2018-01-25:
@jgvictores given https://github.com/roboticslab-uc3m/installation-guides/issues/24#issuecomment-360429885, can we merge this now to `develop`?

PeterBowman, 2018-01-26:
OK, hack removed at #137.

commit 9091d601 (2017-05-06): fix #97 as commented by @PeterBowman

fix #97 as commented by @PeterBowman at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/97#issuecomment-299449139, until fixed upstream