roboticslab-uc3m/kinematics-dynamics #99: movj() error: invKin failed.

rsantos88, 2017-05-16:
I'm trying to move the right arm of TeoSim with BasicCartesianControl using `movj`, but every time it faults showing this error:
```bash
[error] ICartesianSolverImpl.cpp:110 invKin(): -5: The maximum number of iterations is exceeded
[error] ICartesianControlImpl.cpp:54 movj(): invKin failed
```
There is no problem when I use `movl` if I want to do the same movement.

jgvictores, 2017-05-16:
This is related to [kinematic reachability](http://openrave.org/docs/latest_stable/openravepy/databases.kinematicreachability/), so I've revived and assigned https://github.com/roboticslab-uc3m/teo-openrave-models/issues/3.

PeterBowman, 2017-07-08:
Same was happening with ASIBOT every time I issued a `inv` RPC command. I managed to avoid it by relaxing the solver's precision from `1E-9` to `1E-6` (speaking in meters, I wonder why it was set to such a low value, even if considering dcce8fd). BTW compare with the defaults in [KDL::ChainIkSolverPos_NR_JL](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1ChainIkSolverPos__NR__JL.html#a5f6d22a47659005408dce6ae0937f7e1). Therefore, this issue might be related to #100.

PeterBowman, 2018-02-12:
Possibly related: #143.

PeterBowman, 2018-02-22:
Alternative, non-numerical IK solutions proposed at #119 could help here.

jgvictores, 2018-02-23:
Will close issue briefly if no particular examples are given. Possibly due to angleRepr, as in #143.

closed 2022-05-22 by PeterBowman