roboticslab-uc3m/kinematics-dynamics #102: BasicCartesianControl stream FK

jgvictores, 2017-07-24:
BasicCartesianControl stream FK

jgvictores, 2017-07-27:
@PeterBowman Before merging 63f105a5758197eda0db852e3af0121b65639b4a I'd like your opinion on:
1. Maybe a `--fkPeriod` parameter to establish the frequency? Different parameter name proposal (e.g. `--msStateOut`?
2. Most importantly, should the port always be open? Maybe we could avoid opening the port if `--fkPeriod 0`? Or would that be bad because we may want the port as some kind of dependency (and all the `if` clauses would make it a bit messy)?

PeterBowman, 2017-07-27:
@jgvictores I'm worried about race conditions, perhaps we should guard the `iCartesianControl` handle behind a mutex/semaphore. Have you tested this code while issuing simultaneous RPC commands? Maybe I'm misunderstanding the concept of locks/races and these considerations pertain to the implementors of `ICartesianControl`, that is `BasicCartesianControl` for such instance, and the actual thread guards should be considered in those subdevices instead. Whether the port should be always left open might find an answer through previous questions.

Regarding the name proposal: unless we expect to publish more data types (i.e. not just FK), I'm OK with `--fkPeriod`.

jgvictores, 2017-08-29:
@PeterBowman You're right about your concerns... maybe some semaphores would be good.

PeterBowman, 2017-08-29:
Also, I wonder if `CartesianControlClient` should (try to) open a receiver port and buffer incoming data, just like YARP's `remote_controlboard` does (`state:i` and `stateExt:i`).

BTW I'd rebase `issue201fkStream` on current `develop` before resuming work or merging.

jgvictores, 2017-08-29:
> Also, I wonder if `CartesianControlClient` should (try to) open a receiver port and buffer incoming data, just like YARP's `remote_controlboard` does (`state:i` and `stateExt:i`).

Makes sense.

> BTW I'd rebase `issue201fkStream` on current `develop` before resuming work or merging.

:+1:

jgvictores, 2017-12-03:
As commented with @PeterBowman latest changes (even folder name changes) are at `develop`. Should consider rebase or even posible re-impementation before merging.

PeterBowman, 2017-12-07:
@jgvictores I rebased your changes on `develop` (30c955f), now I'm working on `fix-102-fkstream-rebased`. You may want to delete `issue201fkStream` if not needed anymore.

Support for `--fkPeriod` was added in 1300cdb.

> Most importantly, should the port always be open?

By now, I'll stick with *yes* as the answer unless usage proves that we may incur in concurrency issues. A short glimpse at `yarp::dev::ControlBoardWrapper`'s code revealed that they are actually **not** using semaphores to guard [all those interface handles](https://github.com/robotology/yarp/blob/c83719c0f0e047c527e16a640651e73a8689237a/src/libYARP_dev/src/devices/ControlBoardWrapper/SubDevice.h#L82-L104). Well, I found a [lonely mutex](https://github.com/robotology/yarp/blob/c83719c0f0e047c527e16a640651e73a8689237a/src/libYARP_dev/src/devices/ControlBoardWrapper/RPCMessagesParser.h#L100), but it's not used anywhere.

I expected to see something like this used massively ([ref](https://github.com/robotology/yarp/blob/c83719c0f0e047c527e16a640651e73a8689237a/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.h#L316)):

```c++
yarp::os::Semaphore rpcDataMutex; // mutex to avoid concurrency between more
                                  // clients using rppc port
```

However, YARP folks call it exclusively whenever a `memset` is involved for multiple-joint data shuffling.

PeterBowman, 2017-12-07:
> Maybe we could avoid opening the port if --fkPeriod 0? Or would that be bad because we may want the port as some kind of dependency (and all the if clauses would make it a bit messy)?

> By now, I'll stick with yes as the answer unless usage proves that we may incur in concurrency issues.

Actually, it's quite simple to achieve and could prove useful if latency is an issue (keep in mind that there'll be a loop computing FK every few milliseconds). The client device may easily fall back to RPC if there is no `/state:o` port.

PR #129 (2018-01-02): Stream forward kinematics data and fetch it in the client device
Resolves #102. Marking as work in progress: needs proper testing (preferably with high-frequency, streaming commands) so that we make sure there are no race conditions caused by concurrent access.

A few design decisions were inspired by YARP's remote_controlboard & controlboardwrapper2 devices. It's worth noting that this pair sends/receives timestamp envelopes along with actual data (e.g. encoder values, control mode, etc.). In the future, we could take advantage of this technique to provide a more accurate way of describing ca [... cut]
closed 2018-01-02 by jgvictores