roboticslab-uc3m/kinematics-dynamics #103: Libraries being installed to /usr/local/lib/x86_64-linux-gnu

jgvictores, 2017-07-24:
Ubuntu 14.04,
```bash
Install the project...
-- Install configuration: "Release"
/usr/local/lib/x86_64-linux-gnu/libKdlVectorConverterLib.so
/usr/local/lib/x86_64-linux-gnu/libTrajectoryLib.so
```

Temporal hack using `export LD_LIBRARY_PATH=/usr/local/lib/x86_64-linux-gnu`...
**/me** confused... @PeterBowman Do you think this behaviour expected/good?

PeterBowman, 2017-07-24:
@jgvictores please check the `CMAKE_INSTALL_PREFIX` variable in your CMake cache. This is definitely not happening on CI Trusty builds: https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/jobs/256819339#L2544.

PeterBowman, 2017-07-24:
Oh, well, the `x86_64...` part actually happens after the common prefix. Could you look at `CMAKE_INSTALL_LIBDIR` instead? (also: `CMAKE_INSTALL_BINDIR`, `CMAKE_INSTALL_INCLUDEDIR`, etc.).

PeterBowman, 2017-07-24:
I dug a bit further:

* https://github.com/Kitware/CMake/blob/v2.8.12.2/Modules/GNUInstallDirs.cmake#L71-L100
* https://cmake.org/cmake/help/v3.0/variable/CMAKE_LIBRARY_ARCHITECTURE.html
* https://cmake.org/cmake/help/v3.0/variable/CMAKE_LANG_LIBRARY_ARCHITECTURE.html

Are you using a Debian OS?

jgvictores, 2017-07-25:
Nope, just Ubuntu 14.04. Only hack was commenting out `cmake_policy(SET CMP0046 NEW)` from the root CMake.

PeterBowman, 2017-07-25:
Sorry, I overlooked that first sentence. What do you mean by *only hack*? Did you solve it by commenting out that policy line (apart from tweaking `LD_LIBRARY_PATH`)? We may still need to take a look at the `CMAKE_INSTALL_*` variables.

CMP0046 should have nothing to do with installation paths: https://cmake.org/cmake/help/latest/policy/CMP0046.html.

jgvictores, 2017-07-27:
>  Did you solve it by commenting out that policy line (apart from tweaking `LD_LIBRARY_PATH`)?

No, the policy line was commented out just to be able to run the initial `cmake` (I recall we were with `cmake 3.5`).

Right now on `teo-main` I created a symbolic link to avoid tweaking `LD_LIBRARY_PATH`, but this is obviously still a hack.

PeterBowman, 2017-07-27:
The `CPM0046` thing could deserve a separate issue, but please `git clone` the project anew and configure from scratch prior to writing a report. BTW you'll see a warning upon `cmake ..` if using CMake 3.0 or later (it strikes me that you are running CMake 3.5 on Trusty).

Please do the following:

1. in case `color-debug` is cloned on your system, pull the latest code and configure (no need to install it)
2. remove the contents of the build-tree (`/build` directory or whichever name you chose) at kinematics-dynamics and configure again
3. paste the output of the `cmake` command
4. upload here `CMakeCache.txt`

jgvictores, 2017-07-28:
> running CMake 3.5 on Trusty

Just before I forget: https://packages.ubuntu.com/trusty/cmake3

PeterBowman, 2017-07-28:
Wow, I missed that package. It's time to think on bumping those `cmake_minimum_required`...

PeterBowman, 2017-10-16:
Any update on this?

jgvictores, 2017-11-07:
Haven't been using Trusty for a while. I believe this was on [[_name_ommited_for_privacy_]]'s PC, which needs an upgrade anyway. Closing as inactive.

closed 2017-11-07 by jgvictores

PeterBowman, 2017-12-12:
> /me confused... @PeterBowman Do you think this behaviour expected/good?

@jgvictores see roboticslab-uc3m/QA#39.

jgvictores, 2017-12-12:
@PeterBowman Ok, thanks!

jgvictores, 2017-12-12:
PS: Removed the `inactivity` label.