roboticslab-uc3m/kinematics-dynamics #105: CartesianControlServer should handle streaming commands

PeterBowman, 2017-07-26:
This was implemented on [`CartesianServerLib`](https://github.com/roboticslab-uc3m/asibot-main/tree/c975cd5b6e049be9a59e1c11b807faefe51fee6f/src/libraries/CartesianServerLib) at asibot-main (see [xCallbackPort.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5b6e049be9a59e1c11b807faefe51fee6f/src/libraries/CartesianServerLib/xCallbackPort.cpp)), but never made its way to CCS. The `vmos` command - velocity control in cartesian space - seems the most interesting option to start working at.

PeterBowman, 2017-07-26:
Started with some refactoring at the `streaming-cartesian-server` branch.

jgvictores, 2017-07-27:
Great to see improvements in `CartesianControlServer`. In #102 we're working on another improvement, I'll comment there.

PeterBowman, 2017-07-27:
`vmos` is mostly implemented. I've created the `streamingSpnav` program (109ec2c at `streaming-spnav` branch) too give it a try - it works fine on simulator with ASIBOT (default) and TEO, we are about to test it on real ASIBOT (brace yourselves). You'll need a SpaceNavigator 3D mouse or similar (current implementation launches `SpaceNavigator` from yarp-devices, but creating your own `yarp::dev::AnalogSensor` device may be trivial).

@jgvictores the current WIP list for old [xCallbackPort](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5b6e049be9a59e1c11b807faefe51fee6f/src/libraries/CartesianServerLib/xCallbackPort.cpp) commands is: `fwd`, `bkwd`, `rot`, `pose` (and the first one, left without a name). Is there any other streaming command you are especially keen on having brought to kin-dyn?

jgvictores, 2017-07-28:
> @jgvictores the current WIP list for old xCallbackPort commands is: fwd, bkwd, rot, pose (and the first one, left without a name). Is there any other streaming command you are especially keen on having brought to kin-dyn?

I think the current WIP is great!

PeterBowman, 2017-09-11:
The current WIP list is:

```c++
// move forward (relative to end-effector)
void fwd(const std::vector<double> &rot, double step);

// move backwards (relative to end-effector)
void bkwd(const std::vector<double> &rot, double step);

// rotate in end-effector frame
void rot(const std::vector<double> &rot);

// pan in end-effector frame
void pan(const std::vector<double> &transl);

// instantaneous velocity steps (inertial frame)
void vmos(const std::vector<double> &xdot);

// instantaneous velocity steps (end-effector frame)
void eff(const std::vector<double> &xdotee);

// achieve pose in inertial frame
void pose(const std::vector<double> &x, double interval);
```

See also full documentation here:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2fd0b37f7e35de8a0a73e16ca0833fe3dda5ef4c/libraries/TeoYarp/ICartesianControl.h#L239-L339

As you can see, I added `pan` and `eff` to the initial list. The former acts just like `rot`, but only with linear translations (while `rot` changes orientation about the same point in space). The latter is exactly the same as `vmos`, but it moves in the end-effector frame.

Additionally, I've introduced the `step` parameter at `fwd` and `bkwd` to provide a means of controlling the step velocity at which the robot arm moves forward or backwards. Furthermore, `pose` is a bit different from the original command at asibot-main: it overrides the trajectory generator and applies inverse differential kinematics on the velocity obtained by numerical differentiation between current and target poses, using `interval` as the time step (see [BasicCartesianControl/ICartesianControlImpl.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2fd0b37f7e35de8a0a73e16ca0833fe3dda5ef4c/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp)).

The unnamed command has been set aside by now - I'd like to improve control with the `IPositionDirect` interface, although I'm a bit worried about computational power (the controller would perform inverse kinematics every 10 milliseconds).

For completeness, note that end-effector frame commands require a new solver method, `ICartesianSolver::diffInvKinEE`:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/2fd0b37f7e35de8a0a73e16ca0833fe3dda5ef4c/libraries/TeoYarp/ICartesianSolver.h#L130-L144

PeterBowman, 2017-10-06:
@jgvictores I'm a bit worried about the ammount of commands resulting from this, some of them apparently redundant. There are three main variants: RPC vs streaming commands, position vs velocity control (vs torque), and inertial vs end-effector frame. And there is more to come: I'd find quite useful a `movv` RPC command using end-effector coordinates for my `keyboardController` application, but I'm running out of ideas for choosing the best name (`move`? `movve`?).

Are you OK with adding new commands as they render convenient for our applications, or is it preferable to keep the lowest number of them and parameterize if possible?

I was thinking about a new interface (inheritable by `BasicCartesianControl` and the like) for managing controller configuration during runtime. In fact, it would be nice to have the ability to select a different gain, tweak the maximum joint velocity and trajectory time (along the lines of #46), change the trajectory generator, enable RPC/streaming mode, or even choose the reference frame for input coordinates. Do you think it's feasible?

PeterBowman, 2017-10-28:
Resolved at #117, improvements will be tracked in several follow-up issues:

* #118 - simultaneous execution of RPC and streaming commands
* #119 - implement last streaming command inherited from asibot-main
* #120 - use UDP carrier instead of TCP
* #121 - parameterize reference frames to shrink the list of commands and avoid duplicity

closed 2017-10-28 by PeterBowman