roboticslab-uc3m/kinematics-dynamics #106: Make ICartesianControl::stopControl actually stop motors - implement wait() method?

PeterBowman, 2017-07-26:
At the time of writing, you cannot issue a `stopControl` command during a blocking operation like `movj`, `relj` or `movl` (see [BasicCartesianControl/ICartesianControlImpl.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/8c1e7b04e1ea9471794634626fc28c96ed55a2e6/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp)) without creating a new client instance.

My proposal: make these RPC commands return immediately, and add a new `wait` method to the `ICartesianControl` interface (as in [CartesianServerLib/xRpcResponder.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5b6e049be9a59e1c11b807faefe51fee6f/src/libraries/CartesianServerLib/xRpcResponder.cpp#L35-L52) at old asibot-main code). We'd have to deal with restoring previous reference speeds *after* the command has finished execution ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/8c1e7b04e1ea9471794634626fc28c96ed55a2e6/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp#L131-L135)).

jgvictores, 2017-07-28:
LGTM!

PeterBowman, 2017-08-02:
BTW `movl` and similar commands currently return `VOCAB_OK` even if the cmc was forced to abort, e.g. because of exceeding the hardcoded joint velocity limit. The proposed `wait` command could take care of this and return failure vocabs in such circumstances.

PeterBowman, 2017-10-17:
Cf. `yarp::dev::ICartesianControl::waitMotionDone`:

http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#af1a253cb67ed5cf091854a58248e6446

jgvictores, 2017-10-17:
LGTM!

PeterBowman, 2018-02-03:
The wait period is no longer hardcoded, we can now tweak it on runtime via RPC config params: 1f62c00 (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-362797658).

jgvictores, 2018-02-03:
Cool!!!

PeterBowman, 2018-02-05:
Hmmm. Having a timeout is nice, but it's designed to work only on the server side. If network delays/errors occur, how can I enforce a timeout on the client side, too?

jgvictores, 2018-02-05:
- Delays: well, we just wait some more.
- Disconnection: tried a `yarp rpc` (client) against a read port (emulates a server, as `yarp rpc --server` is not working). Writing something from the client, awaiting for response, then killing the read port actually makes the client die:
   > yarp: BottleImpl reader failed, unrecognized object code 25
   > Cannot make connection
   > yarp: Alternative method: precede port name with --client

This is, I think server-side wait is enough.

PeterBowman, 2018-02-05:
Nice to know that, thanks!