roboticslab-uc3m/kinematics-dynamics #108: [AsibotSolver] Implement remaining ICartesianSolver methods

PeterBowman, 2017-08-07:
That is:

- [ ] `appendLink(const std::vector<double> &)`
- [ ] `restoreOriginalChain()`
- [ ] `invDyn(const std::vector<double> &, std::vector<double> &)`
- [ ] `invDyn(const std::vector<double> &, const std::vector<double> &, const std::vector<double> &, const std::vector< std::vector<double> > &, std::vector<double> &)`

And add their corresponding unit tests in `testAsibotSolverFromFile`, too.

PeterBowman, 2018-02-10:
`appendLink` and `restoreOriginalChain` could be implemented quite easily with the latest improvements, particularly #141.  Dynamics are a different beast, and AmorCartesianControl also lacks an implementation of `invDyn`-related commands. I'm prone to leaving them out of this issue and tracking their development in a new ticket.

jgvictores, 2018-02-10:
> Dynamics are a different beast, and AmorCartesianControl also lacks an implementation of invDyn-related commands. I'm prone to leaving them out of this issue and tracking their development in a new ticket.

+1

PeterBowman, 2018-02-10:
Both overloads of the `invDyn` method have been moved to #142, description updated accordingly.

PeterBowman, 2018-02-12:
Done at a12a9dd.

closed 2018-02-12 by PeterBowman