roboticslab-uc3m/kinematics-dynamics #115: Implement actuator control

PeterBowman, 2017-08-23:
The `ICartesianControl` interface could provide a method to control an end-effector actuator (e.g. a gripper tool). To not interfere with `tool` (see #95), a new `act`/`VOCAB_CC_ACT` method/RPC vocab pair is proposed.

PeterBowman, 2017-09-09:
This is easier to accomplish on AmorControlBoard since we have dedicated API methods for controlling the gripper. On the other hand, BasicCartesianControl connects (as of its current implementation) to a remote controlboard and use all of its joints, regardless whether there are *fake* ones (i.e. actuators such as the LacqueyFetch or the TextilesHand, see [yarp-devices/libraries/BodyYarp](https://github.com/roboticslab-uc3m/yarp-devices/tree/b3d27fc5e7c2bf65bc9c0b28eb0af1e1e8f4355a/libraries/BodyYarp)) or not. Didn't try it, but I guess that the robot controlboard plugin was launched without these additional "joints" so far.

To sum up, BasicCartesianControl should be able to distinguish motor and actuator joints via explicit `--param value` options, or fetch this information from the remote controlboard in case it implements the appropriate interface. I was thinking more about [yarp::dev::IAxisInfo](http://yarp.it/classyarp_1_1dev_1_1IAxisInfo.html) ([`VOCAB_JOINTTYPE_UNKNOWN`](http://yarp.it/namespaceyarp_1_1dev.html#ae65674b2d1ecaec0e4877e48ba9a0eef) for actuator joints?) or [yarp::dev::IRemoteVariables](http://yarp.it/classyarp_1_1dev_1_1IRemoteVariables.html) (retrieve the list of motor/actuator joints with [`getRemoteVariable`](http://yarp.it/classyarp_1_1dev_1_1IRemoteVariables.html#a22b53f8a7502c0d0bd5d25e563756a62)?). That said, the remote fetch solution would entail changes in the [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices/tree/b3d27fc5e7c2bf65bc9c0b28eb0af1e1e8f4355a/libraries/BodyYarp) repository.

jgvictores, 2017-09-09:
Sounds interesting. If implemented on yarp-devices, should also go in simulator, as in https://github.com/roboticslab-uc3m/openrave-yarp-plugins

PeterBowman, 2017-09-09:
My current concept for the future actuator-related method in the `ICartesianControl` interface is the following signature:

`bool act(int command);`

Where `command` would evaluate to one vocab of the following new set:

* `VOCAB_CC_ACTUATOR_NONE`: no actuator available or just *do nothing*
* `VOCAB_CC_ACTUATOR_CLOSE_GRIPPER`
* `VOCAB_CC_ACTUATOR_OPEN_GRIPPER`
* `VOCAB_CC_ACTUATOR_STOP_GRIPPER`
* `VOCAB_CC_ACTUATOR_GENERIC`
* ...

Still wondering how would this cope with hypothetical implementations of generic (non-gripper) actuators of any kind. Should the controller/remote controlboard/both be aware of the actuator type?

PeterBowman, 2017-10-28:
WIP at the `amor-cc-actuator` branch (locally) to get a first insight of this interface working on AMOR. As noted before, broader changes are required to accomplish this in the generic controller with custom actuators.

`ICartesianControl` vocabs:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/libraries/TeoYarp/ICartesianControl.h#L86-L91

New method signature:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/libraries/TeoYarp/ICartesianControl.h#L245-L254

Implementation at `AmorCartesianControl`:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/libraries/TeoYarp/AmorCartesianControl/ICartesianControlImpl.cpp#L224-L251

Used in several apps: [*keyboardController*](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/programs/keyboardController), [*streamingDeviceController*](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/programs/streamingDeviceController), [*haarDetectionController*](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/a9b6ecd1cfc2ea28041e2dc752b25c7d809bfe47/programs/haarDetectionController).

PeterBowman, 2018-01-22:
> That said, the remote fetch solution would entail changes in the yarp-devices repository.

Marking as blocked.

PeterBowman, 2019-07-03:
@jgvictores OK to merge amor-cc-actuator branch into develop as a first approach for tackling this problem (at least) on AMOR? We'll most probably rewrite that part sooner or later, but those changes would not be lost (and it will spare me doing a lot of merge commits on the already-too-long run).

PeterBowman, 2019-08-05:
> Update: I'm not convinced anymore about the IRemoteVariables interface, see [roboticslab-uc3m/yarp-devices#198 (comment)](https://github.com/roboticslab-uc3m/yarp-devices/issues/198#issuecomment-495322968). Also, the upstream work on yarp-devices would imply [roboticslab-uc3m/yarp-devices#211](https://github.com/roboticslab-uc3m/yarp-devices/issues/211).

Note: https://github.com/roboticslab-uc3m/yarp-devices/issues/20 will enable arbitrary CAN bus + YARP wrapper configurations, and https://github.com/roboticslab-uc3m/yarp-devices/issues/171 is going to wrap raw joints involved in the actuation. Two remarks:

* BCC needs to know how to treat/map ports (remote connections, i.e. we opened a controlboardwrapper2 exclusively dedicated for the gripper joints) and joints (local devices, i.e. we instantianted the CanBusControlboard instead of a remote_controlboard).
* In addition, BCC should be aware of the gripper type and its behavior on open/close commands. Changing my mind once again: the IRemoteVariables interface might do...
* Loose idea: migrate actuator vocabs to yarp-devices, expose them via installed header, implement IRemoteVariables on Dextra/TextilesHand/LacqueyFetch to perform the desired action, either include that header in user apps or call vocabs via RPC to command the tool.

PeterBowman, 2022-07-13:
Per https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/162#issuecomment-532344770:

> I plan to merge `numRobotJoints` and `numSolverJoints` into one entity that represents motor joints, only. That is, the (until and for now) only additional YARP-handled joint, which corresponds to the LacqueyFetch single-motor gripper, will be exposed via a dedicated YARP port to which BasicCartesianControl would have to connect via a separate remote_controlboard device in case it needs to access actuator commands (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/115). See rationale at https://github.com/roboticslab-uc3m/yarp-devices/issues/228#issuecomment-532200863.

We might want to rethink the whole `act` command stuff. See also https://github.com/roboticslab-uc3m/yarp-devices/issues/228.