roboticslab-uc3m/kinematics-dynamics #118: Initial configuration and internal state in BasicCartesianControl/CartesianControlServer

PeterBowman, 2017-10-20:
Some thoughts arised while working on the BCC/CCS at #117:

- [ ] Forbid simultaneous execution of RPC and streaming commands.
- [ ] Allow only one active connection for streaming commands (maybe RCP too)? Might be necessary to enforce this in the CartesianControlClient plugin.
- [ ] Keep track of run/stop states to avoid frequent mode changes and reduce amount of messages sent to the real robot.
   See following excerpt from `BasicCartesianControl::stopControl`.
   https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/37571fa1329c89ad77bd5b52255b2bfc558332c6/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp#L243-L246
  Perhaps we can solve it upstream (that is, in the robot controller), see roboticslab-uc3m/yarp-devices#120).
- [ ] Stop control if erroring at streaming commands? (remember that these return no success/failure flag)
- [ ] Set controller parameters on initial configuration: gain, maximum joint speed, trajectory time for `movl` commands, set velocity instead of duration (#46).
- [ ] Set those parameters above on runtime by implementing a new controller interface (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/105#issuecomment-334736223). Additionally, let users choose the current reference frame (inertial or end-effector) before sending the command.

PeterBowman, 2017-10-28:
This issue was repurposed to focus solely on internal state considerations regarding the BasicCartesianControl implementation and its network wrapper (possibly spanning to AmorCartesianControl, too). Initial and on-runtime controller configuration will be discussed at #121.

PeterBowman, 2018-01-12:
Regarding internal state and concurrency issues, check https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/129#issuecomment-354472379. Conveniently pasted excerpt:

> * multi-threading within the controller itself (e.g. `RateThread`): current locks cover the state identifier (`getCurrentState`, `setCurrentState`), but some other members are shared between threads without mutex guards (e.g. desired cartesian velocities and forces); by now, we could assume that issuing multiple commands simultaneously is not allowed and re-think the internal behavior in #118 :question:

PeterBowman, 2018-01-19:
Idea (from https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-350555938): stream configuration parameters (#121) in the recently created `state:o` port.

PeterBowman, 2018-01-20:
> Idea (from #121 (comment)): stream configuration parameters (#121) in the recently created state:o port.

Not sure about this, we'll rarely want to access config params in a high frequency manner. RPC set/get calls should be enough. The only advantage I see here is that `yarp rpc` access to the `stat` command will retrieve a comprehensive list of these parameters and their current value, but this would obviously render no benefits when querying the data programatically (i.e. via `CartesianControlClient`).

Another idea: provide a *bonus* RPC vocab to retrieve **all** config parameters with their value. For both interfaces:

* CLI (`yarp rpc`): simply list the values, similarly to `[help] [more]`
* C++ code: store keys and values in a `std::map`

jgvictores, 2018-01-20:
> Not sure about this, we'll rarely want to access config params in a high frequency manner. RPC set/get calls should be enough. The only advantage I see here is that yarp rpc access to the stat [...]

I agree.

> CLI (yarp rpc): simply list the values, similarly to [help] [more]
> C++ code: store keys and values in a std::map

Cool. For `yarp rpc`, something like `[get] [prms]` sounds good to me!

PeterBowman, 2018-01-21:
Added reference to related issue in description: #106.

PeterBowman, 2018-01-22:
In spite of 7d2e073, I'd like to decouple our internal vocabs from those used by YARP controlboard interfaces (which have nothing to do with cartesian control). @jgvictores OK to create `VOCAB_CC_OK`, `VOCAB_CC_FAILED`, `VOCAB_CC_SET` and `VOCAB_CC_GET`? They would resolve to the values users are already familiar with (`[ok]`, `[fail]`, `[set]`, `[get]`, respectively).

PeterBowman, 2018-01-23:
New vocab definitions and group parameter accessors: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-359628756.

PeterBowman, 2018-01-28:
Keyword: **state machines** (FSM). Several checks should be performed against internal state variables upon entering most of the control commands (e.g. current control mode, whether the trajectory instance is cleared and ready for use, etc.). The CMC thread would take care of any necessary cleanup before and after execution.

PeterBowman, 2018-02-21:
For now, marking as blocked by #145.

PeterBowman, 2018-12-27:
Moving this away from the issue description regarding control modes:

> See the following excerpt from `BasicCartesianControl::stopControl`:
> https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/37571fa1329c89ad77bd5b52255b2bfc558332c6/libraries/TeoYarp/BasicCartesianControl/ICartesianControlImpl.cpp#L243-L248
> Perhaps we can solve it upstream (that is, in the robot controller), see roboticslab-uc3m/yarp-devices#120).

PeterBowman, 2019-12-07:
Regarding locks and race conditions because of port callbacks being invoked at the same time, see [`yarp::os::Contactable::lockCallback`](http://www.yarp.it/classyarp_1_1os_1_1Contactable.html#a53c6446cec1b3cf84ee2e5d636fb4d45) and other lock-related methods.

closed 2022-07-13 by PeterBowman