roboticslab-uc3m/kinematics-dynamics #119: Investigate new streaming command: non-linear, position (direct) mode

PeterBowman, 2017-10-28:
The last command scheduled at #105 and present in old asibot-main code ([ref](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5/src/libraries/CartesianServerLib/xCallbackPort.cpp#L8-LL24)) was temporarily excluded due to current limitations. As pointed out in the title, it's meant to perform non-linear movements in position mode, that is, it's the *streaming* counterpart of `movj`.

The proposed signature is (no name yet):

```c++
void xxx(const std::vector<double> & x, double interval);
```

Due to the high-frequency nature of streaming commands, I'd propose switching to the [`yarp::dev::IPositionDirect`](http://www.yarp.it/classyarp_1_1dev_1_1IPositionDirect.html) to bypass trajectory generation at the iPOS level, hence the interval parameter intended for sampling point coordinates of the desired path (similarly to `ICartesianControl::pose`). However, this won't work on our simulator code (yet) unless we update the control boards to handle this interface.

jgvictores, 2017-10-29:
For TEO we implement [`yarp::dev::IPositionDirect`](http://www.yarp.it/classyarp_1_1dev_1_1IPositionDirect.html) via the Technosoft iPos `PT Mode`, which turns out to work quite nicely.

For the simulator, we could map `IPositionDirect` to `IPositionControl` or `IPositionControl2`. Regarding simulators, we've gone through quite a bit of iterations. Which one are you using currently, teoSim or the yarp-openrave-plugins via python loader?

PS: Put an issue on the https://github.com/roboticslab-uc3m/openrave-yarp-plugins repo if you want this implementation!

PeterBowman, 2017-12-12:
One drawback of the implementation I had in mind is that it would require high frequency computation of the inverse kinematics, which is most probably out of the question in KdlSolver (reminder: this solver currently works with numeric solutions). As a workaround, I'd consider exploring [screw theory](https://en.wikipedia.org/wiki/Screw_theory) and available KDL [solver interfaces](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1SolverI.html) (might end up creating our own implementation).

jgvictores, 2017-12-14:
Obviously screw theory is beautiful. :-)
However, I'd first look into:
1. Velocity/torque control via Jacobian (think of how to connect this into PT Mode or PVT Mode).
2. Wrap OpenRAVE's IKFast in our solvers.
3. AFAIK, OpenRAVE's IKFast generates analytical solvers given the robot geometry. I've seen more software that does similar, oriented to small robots that use microcontrollers (Bioloid, etc). Found: https://github.com/AversivePlusPlus/ik and also have asked a contact I recall using a similar software.

PeterBowman, 2017-12-14:
1. Perhaps through infinitesimal displacement twists ([Bruyninckx, *Robot Kinematics and Dynamics*, 4.1.8](https://people.mech.kuleuven.be/~bruyninc/tmp/HermanBruyninckx-robotics.pdf)), which somehow resemble virtual displacements, and then solve the usual delta_q = J^-1 * delta_x equation (note the deltas instead of time derivatives).

jgvictores, 2017-12-19:
>  However, this won't work on our simulator code (yet) unless we update the control boards to handle this interface.

Implemented on `YarpOpenraveControlboard` via https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/14da8de33a92987cf6c3f63f8afce8abd1aa05eb (example launch: [1](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/14da8de33a92987cf6c3f63f8afce8abd1aa05eb/examples/openraveYarpControlboard.py) and [2](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/examples/openraveYarpPluginLoader-controlboard.py)).

jgvictores, 2018-01-05:
> have asked a contact I recall using a similar software

https://vanadiumlabs.github.io/pypose/nuke-intro.html [thanks @JavierIH !!]

jgvictores, 2018-01-16:
> I'd consider exploring screw theory and available KDL solver interfaces (might end up creating our own implementation).

In https://github.com/robotology/codyco-superbuild they use https://github.com/ocra-recipes/eigen_lgsm

PeterBowman, 2018-02-16:
@jgvictores proposes that we use `movi` (immediate move) as the name for the new command, following the convention of `movj` (joint move) and `movl` (linear move).

PeterBowman, 2018-07-12:
See also https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/155#issuecomment-404444963 (ikfast).

PeterBowman, 2018-11-02:
Thoughts on implementing a screw theory-based solver in KDL (mutually exclusive options):

* Parse geometric data from DH-information stored in current .ini files as is. Most elaborate way, has to search for valid Paden-Kahan solutions through an iterative process - just like manually solving a math equation through variable isolation and any necessary intermediate steps.

* Provide some hints along with the DH parameters, e.g. which PK problem is most suitable, how to find coplanar axes and so on. Somewhat less convolute than iterating through every single point in the problem space and checking for valid paths.

* Add a new parameter group in the .ini files so that a full direct PK algorithm is provided to the solver. No step generation is needed, just follow the ones specified by the user, carefully.

The last two options require that users know how to define a screw theory-based problem, leading to data duplication: in the same .ini (or even in separate files), both DH params and ST information would be hosted. Any change that affects the geometry should be reflected (and manually computed/recalculated) for **two** different problem definitions.

I'd choose the hard way, see first option.

PeterBowman, 2018-11-03:
Iterative equation based on the POE:

<a href="https://www.codecogs.com/eqnedit.php?latex=\prod_{i=j}^{j&plus;k}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;=&space;\left&space;[&space;\prod_{i=1}^{j-1}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;\right&space;]^{-1}&space;\cdot&space;H_{ST}(\theta)&space;\cdot&space;\left&space;[&space;H_{ST}(0)&space;\right&space;]^{-1}&space;\cdot&space;\left&space;[&space;\prod_{i=j&plus;k&plus;1}^{N}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;\right&space;]^{-1}" target="_blank"><img src="https://latex.codecogs.com/svg.latex?\prod_{i=j}^{j&plus;k}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;=&space;\left&space;[&space;\prod_{i=1}^{j-1}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;\right&space;]^{-1}&space;\cdot&space;H_{ST}(\theta)&space;\cdot&space;\left&space;[&space;H_{ST}(0)&space;\right&space;]^{-1}&space;\cdot&space;\left&space;[&space;\prod_{i=j&plus;k&plus;1}^{N}&space;e\,^{\hat{\xi}_i\,{\theta_i}}&space;\right&space;]^{-1}" title="\prod_{i=j}^{j+k} e\,^{\hat{\xi}_i\,{\theta_i}} = \left [ \prod_{i=1}^{j-1} e\,^{\hat{\xi}_i\,{\theta_i}} \right ]^{-1} \cdot H_{ST}(\theta) \cdot \left [ H_{ST}(0) \right ]^{-1} \cdot \left [ \prod_{i=j+k+1}^{N} e\,^{\hat{\xi}_i\,{\theta_i}} \right ]^{-1}" /></a>

With `j = {1, 2, ..., N}`, `k = {1, 2, ..., N-1}`, `1 <= j+k <= N`.

PeterBowman, 2018-11-15:
[image]

PeterBowman, 2018-12-04:
I am happy to announce that the Screw Theory algorithm is close to being complete. First results have been obtained for the ABB IRB 120 robot (following @DrPardosGotor's lecture notes). Considering an initial home pose (same as the initial guess) and `q_i = pi/2` as the desired pose, the ST inverse kinematics solver performs 5-6 times faster than a numeric Newton-Raphson algorithm (as implemented in KDL) and produces all 8 available solutions. Execution times vary between 0.04-0.12 milliseconds.

DrPardosGotor, 2018-12-04:
Good for you Bartek!

Dr.Eng. Jose M. PARDOS-GOTOR, PhD.
Associate Professor
Systems Engineering & Automation
UC3M
M. +34 625 606 337

DrPardosGotor, 2018-12-05:
Many Thanks Bartek,
I would propose to make a presentation to understand in detail wha you have done. I suppose this mount is going to be complicated, but as soon as you can I believe It will be very interesting for all involved.
I wait for an appointment to the group (perhaps some Thursday at 19:00h).

Dr.Eng. Jose Manuel PARDOS-GOTOR, PhD.
Associate Professor
SYSTEMS ENGINEERING AND AUTOMATION

jgvictores, 2018-12-14:
I guess we'll be closing this issue soon thanks to #169, so whatever commented here will probably be buried in time... however, may be worthy to mention.

Regarding https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/169#issue-238697237: `5-10 times faster than a numeric Newton-Raphson solver provided by KDL`. Kind of leaves me asking if this is the order of magnitude of acceleration expected, and if we would do any better using some kind of GPU-accelerated solver such as [cuSOLVER](https://developer.nvidia.com/cusolver).

PeterBowman, 2018-12-14:
> Regarding [#169 (comment)](https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/169#issue-238697237): `5-10 times faster than a numeric Newton-Raphson solver provided by KDL`. Kind of leaves me asking if this is the order of magnitude of acceleration expected

Regarding MATLAB code and as stated in the reference I cited therein, a numeric solver (Robotics Toolbox) is about 1000x slower than the proposed Screw Theory implementation (0.1~1ms). Looks like KDL/C++ performs far better, although Screw Theory/C++ is still much faster.

jgvictores, 2018-12-21:
Done:
- Info documented at 35b1499c35feae085da13ff97ef796c2af542fb7
- Proposal followup at #171
Left to do:
- [ ] Make sure the doxygen of the newly developed classes have all the info here: diagrams, `5-10 times faster than a numeric Newton-Raphson solver provided by KDL`, etc.

PeterBowman, 2018-12-21:
Added example app at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/e777bc865dccc405124c687a6f4b3539b6f774f7.

PeterBowman, 2018-12-25:
>     * Make sure the doxygen of the newly developed classes have all the info here: diagrams (...)

@jgvictores I might have some useful diagrams generated in a couple of weeks. For now, though, I don't think https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/119#issuecomment-438997749 could be of any use. I did some work on describing the class internals either with Doxygen (preferable) or through inline comments only seen in source code, so that said diagram (severely WIP back then) has been already expressed in text/formula form. It would be much easier if you could state what exactly is missing as of the current state of the documentation in ScrewTheoryLib. That performance notice could fit inside KdlSolver (ST-KDL wrapper).

jgvictores, 2018-12-25:
1. Really cool: http://robots.uc3m.es/dox-kinematics-dynamics/classroboticslab_1_1ScrewTheoryIkSubproblem.html#ad2d0dbe398d37d2e2582e8730299dfa6
2. I understand you say https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/119#issuecomment-438997749 is already there in some updated but different format, could you please indicate (here on this same issue) a URL or how to reach it?
3. http://robots.uc3m.es/dox-kinematics-dynamics/group__ScrewTheoryLib.html or click on `more...` to readh http://robots.uc3m.es/dox-kinematics-dynamics/group__ScrewTheoryLib.html#details which is empty. I'm missing a small comment saying "this is faster and may provide better solutions than numerical methods because XXXX [ref]". I know there is a reference already, but I believe this is kind of the "first stop" before someone will decide to actually give a shot at the library (at least my own personal case).

PeterBowman, 2018-12-25:
> I understand you say [#119 (comment)](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/119#issuecomment-438997749) is already there in some updated but different format, could you please indicate (here on this same issue) a URL or how to reach it?

Mostly inline comments scattered all over [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/e777bc865dccc405124c687a6f4b3539b6f774f7/libraries/ScrewTheoryLib/ScrewTheoryIkProblemBuilder.cpp) plus that iterative formula you linked to in 1. Please mind that I never considered that implementation to be stable yet, several details would benefit from a :alien:2:alien: discussion and, in that vein, most of those comments could/should be rewritten.

> I'm missing a small comment saying "this is faster and may provide better solutions than numerical methods because XXXX [ref]".

👍

PeterBowman, 2018-12-25:
Docs improved at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/9c580a58509f187858a7d54341784c95e8653dcf.

jgvictores, 2018-12-26:
Thanks a lot, I think it's great right now. Ok to close issue!

PeterBowman, 2018-12-26:
>     * Add the `movi` streaming command.

Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/b9ca4d6edd69431c92f3926efde513b953108b2a, closing.

closed 2018-12-26 by PeterBowman