roboticslab-uc3m/kinematics-dynamics #120: Use UDP carrier in streaming contexts

PeterBowman, 2017-10-28:
As a follow-up to #105 and related to #102 (once the CartesianControlClient is capable of establishing a connection to the remote port), we'd like to investigate the consequences of switching from the default TCP carrier in the usual `yarp::os::Connect` calls to UDP. After all, the whole point of using streaming commands is that we seek high-speed comunication and don't care of casual data loss.

This is how it's done at `yarp::dev::RemoteControlBoard`: [ref](https://github.com/robotology/yarp/blob/8efdca9a8ae994bfdd28f2ad72bcefcdce2eedba/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L1050-L1053).

jgvictores, 2017-10-28:
Yes, that makes perfect sense!

PeterBowman, 2017-10-28:
More thoughts on the choice of a carrier: #123.

commit 082597aa (2017-12-05): Use UDP carrier in streaming output, use Contactable::addOutput

Closes #120.
closed 2017-12-06 by PeterBowman