roboticslab-uc3m/kinematics-dynamics #121: Let users access and modify low-level controller configuration

PeterBowman, 2017-10-28:
As discussed with @jgvictores, we'd like to stop and avoid an increasing expansion of the list of commands at `ICartesianControl` ([totalling 17 at the time of filing this issue](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/1236178e489ea74fdd1b3d5cf68f69906698f4fb/libraries/TeoYarp/ICartesianControl.h)). See background at #105.

Influenced by [`yarp::dev::ICartesianControl::tweakSet`](http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a07eb642241696232c9f142820d07ccb7) and [`yarp::dev::ICartesianControl::tweakGet`](http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#ae8b0254bdf597dad1eb4308f8c7affd5), consider adding a pair of setter/getter RPC methods that would allow users to change a set of controller parameters on runtime, such as:

* gain
* maximum joint velocity
* trajectory duration
* switch between velocity and duration as a limiter for `movl` and the like
* trajectory generator?
* switch between RPC/streaming modes?
* switch reference frame (intertial vs end-effector) -> palliates command duplication

Iin parallel to this, new configuration options should be implemented in the corresponding YARP devices (e.g. `yarpdev --device BasicCartesianControl --gain 0.15 ...`).

Example signature:

```c++
// for doubles...
bool setParameter(int vocab, double value);
bool getParameter(int vocab, double *value);

// for bools...
bool setParameter(int vocab, bool value);
bool getParameter(int vocab, bool *value);

// other types?
```

Multiple overloads may seem like overkill and go against the initial movation of this issue, but C++ lacks a standard container similar to `yarp::os::Value` (at least not until C++17, see [std::any](http://en.cppreference.com/w/cpp/utility/any)).

PeterBowman, 2017-12-07:
`ros::NodeHandle` implements a similar solution for accessing the parameter server ([ref1](http://docs.ros.org/jade/api/roscpp/html/classros_1_1NodeHandle.html), [ref2](http://wiki.ros.org/roscpp/Overview/Parameter%20Server)). ROS devs didn't hesitate to create overloads: `ros::NodeHandle::getParam` x 16, `ros::NodeHandle::setParam` x 16, plus their cached counterparts, and more, including [generic methods](http://docs.ros.org/jade/api/roscpp/html/classros_1_1NodeHandle.html#af0bf7baa0ab2385275bb657cc358111b) (which I think we can't use in an abstract base class).

A remark on that: ROS does not place constraints on the parameter identifier (the *key* part, as opposed to the *value*). It's meant to be a plain `std::string` which may point at any type of data, hence such a high number of overloads. Should we make them accept vocabs instead, a decision should be taken whether to define a narrow, basic set of properties in `ICartesianControl.h` (e.g. `VOCAB_CC_GAIN`), or delegate this responsibility to the implementors (e.g. `BasicCartesianControl` and `AmorCartesianControl` might accept different sets of vocabs). The latter imposes a serious restriction: client code must know beforehand *how* is the vocab encoded and construct it with `yarp::os::Vocab::encode` prior to handling the `vocab` parameter. This is obviously true for stringified parameter keys as well, but what is the point of enforcing a four-character limit for these descriptors, then?

PeterBowman, 2017-12-10:
Idea: publish some or all of these configuration parameters along with controller state and FK solution in `stat()` (#129). Discuss it here or at #118.

jgvictores, 2018-01-05:
Maybe just `double` and `string` for now?

PeterBowman, 2018-01-19:
I'm going to forget about `string`s for now... I can't see any valid use case for the time being.

PeterBowman, 2018-01-19:
Done at b0ccbf2. Currently enabled parameters:

* controller gain
* maximum joint velocity
* trajectory duration
* reference frame (implemented as vocabs):
  * base frame (inertial)
  * TCP frame (end effector)

No longer blocking #46.

closed 2018-01-19 by PeterBowman

PeterBowman, 2018-01-20:
Small follow-up at 944ec2a. Relevant changes:

* new configuration parameter: `cmcRateMs` (only `BasicCartesianControl`)
* encoded frame vocabs have been moved to `ICartesianControl::reference_frame`
* a few RPC commands now support the TCP frame, notably `movl` (**we can issue linear offsets**) and `movv`
  * remaining commands will be tracked in #136
  * **important:** `movl` and `pose` will crash unless `controllerGain` is set to zero

jgvictores, 2018-01-20:
> https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/944ec2a6a65b20d875b4fd5f7a6bbeba27b30882/libraries/YarpPlugins/ICartesianControl.h#L100-L105

Classy! Smooth! Nice!!

>   * **important:** `movl` and `pose` will crash unless `controllerGain` is set to zero

Should that be happening? In simulation, zero gain should work, but a non-zero gain should be needed with real motors.

PeterBowman, 2018-01-20:
> Classy! Smooth! Nice!!

Just a shameless copypasta from [YARP](http://www.yarp.it/IInteractionMode_8h_source.html#l00018) 😄 

> Should that be happening? In simulation, zero gain should work, but a non-zero gain should be needed with real motors.

In fact, this is closely related to #136 because of a `fwdKinError` call that does not support (yet) the TCP frame. See:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/944ec2a6a65b20d875b4fd5f7a6bbeba27b30882/libraries/YarpPlugins/BasicCartesianControl/RateThreadImpl.cpp#L63-L69

As a workaround, I suggest setting `gain` to zero so that the buggy command does not influence subsequent operations.

jgvictores, 2018-01-20:
Okay, understood. Thanks!

PeterBowman, 2018-01-23:
Another follow-up: commit 5ad5eea added two new methods for setting and retrieving multiple parameters at once. Also, set/get vocabs have been standardized. Example invocations via `yarp rpc`:

* `get cpjv`: retrieve maximum joint velocity
* `set cpjv 15`: set maximum joint velocity to `15`
* `get prms`: retrieve **all** configuration parameters
* `set prms (cpjv 15) (cpcr 60)`: set maximum joint velocity to `15` and CMC rate to `60` ms

Vocabs may be passed either as plain strings or in brackets: `cpjv` or `[cpjv]`.

PeterBowman, 2018-02-03:
Yet another follow-up (YAFU, which closely resembles [YAFUD](https://www.urbandictionary.com/define.php?term=YAFUD)): 1f62c00. Now, we can configure the check period within the implementors of `ICartesianControl::wait` (#106):

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/1f62c0012c13b640e5ae407b519ac77f434331b4/libraries/YarpPlugins/CartesianControlServer/RpcResponder.cpp#L104

PeterBowman, 2018-07-19:
https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-359628756

The `cpjv` (maximum joint velocity) parameter shall be removed after https://github.com/roboticslab-uc3m/yarp-devices/issues/188, and convenient calls to [`getVelLimits`](http://www.yarp.it/classyarp_1_1dev_1_1IControlLimits.html#a1c97fd5111935d2ee3dfc8dbade3d784) performed on controller init (similarly to `getLimits`).