roboticslab-uc3m/kinematics-dynamics #122: Store vocab definitions as enumerator values

PeterBowman, 2017-10-28:
Currently, we use `#define`s:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/1236178e489ea74fdd1b3d5cf68f69906698f4fb/libraries/TeoYarp/ICartesianControl.h#L26-L35

Enumerations seem like a cleaner way to achieve the same result (per https://github.com/robotology/yarp/blob/aac0dc7/src/libYARP_sig/include/yarp/sig/Image.h#L363-L392):

```c++
enum ICartesianControlVocabRpcCommandsEnum
{
    VOCAB_CC_STAT = VOCAB4('s','t','a','t'), ///< Current state and position
    VOCAB_CC_INV = VOCAB3('i','n','v'),      ///< Inverse kinematics
    VOCAB_CC_MOVJ = VOCAB4('m','o','v','j'), ///< Move in joint space, absolute coordinates
    VOCAB_CC_RELJ = VOCAB4('r','e','l','j'), ///< Move in joint space, relative coordinates
    VOCAB_CC_MOVL = VOCAB4('m','o','v','l'), ///< Linear move to target position
    VOCAB_CC_MOVV = VOCAB4('m','o','v','v'), ///< Linear move with given velocity
    VOCAB_CC_GCMP = VOCAB4('g','c','m','p'), ///< Gravity compensation
    VOCAB_CC_FORC = VOCAB4('f','o','r','c'), ///< Force control
    VOCAB_CC_STOP = VOCAB4('s','t','o','p'), ///< Stop control
    VOCAB_CC_TOOL = VOCAB4('t','o','o','l')  ///< Change tool
}
```

Some potential benefits: https://stackoverflow.com/a/11648734.

Note that, in #115 and #121, we want to pass similar vocabs to several new methods of the `ICartesianControl` interface and represent them as `int`s. This change would entail using the correct type (e.g. `ICartesianControlVocabRpcCommandsEnum` in the example above) instead of integer primitives.

jgvictores, 2017-10-29:
Interesting.
1. Are we sure we want to specify RPC (might end up in duplicates with streaming commands)?
1. Not my favorite, but just to lay out all the options, yet another way would be via `static const int`, see [this](https://github.com/asrob-uc3m/robotDevastation/blob/6436cb9f23443c2ba7c61e6aa1624e03b6635428/src/libraries/InputLib/Key.hpp#L37-L45) and [this](https://github.com/asrob-uc3m/robotDevastation/blob/6436cb9f23443c2ba7c61e6aa1624e03b6635428/src/libraries/InputLib/Key.cpp#L72-L81).

PeterBowman, 2017-10-29:
1. Not sure if I understood this. Could you provide an example?
2. We'd have to deal with the intricacies of the One Definition Rule (those vocabs are defined in a header file). Apart from that, we'd end up adding data to a pure virtual class ([SO question](https://stackoverflow.com/q/33197720) - I recall having referenced this somewhere).

jgvictores, 2017-10-30:
1. Just thinking if something among the lines of `ICartesianControlVocabCommandsEnum::Rpc` and `ICartesianControlVocabCommandsEnum::Streaming` would be interesting vs ugly.
1. Agreed, `static const int` is definitely not my favorite option.

PeterBowman, 2017-10-31:
I didn't notice that switching to enumerations will probably require some changes in the function signatures proposed in #121. Note to self: investigate implicit conversions (unscoped `enum` to `int`) per [this SO answer](https://stackoverflow.com/a/23550073), or adapt the signature to these new types.

PeterBowman, 2018-01-20:
While working on #121, I committed aeae9df (see also https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/121#issuecomment-359174013). Now, we have a mixture of macro definitions and enumerators in `ICartesianControl.h`. Conversions from `int`/`double` to an enumerator type require a `static_cast`. Function signatures of the newly added `getParameter` and `setParameter` methods actually accept a `double`, but we can pass a variable of enumeration type and the compiler will perform an implicit conversion.

PeterBowman, 2018-04-19:
> yet another way would be via `static const int`

Or even `constexpr` (C++11), see https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/150#issuecomment-382790532.

PeterBowman, 2018-06-28:
Following https://github.com/robotology/yarp/issues/1755, namespace-aware `constexpr` *macros* are the new hype.

PeterBowman, 2021-06-08:
I think it's OK as the way it is now: global vocabs in ICartesianControl.h, enumerations in ICartesianSolver.h. Moving that in either direction (convert to enumeration or global variable) would break the interfaces. Grouping vocabs as enums for the sake of classification is not a must, we already have that in doxy.