roboticslab-uc3m/kinematics-dynamics #128: Support KDL 1.3.0 ?

jgvictores, 2017-11-27:
The reason for not supporting KDL 1.3.0 was potentially supoorting #15 . Given the final solution to #15, I'm not sure KDL 1.4.0 is actually a hard requirement (current lack of support could be solved as stated [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/72#issuecomment-272325956)). Must see if providing support to KDL 1.3.0 does not affect 15.

PeterBowman, 2018-01-24:
Just to raise awareness: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/72#issuecomment-360289354.

jgvictores, 2018-01-27:
Thoughts about supporting both, e.g. via pre-processor checks.

PeterBowman, 2018-01-27:
> Just to raise awareness: #72 (comment).

That comment refer to an edge case that may arise whenever two conditions are met:

* multiple versions of a dependency (e.g. YARP, KDL) found on system
* more than one distinct version required in `find_package()` calls

The latter is not that unusual. In the openrave-yarp-plugins repo, some plugins require a recent YARP release to be compiled against. The former plays a relevant role when linking, but as stated in the comment, I discovered that CMake handles headers in the same dependency-driven manner. Thus, non-compatible headers may be requested for compiling sources that depend on a more recent version of the 3rd party software.

Multiple versions should be always handled with care, but it's more a concern for the user than for the developer, that is, it's hard to cover all kind of system configurations on the user side. The KDL thing was caused by ROS having its own package installed in `/opt/ros` **and** then literally forced to be picked first by CMake due to some environment variable conveniently set (see last comments at #40). This is not a good practice anyway, but experience has taught us that attempts to override that imposed behavior may lead to more problems (check reasons behind 65c4969).

Long story short, @jgvictores's comment refers to the fact that we can sometimes anticipate version issues within the code itself. Instead of letting CMake try to solve the problem with system instrospection and deduce whatever library version is requested (actually, with ROS hiding around the corner, I'd compare this with walking on a minefield), we can write code that supports more than one version and handle it with the usual `#ifdef` directives, while CMake simply queries and tells us the library version determined by `find_package()`.

PeterBowman, 2018-06-28:
CMake code is now a bit cleaner as a result of https://github.com/roboticslab-uc3m/questions-and-answers/issues/54. There is a single `find_package(orocos_kdl 1.4 QUIET)` call at root *CMakeLists.txt* that affects all devices. However, we could lower it down to 1.3 and let devices decide whether they support 1.4, too, via preprocessor definitions. That is, I think the only piece of code that **requires** KDL 1.4 are the `idsolver.strError` occurrences at [KdlSolver/ICartesianSolverImpl.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/9e87f28b187aa7d384698cb4dc6f50283789ece1/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp). Solution: either drop this and return to KDL 1.3-compatible code only, or apply preprocessor macros and conditionals (perhaps a bit too overkill?).

jgvictores, 2018-06-28:
Trying to track KDL version on ROS, both Indigo and Kinetic (as well as Lunar) use 1.3.1. Have to wait for Melodic for 1.4.0. For me it's no real issue to drop `idsolver.strError`.

For reference:
- https://github.com/orocos/orocos-kdl-release/blob/release/indigo/orocos_kinematics_dynamics/package.xml
- https://github.com/orocos/orocos-kdl-release/blob/release/kinetic/orocos_kinematics_dynamics/package.xml
- https://github.com/orocos/orocos-kdl-release/blob/release/lunar/orocos_kinematics_dynamics/package.xml
- https://github.com/orocos/orocos-kdl-release/blob/release/melodic/orocos_kinematics_dynamics/package.xml

PeterBowman, 2018-06-28:
(note to self?) If we go that way (drop 1.4), KDL builds via Travis should check out the 1.3.x tag.

BTW it's a pity that Trusty didn't get those convenient apt packages ([ref](https://packages.ubuntu.com/search?keywords=liborocos-kdl-dev), perhaps through a PPA?). On the other hand, https://github.com/roboticslab-uc3m/installation-guides/commit/3f543d15e63440bc5779440fb4a74d9d18a53c2d could make sense now.

PeterBowman, 2018-06-28:
Expanding on my previous comment (see "BTW"), since users are already encouraged to clone and install KDL master, I think there is no use of supporting v1.3.x anyway. On the other hand, if we make `apt install liborocos-kdl-dev` the preferred way to work with kin-dyn and others (and I kinda like it for its simplicity, but **as long as we don't depend on new KDL features**), then OK.

jgvictores, 2018-06-28:
> as long as we don't depend on new KDL features

Such as a future LMA_JL? :smile:

PeterBowman, 2018-06-28:
That's https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/155. Yeah, perhaps, I'm just a bit skeptical about seeing it added upstream anytime soon. If we don't care about entropy, we'll have another copy of its implementation in kin-dyn...

PeterBowman, 2018-06-30:
As spoken with @jgvictores, and especially accounting on future PRs (e.g. the LMA_JL thing), we'll stick to KDL master (currently 1.4.0).

jgvictores, 2018-12-13:
PS: At time of writing, we are using `updateInternalDataStructures`, which is not even in all 1.4.0 commits.

closed 2022-05-22 by PeterBowman