roboticslab-uc3m/kinematics-dynamics #133: Think about a gcode interpreter

jgvictores, 2018-01-06:
Just a fun idea commented with @PeterBowman yesterday: Develop a program that opens a `.gcode` file (pretty sure that inkscape can export this, which is cool for example after vectorizing a bitmap), and can create an instance of an object that inherits from `Trajectory` that is runnable via `BasicCartesianControl` (note that some methods to do this still do not exist).

jgvictores, 2018-01-06:
Blocked by #134 and #135.

jgvictores, 2018-01-06:
Just for completeness, the syntax we were thinking about is similar to:
```bash
./launchGcode --file whatever.gcode --orient 1 0 0 0
```

David-Estevez, 2018-01-06:
There is a specific need for the gcode? You can obtain the trajectory directly from a SVG file, without the need to convert to gcode, if you are careful with the desired dimensions of the trajectory.

jgvictores, 2018-01-06:
No, it's just an idea, no practical use-case. The benefit of `.gcode` is we could even use `slic3r`-generated trajectories. :smile:

PeterBowman, 2022-07-13:
Any trajectory classes we envisioned have been superseded by KDL: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134#issuecomment-794102173. Perhaps we can write a small script to convert from gcode to text, and then use our playback app ([see tools repo](https://github.com/roboticslab-uc3m/tools/tree/5c6251226b3b37041a72be4ee02bde79c5279b90/examples/cpp/examplePlaybackThread)). @jgvictores shall we close this as invalid or wontfix?

jgvictores, 2022-07-17:
> shall we close this as invalid or wontfix?

Yup!

closed 2022-07-17 by jgvictores

jgvictores, 2024-01-25:
- https://github.com/DAguirreAg/GCode-to-ABB
- "G-Code to RAPID translator for RobotStudio" by Daniel Nilsson (2016), https://www.diva-portal.org/smash/get/diva2:1034182/FULLTEXT01.pdf