roboticslab-uc3m/kinematics-dynamics #134: Improve TrajectoryLib

jgvictores, 2018-01-06:
Improve TrajectoryLib, starting by decoupling one and two limb trajectory generation.

Blocks #133, and more to come.

jgvictores, 2018-01-06:
Blocks #135 in addition to #133.

jgvictores, 2018-01-11:
Additionally, the current [Trajectory](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/3b3600e4268e9580365bf3fc71bd0f0fcb46658b/libraries/TrajectoryLib/Trajectory.hpp) class is currently 1 DoF. We need high DoF and unambiguously named classes for each.

jgvictores, 2018-01-11:
Note that several design decisions must be taken. We might have a meeting in person for this, I'll keep you informed.
cc: @PeterBowman @rsantos88

jgvictores, 2018-01-12:
@PeterBowman commented that he had started refractoring TrajectoryLib a while ago. Maybe some code or ideas hat we can reuse.

jgvictores, 2018-01-12:
[image]

jgvictores, 2018-01-16:
Also blocks https://github.com/roboticslab-uc3m/teo-grasp/issues/4

jgvictores, 2018-01-16:
@PeterBowman expressed concerns regarding being able to add several waypoints in a single function call.
@rsantos88 this is also useful for teo-grasp in relation to porting trajectories from openrave.

With https://github.com/roboticslab-uc3m/tools also in mind, I'm raising the bar and invoking some magic keywords:
```ominous-but-enigmatic-warning
serialization/deserialization
```

A.k.a. marshalling and unmarshalling.

rsantos88, 2018-01-18:
Just an idea, I think it would be interesting to make an example with the functional part of this library (in case it is usable) for example with `LineTrajectory`

jgvictores, 2018-01-18:
> Just an idea, I think it would be interesting to make an example with the functional part of this library (in case it is usable) for example with `LineTrajectory`

Yes! Definitively. As a bare minimum, we should at least `test` the resulting class(es).

jgvictores, 2018-01-19:
Lots of progress (developed `ITrajectoy`, `ICartesianTrajectoy`, `KdlTrajectory`) at f02a0d8283c063eddd347f1ed10ca0221c01ef2d, actually improving `BasicCartesianControl` too (#135).

jgvictores, 2018-01-19:
> Just an idea, I think it would be interesting to make an example with the functional part of this library (in case it is usable) for example with LineTrajectory

Added [this test](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/92a0c0b018ce55c936b812b0d204281125516822/test/testKdlTrajectory.cpp) at 92a0c0b018ce55c936b812b0d204281125516822.

Still pending (but coming soon!): trajectories with paths other than `ICartesianTrajectory::LINE`.

jgvictores, 2018-01-20:
> Next thoughts should be in the line of discrete trajectories, to address the use case of, say, wanting to retrieve the timestamp of the n-th waypoint.

In addition to a potential (linear?) interpolation if requesting a [movementTime from any of the methods](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/92a0c0b018ce55c936b812b0d204281125516822/libraries/TrajectoryLib/ITrajectory.hpp#L35-L63) that does not correspond precisely to a waypoint (and e.g. avoiding the interpolation if it does correspond.... within a certain tolerance xD).

jgvictores, 2018-02-13:
Began `IJointTrajectory` at bd660a4dc94cf10b76ca84037211c2d4f2653c20, currently at `issue-164-joint` branch.

PeterBowman, 2018-07-19:
https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/157 affects on how trajectories are to be generated: pass a duration parameter straight to the KDL API, or do some math with distances/velocities and obtain approximate duration first.

PeterBowman, 2018-10-27:
Currently, maximum velocity and acceleration defaults are enforced:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/8462fa859847378afe6e75d7bd4fb463d61f2b2b/libraries/TrajectoryLib/KdlTrajectory.hpp#L17-L18

I believe we should enable their customization somehow (through additional interface methods or via constructor?).

PeterBowman, 2018-10-28:
> I believe we should enable their customization somehow (through additional interface methods or via constructor?).

Added to constructor in https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/d223d392e0cbb63b28e3e7594ee16251148045ae.

jgvictores, 2018-12-13:
WIP at 6f3e7d8.

jgvictores, 2018-12-13:
Also blocks https://github.com/roboticslab-uc3m/teo-grasp/issues/3

jgvictores, 2018-12-14:
It turns out that [libraries/YarpPlugins/AsibotSolver/TrajGen.hpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/6ed8110a8a5879d9d119c2312256e154af73f880/libraries/YarpPlugins/AsibotSolver/TrajGen.hpp) would be useful for the new [libraries/TrajectoryLib/BasicJointTrajectory.hpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/ee28500692d1a514c71194bf6dd8781458567cd0/libraries/TrajectoryLib/BasicJointTrajectory.hpp). @PeterBowman Maybe we could move things aroud (also rename a bit, something like `SingleJointTrajectory`)?

PeterBowman, 2018-12-14:
TrajGen was kept for historical reasons, no code of ours is using that ATOW. If I'm correct, we even came close to removing it since KDL provides a similar functionality: [`KDL::VelocityProfile_Spline`](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1VelocityProfile__Spline.html) (supports up to 5th order splines, which we don't). Does it make sense to integrate/move TrajGen in the `IJointTrajectory` inheritance tree (even expand it so that initial/final accelerations are taken into account) and, if we ever need it, support velocity splines in the `ICartesianTrajectory` tree by means of the existing `KdlTrajectory` wrapper?

jgvictores, 2018-12-14:
I'm not a big fan of adding dependencies, but even less a fan of code duplication. After reading your comment, this makes me inclined towards the use of KDL even for joint space trajectories. At time of writing [`KDL::VelocityProfile`](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1VelocityProfile.html) provides 5 different implementations that we may have ended up re-implementing (including the  [`KDL::VelocityProfile_Spline`](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1VelocityProfile__Spline.html) you cite).

Looking at [current `issue-164-joint` branch](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/ee28500692d1a514c71194bf6dd8781458567cd0/libraries/TrajectoryLib), maybe some renaming to be done?
- KdlTrajectory -> KdlCartesianTrajectory
- BasicJointTrajectory -> KdlJointTrajectory

PeterBowman, 2018-12-14:
> [`KDL::VelocityProfile`](http://docs.ros.org/latest/api/orocos_kdl/html/classKDL_1_1VelocityProfile.html) provides 5 different implementations that we may have ended up re-implementing

For the record, we already use `KDL::VelocityProfile_Trap`, and `KDL::VelocityProfile_Rectangular` is WIP for https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/166. But, there is no other re-implementation I know of apart from said TrajLib/`KDL::DL::VelocityProfile_Spline`. That one originated at asibot-main, probably before KDL was first used in RL code.

> maybe some renaming to be done?

If KDL already has everything we need to implement `IJointTrajectory`, fine, then. I'll take care of edit conflicts across branches.

jgvictores, 2019-04-08:
Slightly blocked, focusing on https://github.com/roboticslab-uc3m/teo-main/issues/38 right now.

PeterBowman, 2019-06-25:
ASWJ we might consider dropping the `ITrajectory` wrapper class tree in favor of vanilla [KDL's motion API](http://docs.ros.org/latest/api/orocos_kdl/html/group__Motion.html).

jgvictores, 2019-12-21:
As spoken at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134#issuecomment-505659623, [KDL's motion API](http://docs.ros.org/latest/api/orocos_kdl/html/group__Motion.html) seems reasonable.

Furthermore, here are some notes to future selves:
1. I would not completely forget ideas at figure https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134#issuecomment-357325384
1. I'd also like to keep in mind seeing how this connects to:
    - YARP: https://www.yarp.it/yarpdataplayer.html but also https://www.yarp.it/yarpmotorgui.html
    - OpenRAVE: http://openrave.org/docs/0.8.0/architecture/trajectory/

jgvictores, 2020-03-13:
Recent discovery: https://www.yarp.it/namespaceyarp_1_1rosmsg_1_1trajectory__msgs.html

PeterBowman, 2021-03-09:
@jgvictores, let's ask the bold question: should we drop branch `issue-164-joint` (badly named, by the way) and turn this issue into "Nuke TrajectoryLib"? That is, use KDL directly instead of this thin wrapper of ours. I'm not quite proud of it and its pitfalls, had to code very carefully in the past and yet kept finding memory leaks. Extending this library (e.g. for rectangular velocity profiles) felt kinda useless as I could just call KDL methods instead.

jgvictores, 2021-03-09:
> let's ask the bold question: should we drop branch issue-164-joint (badly named, by the way) and turn this issue into "Nuke TrajectoryLib"? That is, use KDL directly instead of this thin wrapper of [...]

Yesssssssssssssssssss :+1:

PeterBowman, 2021-03-09:
Tactical nuke dropped, there is no more TrajectoryLib at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/692888d12996d1fdd47169979c28839b55dbc16e. Last stable implementation was available [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/a9acac8b8246c77be885f83648b12eddcac43d3c/libraries/TrajectoryLib). The `issue-164-joint` branch will be deleted, here is the full patchset for reference: https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/a2bd56ccb7b6cbaa98adc00c2f7298eb1de4c659...d2665723906c9f4c555714e696f8f4fe8c766e9b.

Keep in mind the KDL motion API allows you not to configure certain parameters on construction, such as trajectory duration, maximum velocity, reference acceleration and so on. However, those must be accounted for anyway. A particular scenario involves trajectories with a starting point, a constant velocity and infinite duration (no end point):

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/a9acac8b8246c77be885f83648b12eddcac43d3c/libraries/TrajectoryLib/KdlTrajectory.cpp#L242-L280

`SetProfileDuration` can be used with a dummy waypoint which we know would never be reached, e.g. 10 meters away from start:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/a2bd56ccb7b6cbaa98adc00c2f7298eb1de4c659/programs/keyboardController/LinearTrajectoryThread.cpp#L91-L95

closed 2021-03-09 by PeterBowman