roboticslab-uc3m/kinematics-dynamics #135: Improve BasicCartesianControl

jgvictores, 2018-01-06:
Improve BasicCartesianControl, adding methods to design and run more complex trajectories. For example,
 start with `MoveC`. Blocked by #134. Blocks #133.

Note: Slight possiblity of having to add a new trajectory generation layer, but for now want to try to do within BasicCartesianControl.

jgvictores, 2018-01-19:
Lots of improvements at f02a0d8283c063eddd347f1ed10ca0221c01ef2d as commented at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134#issuecomment-359099379

jgvictores, 2018-01-19:
Removed `blocked` label, WIP in parallel with #134.

PeterBowman, 2018-01-23:
Idea: perform a dry run on the desired trajectory (usually `movl`) and check for any limit violations, e.g. maximum joint velocity might be exceeded at some point or the operation would take too long (the latter does not apply to current strategies since trajectory duration is one of our input parameters). On success, perform the trajectory; on failure, return immediately and notify callers.

PeterBowman, 2018-06-28:
> Idea: perform a dry run on the desired trajectory (usually movl) and check for any limit violations, e.g. maximum joint velocity might be exceeded at some point or the operation would take too long [...]

Moved to #156.

PeterBowman, 2021-03-17:
Given the outcome of https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134 and https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/168#issuecomment-801017414, we are steering towards the externalization of trajectory generation in BasicCartesianControl. In other words, clients should take care of providing new setpoints (`movi`) to the controller. For instance, one can implement the `MoveC` thing with [`KDL::Path_Circle`](http://docs.ros.org/en/indigo/api/orocos_kdl/html/classKDL_1_1Path__Circle.html), iterate over time and send the usual commands.

@jgvictores are you OK with closing this as WONTFIX, too?

jgvictores, 2021-03-17:
> @jgvictores are you OK with closing this as WONTFIX, too?

Yep!

closed 2022-05-22 by PeterBowman