roboticslab-uc3m/kinematics-dynamics #136: Handle TCP frame in FK solver and controller methods

PeterBowman, 2018-01-20:
As a follow up to #121, we now may use the `diffInvKin` and `diffInvKinEE` methods from `ICartesianSolver` to issue a few commands in the cartesian space referring to the end-effector/TCP frame. However, some other depend on FK/IK transformations (i.e. non-differential kinematics), which do not support TCP frames yet.

That being said, we'll probably need the end-effector counterparts of `invKin` and `fwdKinError`.

PeterBowman, 2018-02-07:
Still WIP, but `diffInvKin` and `diffInvKinEE` have been already merged into one method, which now accepts `ICartesianSolver::reference_frame` as a parameter. No further changes here. Regarding the other two methods:

* `invKin`: similarly to `diffInvKin`, users may pass the reference frame in which the input pose should be expressed.
* `fwdKinError`: I dismissed my original idea since the input cartesian pose should always be a **static** reference, i.e. it doesn't make sense to express it in terms of the TCP frame as it would keep changing during the execution of the trajectory.

To bypass the latter, I propose yet another solver method: `changeReferenceFrame` (name may change).

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/9643744bcfb7eaa21a1c8f7fa3f1efdc0bed74b4/libraries/YarpPlugins/ICartesianSolver.h#L64-L80

PeterBowman, 2018-02-08:
BTW I'd probably need to update `ICartesianSolver`'s docs so that the reference frame of input/output pose vectors is explicitly stated.

PeterBowman, 2018-02-08:
Both available implementations of `changeReferenceFrame` are prepared to handle a world frame, besides the currently available base and TCP frames (`ICartesianSolver::BASE_FRAME` and `ICartesianSolver::TCP_FRAME`, respectively).

Regarding world and TCP frames, let's discuss a reasonable way of handling the `H0` and `HN` .ini config parameters (currently used in KdlSolver, still pending for AsibotSolver at #112).

jgvictores, 2018-02-09:
Ref: "una imagen vale más que mil palabras"

[image]

PR #141 (2018-02-10): Perform kinematics in the TCP frame
Fixes #136.

Notable changes in the `ICartesianSolver` interface:

* `ICartesianControl::reference_frame` enumerator types have been moved to the solver interface. Now, `ICartesianControl.h` includes `ICartesianSolver.h`.
* `diffInvKin` and `diffInvKinEE` have been merged. This method accepts a reference frame.
* `invKin` accepts a reference frame.
* `fwdKinError` has been repurposed into `poseDiff`. Now, it has no knowledge about the robot state.
* New method: `changeOrigin`.
closed 2018-02-10 by jgvictores