roboticslab-uc3m/kinematics-dynamics #138: Disable AmorCartesianControl with warning if AMOR_API not found

jgvictores, 2018-01-31:
Disable AmorCartesianControl with warning if AMOR_API not found

PeterBowman, 2018-01-31:
Related: https://github.com/roboticslab-uc3m/questions-and-answers/issues/18#issuecomment-361898108.

> Perhaps we should move most `find_package()` calls to an upper level (root `CMakeLists.txt`?) and then use the corresponding `_FOUND` variables in `cmake_dependent_option()`/`yarp_prepare_plugin()` (...)

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5913c9f4e02ad52ed6f6095a40e929d89adf533f/libraries/YarpPlugins/AmorCartesianControl/CMakeLists.txt#L9

Proposal:

```cmake
DEPENDS "ENABLE_KinematicRepresentationLib;AMOR_API_FOUND"
```

PeterBowman, 2018-02-02:
I'll try this: https://github.com/roboticslab-uc3m/questions-and-answers/issues/18#issuecomment-362647755.

PeterBowman, 2018-02-02:
Done at 14cf9cb, I'll drop some notes on this at roboticslab-uc3m/questions-and-answers#18.

closed 2018-02-02 by PeterBowman