roboticslab-uc3m/kinematics-dynamics #145: Make solver implementations more aware of their internal state

PeterBowman, 2018-02-21:
Solver devices could hold more information about their internal state, such as the angle representation (removed at #113 so that a single copy is managed by the server device and passed down the chain, i.e. controller & trajectory generator & solver), thus rendering more useful as standalone components not necessarily linked to a controller device at all times (cf. [transCoords app](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/6063082a298605d9202cc2ed17793b0929a2c133/programs/transCoords)).

One potential benefit of said awareness is related to the fact that users are now fully responsible for managing the right angle representation and its conversions between `ICartesianSolver`'s standard format and whatever their app handles. In other words, spare them several annoying calls to `KinRepresentation::encodePose` and `KinRepresentation::decodePose` every time they want to interface with a solver.

As proposed by @jgvictores, there could be an intermediate layer between the actual `ICartesianSolver` interface and the implementation. In that vein, the former accepts the angle representation of choice, as specified by the user. The latter works with KDL standard (scaled axis-angle) or whatever suits it better, and in fact would behave as a `ICartesianSolverRaw` or such. The new layer may now hold the angle representation and use it to manage all conversions needed between interface-implementation. We already work with something similar in the server device, check the [`RpcTransformResponder`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/6063082a298605d9202cc2ed17793b0929a2c133/libraries/YarpPlugins/CartesianControlServer/RpcResponder.cpp#L372-L382) conversion methods.

Several solutions were laid out at #113 and could be now considered again. My main concern, and one of the reason behind how things are currently done, is that angleRepr will be supervised by several components, perhaps introducing duplicate routines and time overheads. Also, consistency must be kept when working with the full chain (server + controller + solver).

PeterBowman, 2018-02-21:
More ideas regarding internal state in solvers (per https://github.com/roboticslab-uc3m/teo-main/issues/15#issuecomment-367320582):

> > @PeterBowman I'd like to discuss the possibility of removing joint limits from the kinematics .ini used from the solver:
>
> Those .ini values are not used at all, so IMO it's totally reasonable to remove them. Speaking of current cartesian controllers, I'd consider opening the robot device before the solver one, pick actual robot joint limits via `IControlLimits` interface, and pass them later to the solver upon configuration (e.g. `solverOptions.put("min", "whatever")`).
>
> One thing to sort out is: should joint limits be constrained as read-only properties in `ICartesianSolver`?
>
> * Yes, we set them upon device initialization and never modify their values again. `ICartesianSolver::setLimits` could be either removed or rewritten to hold the corresponding class invariants (= report failure if called more than once, which is perhaps a bit against the purpose of said method).
> * No, keep `ICartesianSolver::setLimits` so that users may call it at will on runtime, but take some care to ensure it'll never fall into race conditions. Similarly, variables holding the kinematic chain have been guarded against such circumstances, recently.

jgvictores, 2018-02-21:
1. I agree with the general statement `Make solver implementations more aware of their internal state`.
1. We have allowed tool changing during runtime, and this is good because this is frequent in robotics. Regarding joint limits, I believe initial setup would be enough, as in `solverOptions.put("mins", "(val1 ... valn)")`. The implementation side is easier as we avoid race conditions, and it is justified because we have no use case or potential use case with varying joint limits. The only end-user inconvenience is that the solver cannot be initialized until *after* *connecting* to a robot **and getting** joint limits. Not bad, and reasonable practice, IMHO. Regarding the treatment, two options:
   1. Draconian. Force user to establish limits, else fail.
   1. Fall back to +-180 degree defaults or similar, simply `CD_WARNING` (may pass unperceived anyway).

jgvictores, 2018-02-21:
Blocks #118.

PeterBowman, 2018-02-21:
> may pass unperceived anyway

Indeed, I forgot to call `setLimits` in [TransCoords::configure](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/6063082a298605d9202cc2ed17793b0929a2c133/programs/transCoords/TransCoords.cpp#L50-L93). Ouch.

jgvictores, 2018-02-23:
+1 to draconian.

PeterBowman, 2018-02-27:
With #146, I started using `yarp::os::Value::makeList(const char *)` to populate a bottle of minimum or maximum joint values, thus simplifying the process of working with bottles and values (`yarp::os::Property::put` requires a pointer to a `Value`). Example:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/7ea960b810161ae0d398b3c668b3544683cac49f/programs/transCoords/TransCoords.cpp#L98-L99

However, I didn't realize that this kind of creator funcion uses dynamic allocation, and returns a pointer to an object that needs to be destroyed further on. In other words, without a `delete` instruction, memory leaks are likely to happen. Check [docs and source code](http://yarp.it/classyarp_1_1os_1_1Value.html#a0fb1d891b92b71d9dd00dc347e2d6bba).

jgvictores, 2018-03-02:
> I didn't realize that this kind of creator funcion uses dynamic allocation, and returns a pointer to an object that needs to be destroyed further on. In other words

Ok, yes, that es important!

PeterBowman, 2018-03-02:
> However, I didn't realize that this kind of creator funcion uses dynamic allocation, and returns a pointer to an object that needs to be destroyed further on. In other words, without a delete instruction, memory leaks are likely to happen. Check docs and source code.

It's fine. Per [`yarp::os::Property::put(const ConstString &, yarp::os::Value *)`](http://yarp.it/classyarp_1_1os_1_1Property.html#afb5e63e83766e775c2ce23faa4aa9887):

>  After the association find(key) will return that value. **The Property object is responsible for deallocating the value.** If key is already associated, the value will be replaced with the new one.

PeterBowman, 2018-03-02:
[image]

jgvictores, 2018-03-05:
I've been thinking... Perhaps the CCRH wouldn't even need an additional interface class, since there is only going to be one, right?
(would inherit from DeviceDriver, but then be view()ed directly)

jgvictores, 2018-03-05:
There are actually 3 possiblies:
1. What I just proposed, which is an intermediate proposal. 
1. Not a YARP device at all, just a helper class. Would provide the cleanest syntax, but would require the end-user most code changes to switch between RH and non-RH. 
1. What @PeterBowman proposed, `ICCRH : public ICC` would perhaps provide the interface with an API in line with the rest, and only add that single line of code. For now, I think I'd go with this one.

PeterBowman, 2018-05-03:
91fec270146b28def7635443666f64b02c8d6fc7 has been partially reverted in 4557220c504b9d5f535d4c7b5c5b7295ca8cb560 because of AMOR returning 0.0 as minimum and maximum joint limits (check #152).

PeterBowman, 2018-06-08:
@jgvictores after having taken another look at my `fix-145-solver-state` branch, I think it could do as a starting point for implementing the concepts behind https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/145#issuecomment-370009336. So far, an abstraction layer (following the [proxy pattern](https://en.wikipedia.org/wiki/Proxy_pattern)) has been put on top of current solver code (*AsibotSolver* + *KdlSolver*), but there is some duplication that should be refactored. Once done, there it goes: ICSRH + ICS.

Sorry for the mess, let's discuss the next steps face to face.

PeterBowman, 2018-06-09:
There is much more work to do. I started from the bottom of the chain (cartesian solver), but all components should be representation-aware prior to merging anything. Also, we need to discuss the main shortcoming of the current solution (#113) which went largely unnoticed: transformations of velocities and **accelerations**.

PeterBowman, 2021-06-09:
ASWJ @jgvictores we have decided to remove the hackish `--transform` from the NWC (https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/92dfaed1592a1c6e2ce451f6249397bf5fc67047) and leave the NWS unchanged. In this way, there is no violation of the public interface (we'll always want a specific, standard representation to be used as input/output). Still, we allow users to make the cartesian server open an additional, optional `/rpc_transform:s` port configured as they wish. This is intentionally left to facilitate an easy access to cartesian commads via terminal. Our C++/Python examples have been updated accordingly (https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/cb2b185d8d3b08f40f5f8c203f79ebc4a039c351).

The legacy "fix-145-solver-state" WIP branch will be deleted now: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/6d252ce41c96f8eea602dd238e0e07b7e2c2685d, https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/821234cc2779781f47e54f6d711c3fa1c5def83a...6d252ce41c96f8eea602dd238e0e07b7e2c2685d. Closing the issue as solved, but WONTFIX.

closed 2022-05-22 by PeterBowman