roboticslab-uc3m/kinematics-dynamics #147: ImportError: No module named kinematics_dynamics running exampleCartesianControlClient

rsantos88, 2018-02-26:
I want to run the `exampleCartesianControlClient.py` but it shows this error:
```bash
Traceback (most recent call last):
  File "exampleCartesianControlClient.py", line 4, in <module>
    import kinematics_dynamics
ImportError: No module named kinematics_dynamics
```
I've followed the steps described to install [kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/doc/kinematics-dynamics-install.md#kinematics-dynamics-installation-from-source-code) but maybe there is something I am passing over..

jgvictores, 2018-02-26:
True!

We must document how to install the bindings at https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/doc/kinematics-dynamics-install.md !!

PeterBowman, 2018-02-26:
I presume you compiled bindings. Did you set `LD_LIBRARY_PATH` and `PYTHONPATH` to the directories the `.so` and `.py` files are built in, respectively?

jgvictores, 2018-02-26:
Also, in Python, extra care should be taken with Python 2 vs 3 (e.g. paths 2.7 and 3.5m).

jgvictores, 2018-02-26:
Added bindings section at https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/doc/kinematics-dynamics-install.md#install-bindings (swig + python)

@rsantos88 Would be great if you could beta-test and provide feedback!

rsantos88, 2018-02-27:
Ok. Some notes:
> I presume you compiled bindings. Did you set `LD_LIBRARY_PATH` and `PYTHONPATH` to the directories the `.so` and `.py` files are built in, respectively?

I've inspected `.bashrc` file and this environmental variables are configured but, there are ok? . Here a small fragment:
```bash
export LD_LIBRARY_PATH=/usr/local/lib
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig
export GST_PLUGIN_PATH=/usr/local/lib/gstreamer-1.0
export ROBOTICSLAB_SPEECH_DIR=/home/teo/repos/speech/build
export PYTHONPATH=$PYTHONPATH:`openrave-config --python-dir`
export ROBOTICSLAB_TOOLS_DIR=$HOME/repos/tools/build  # Points to where TEOConfig.cmake is generated $

# For CMake find_package(ROBOTICSLAB_KINEMATICS_DYNAMICS REQUIRED)
export ROBOTICSLAB_KINEMATICS_DYNAMICS_DIR=$HOME/repos/kinematics-dynamics/build

```

I've noticed that when I compile `kinematics-dynamics` and the binding, it can't find `ROBOTICSLAB_YARP_DEVICES`
```bash
teo@teo-oliver:~/repos/kinematics-dynamics/build$ cmake ..
-- YCM found in /usr/local/share/YCM.
-- Setting up installation of roboticslab-kinematics-dynamics.ini to /usr/local/share/yarp/config/path.d folder.
--  --- plugin AmorCartesianControl: disabled (dependencies unsatisfied)
--  --- plugin AsibotSolver: disabled (dependencies unsatisfied)
--  +++ plugin BasicCartesianControl: enabled
--  --- plugin BasicTwoLimbCartesianControl: disabled (dependencies unsatisfied)
--  +++ plugin CartesianControlClient: enabled
--  +++ plugin CartesianControlServer: enabled
--  +++ plugin KdlSolver: enabled
--  --- plugin TwoLimbCartesianControlServer: disabled
-- ROBOTICSLAB_YARP_DEVICES not found, building streamingDeviceController without sensor support
-- Configuring done
-- Generating done
-- Build files have been written to: /home/teo/repos/kinematics-dynamics/build
```
on the other hand, I've compiled and installed binding of kinematics-dinamics. Here is the `cmake` out information:
```bash
teo@teo-oliver:~/repos/kinematics-dynamics/bindings$ cmake .. -DCREATE_PYTHON=ON
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- YCM found in /usr/local/share/YCM.
-- Setting up installation of roboticslab-kinematics-dynamics.ini to /usr/local/share/yarp/config/path.d folder.
CMake Warning at libraries/YarpTinyMathLib/CMakeLists.txt:4 (message):
  YARP math lib not found, disabling YarpTinyMathLib

CMake Warning at libraries/YarpPlugins/AmorCartesianControl/CMakeLists.txt:7 (message):
  AMOR_API package not found, disabling AmorCartesianControl

--  --- plugin AmorCartesianControl: disabled (dependencies unsatisfied)
CMake Warning at libraries/YarpPlugins/AsibotSolver/CMakeLists.txt:2 (message):
  YARP math lib not found, disabling AsibotSolver

--  --- plugin AsibotSolver: disabled (dependencies unsatisfied)
--  +++ plugin BasicCartesianControl: enabled
--  --- plugin BasicTwoLimbCartesianControl: disabled (dependencies unsatisfied)
--  +++ plugin CartesianControlClient: enabled
--  +++ plugin CartesianControlServer: enabled
--  +++ plugin KdlSolver: enabled
--  --- plugin TwoLimbCartesianControlServer: disabled
-- ROBOTICSLAB_YARP_DEVICES not found, building streamingDeviceController without sensor support
CMake Warning at tests/CMakeLists.txt:4 (message):
  GTestSources package not found, disabling tests

-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CREATE_PYTHON

-- Build files have been written to: /home/teo/repos/kinematics-dynamics/bindings
```
when I try to run the example it says the same:
```bash
Traceback (most recent call last):
  File "exampleCartesianControlClient.py", line 4, in <module>
    import kinematics_dynamics
ImportError: No module named kinematics_dynamics
```

PeterBowman, 2018-02-27:
> I've noticed that when I compile kinematics-dynamics and the binding, it can't find ROBOTICSLAB_YARP_DEVICES

Perhaps you need to `git pull` and configure ROBOTICSLAB_YARP_DEVICES again, CMake should be able to locate its build/install dir automatically. Anyway, this is unrelated.

```
CMake Warning:
  Manually-specified variables were not used by the project:

    CREATE_PYTHON
```

You are building the kinematics-dynamics project alone, but its bindings are handled differently. Take a look at the updated install docs, you are missing a `build/` subdirectory within `bindings/`. In the snippet above, you are building the kin-dyn project again (that is, CMake is reading from root `/CMakeLists.txt` again, instead of `/bindings/CMakeLists.txt`).

PeterBowman, 2018-02-27:
Actually, there is no need to tweak env vars after you install bindings (at least for Python). To make sure, run:

```bash
python -c 'import kinematics_dynamics' && echo success || echo failure
```

rsantos88, 2018-02-27:
> You are building the kinematics-dynamics project alone, but its bindings are handled differently. Take a look at the updated install docs, you are missing a build/ subdirectory within bindings/. In [...]

Sorry @PeterBowman for the last snippet. You're right. I hadn't created a build directory :confounded:
This has been the process:
```bash
teo@teo-oliver:~/repos/kinematics-dynamics/bindings$ cd build/
teo@teo-oliver:~/repos/kinematics-dynamics/bindings/build$ cmake .. -DCREATE_PYTHON=ON
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: /usr/local/share/OpenCV/OpenCVConfig.cmake  
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.2") 
-- /usr/local/share/cmake-3.11/Modules/UseSWIG.cmake
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.4m.so (found version "3.4.3") 
CMake Deprecation Warning at /usr/local/share/cmake-3.11/Modules/UseSWIG.cmake:272 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMakeLists.txt:59 (swig_add_module)

-- Found PythonInterp: /usr/bin/python3.4 (found version "3.4.3") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/teo/repos/kinematics-dynamics/bindings/build
teo@teo-oliver:~/repos/kinematics-dynamics/bindings/build$ make -j$(nproc)  # compile
Scanning dependencies of target gen_kinematics_dynamics_kinematics_dynamicsPYTHON_wrap
[ 33%] Swig source
[ 33%] Built target gen_kinematics_dynamics_kinematics_dynamicsPYTHON_wrap
Scanning dependencies of target _kinematics_dynamics
[ 66%] Building CXX object CMakeFiles/_kinematics_dynamics.dir/kinematics_dynamicsPYTHON_wrap.cxx.o
[100%] Linking CXX shared module _kinematics_dynamics.so
[100%] Built target _kinematics_dynamics
teo@teo-oliver:~/repos/kinematics-dynamics/bindings/build$ sudo make install; sudo ldconfig; cd # install and go home
[ 33%] Built target gen_kinematics_dynamics_kinematics_dynamicsPYTHON_wrap
[100%] Built target _kinematics_dynamics
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/python3/dist-packages/kinematics_dynamics.py
-- Installing: /usr/local/lib/python3/dist-packages/_kinematics_dynamics.so
teo@teo-oliver:~$ cd repos/kinematics-dynamics/examples/python/
teo@teo-oliver:~/repos/kinematics-dynamics/examples/python$ ls
exampleCartesianControlClient.py
teo@teo-oliver:~/repos/kinematics-dynamics/examples/python$ python exampleCartesianControlClient.py 
Traceback (most recent call last):
  File "exampleCartesianControlClient.py", line 4, in <module>
    import kinematics_dynamics
ImportError: No module named kinematics_dynamics

```

jgvictores, 2018-02-27:
Could you please type the output of
`python --version`
?

PeterBowman, 2018-02-27:
> -- Installing: /usr/local/lib/python3/dist-packages/kinematics_dynamics.py

Try this:

```shell
python3 exampleCartesianControlClient.py
```

rsantos88, 2018-02-27:
> Could you please type the output of
python --version
?

Python 2.7.6

> Try this:
>
>```shell
>python3 exampleCartesianControlClient.py
>```
No more this error ^^. But I think this demo has been programed in order to use in python 2.
```shell
  File "exampleCartesianControlClient.py", line 9
    print '[error] Please try running yarp server'
                                                 ^
SyntaxError: Missing parentheses in call to 'print'
```
Anyway, I've fixed the parenthesis of the `print` function, but after that the example crashs showing this:
```shell
Traceback (most recent call last):
  File "exampleCartesianControlClient.py", line 3, in <module>
    import yarp
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 28, in <module>
    _yarp = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 24, in swig_import_helper
    _mod = imp.load_module('_yarp', fp, pathname, description)
  File "/usr/lib/python3.4/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)
ImportError: dynamic module does not define init function (PyInit__yarp)
```

jgvictores, 2018-02-27:
Yes, with the current CMakes, YARP tends to install in the default (2.7), while kinematics-dynamics usually goes to the newest Python.

I'd recommend as indirectly commented [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/doc/kinematics-dynamics-install.md#install-bindings): in kinematics-dynamics bindings cmake, toggle to advanced via `t` and replace all `3.4` or `3.4m` or similar manually to `2.7`. Usually 3 of them.
Then compile, install, and go for python=python2.

rsantos88, 2018-02-27:
Ok. This are the fields modified using `ccmake` in toogle mode (only compiling binding of kinematic-dynamic):
```shell
 PYTHON_EXECUTABLE                /usr/bin/python2.7                                                 
 PYTHON_INCLUDE_DIR               /usr/include/python2.7                                             
 PYTHON_LIBRARY                   /usr/lib/x86_64-linux-gnu/libpython2.7.so
```
I've checked the existence of these file before writing the path, because there are `python3.4m` but not `python2.7m` . For example, in `/usr/include` I can see this:
```shell
teo@teo-oliver:/usr/include$ python
python                                         python3.4m/
python2                                        python3.4m-config
python2.7/                                     python3-config
python2.7-config                               python3m
python2-config                                 python3m-config
python3                                        python-argcomplete-check-easy-install-script3
python3.4/                                     python-config
python3.4-config 
```
So, once I've configured and compiled it, I tried once again to run the example but it crashs with this error:
```shell
teo@teo-oliver:~/repos/kinematics-dynamics/examples/python$ python exampleCartesianControlClient.py 
error while opening /usr/local/lib/yarp/yarp_priority.so:
  /usr/local/lib/yarp/yarp_priority.so: undefined symbol: _ZTIN4yarp2os4impl27PortReaderBufferBaseCreatorE
yarp: Port /cartesianControlExample/rpc:c active at tcp://2.2.2.101:10042/
error while opening /usr/local/lib/yarp/yarp_priority.so:
  /usr/local/lib/yarp/yarp_priority.so: undefined symbol: _ZTIN4yarp2os4impl27PortReaderBufferBaseCreatorE
yarp: Port /cartesianControlExample/command:o active at tcp://2.2.2.101:10043/
error while opening /usr/local/lib/yarp/yarp_priority.so:
  /usr/local/lib/yarp/yarp_priority.so: undefined symbol: _ZTIN4yarp2os4impl27PortReaderBufferBaseCreatorE
yarp: Port /cartesianControlExample/state:i active at tcp://2.2.2.101:10044/
[error] DeviceDriverImpl.cpp:36 open(): Error on connect to remote RPC server.
yarpdev: ***ERROR*** driver <CartesianControlClient> was found but could not open
Cannot open the device!
Traceback (most recent call last):
  File "exampleCartesianControlClient.py", line 20, in <module>
    sys.exit()
NameError: name 'sys' is not defined
```
I have running at the same time `BasicCartesianControl` device

jgvictores, 2018-02-28:
>   sys.exit() NameError: name 'sys' is not defined

Was minor, should be fixed at 127a768baa478bb7194762435c64220f2022bd19.

>   /usr/local/lib/yarp/yarp_priority.so: undefined symbol: _ZTIN4yarp2os4impl27PortReaderBufferBaseCreatorE

Looks like an issue of multiple installations of YARP. At minimum would require reinstalling YARP and k-d plus bindings, best if first removing all traces of installed versions.

However, a quick hack could maybe simply be: `sudo rm /usr/local/share/yarp/plugins/priority.ini`

rsantos88, 2018-03-01:
Thanks @jgvictores . These errors was fixed. Edited also the port name of RPC server (6e77c2aa27c959f94f7c6671179c9870f96f867f) to correspond with xml of [teoBase](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/develop/share/teoBase/scripts/teoBase.xml#L56)

jgvictores, 2018-03-01:
Great, perfect!

Feel free to merge 6e77c2aa27c959f94f7c6671179c9870f96f867f [thanks!] from `issue#147` into `develop` and close the issue!

rsantos88, 2018-03-01:
Maybe we've to do this example more functional.
```shell
teo@teo-oliver:~/repos/kinematics-dynamics/examples/python$ python exampleCartesianControlClient.py 
yarp: Port /cartesianControlExample/rpc:c active at tcp://2.2.2.101:10042/
yarp: Port /cartesianControlExample/command:o active at tcp://2.2.2.101:10043/
yarp: Port /cartesianControlExample/state:i active at tcp://2.2.2.101:10044/
yarp: Sending output from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc:s using tcp
yarp: Sending output from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i using udp
yarp: Receiving input from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i using udp
[success] DeviceDriverImpl.cpp:60 open(): Connected to remote.
[INFO]created device <CartesianControlClient>. See C++ class roboticslab::CartesianControlClient for documentation.
> stat
[warning] ICartesianControlImpl.cpp:150 stat(): FK stream timeout, sending RPC request.
< ccnc [3.25149848408e-17, -0.34692, -0.221806, 2.40458836715e-16, 1.57079632679, -4.8091767343e-17]
> inv [3.25149848408e-17, -0.34692, -0.221806, 0, 1, 0, 90]
< [fail]
> movj [3.25149848408e-17, -0.34692, -0.221806, 0, 1, 0, 90]
< [fail]
delay(1)
bye!
yarp: Removing output from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc:s
yarp: Removing output from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i
yarp: Removing input from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i
```

PeterBowman, 2018-03-01:
I forgot to use the `--transform` trick here, sorry. Add this before [line 16](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/e1d2a66de87962fc74b17a18fdd6cc123b09956b/examples/python/exampleCartesianControlClient.py) (or obtain the equivalent scaled axis-angle representation):

```python
options.put('transform', 1)
```

rsantos88, 2018-03-05:
with that change (160b2a9cbf8b064e420411667a0c63040d565128), the result is:

**exampleCartesianControlClient.py:**
```shell
yarp: Port /cartesianControlExample/rpc:c active at tcp://2.2.2.101:10037/
yarp: Port /cartesianControlExample/command:o active at tcp://2.2.2.101:10038/
yarp: Port /cartesianControlExample/state:i active at tcp://2.2.2.101:10039/
yarp: Sending output from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc_transform:s using tcp
yarp: Sending output from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i using udp
yarp: Receiving input from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i using udp
[success] DeviceDriverImpl.cpp:60 open(): Connected to remote.
[INFO]created device <CartesianControlClient>. See C++ class roboticslab::CartesianControlClient for documentation.
> stat
[warning] ICartesianControlImpl.cpp:150 stat(): FK stream timeout, sending RPC request.
< ccnc [0.675174970134, -0.34692, 0.25095698028, 5.82132264516e-16, 1.0, 2.55418106437e-16, 19.9999571131]
> inv [0, -0.346927436108, -0.221801094416, 0, 1, 0, 90]
< [2880.19297392, -2160.00129458, 0.146176036394, -360.507285228, -2160.14618177, 5040.31430965]
> movj [0, -0.346927436108, -0.221801094416, 0, 1, 0, 90]
< [fail]
delay(1)
bye!
yarp: Removing output from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc_transform:s
yarp: Removing output from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i
yarp: Removing input from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i
```
**Cartesian Control Server:**
```shell
yarp: Receiving input from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc_transform:s using tcp
yarp: Receiving input from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i using udp
yarp: Sending output from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i using udp
[debug] RpcResponder.cpp:46 respond(): Got: [stat]
[debug] RpcResponder.cpp:46 respond(): Got: [inv] 0.0 -0.346927436108 -0.221801094416 0.0 1.0 0.0 90.0
[debug] RpcResponder.cpp:46 respond(): Got: [movj] 0.0 -0.346927436108 -0.221801094416 0.0 1.0 0.0 90.0
[info] ICartesianControlImpl.cpp:73 movj(): dist[0]: 386547053469.807068
[info] ICartesianControlImpl.cpp:77 movj():  -->candidate: 38654705346.980705
[info] ICartesianControlImpl.cpp:73 movj(): dist[1]: 2160.001295
[info] ICartesianControlImpl.cpp:73 movj(): dist[2]: 0.146176
[info] ICartesianControlImpl.cpp:73 movj(): dist[3]: 360.507285
[info] ICartesianControlImpl.cpp:73 movj(): dist[4]: 2160.146182
[info] ICartesianControlImpl.cpp:73 movj(): dist[5]: 5040.314310
[info] ICartesianControlImpl.cpp:80 movj(): max_time[final]: 38654705346.980705
[info] ICartesianControlImpl.cpp:94 movj(): vmo[0]: 10.000000
[info] ICartesianControlImpl.cpp:94 movj(): vmo[1]: 0.000000
[info] ICartesianControlImpl.cpp:94 movj(): vmo[2]: 0.000000
[info] ICartesianControlImpl.cpp:94 movj(): vmo[3]: 0.000000
[info] ICartesianControlImpl.cpp:94 movj(): vmo[4]: 0.000000
[info] ICartesianControlImpl.cpp:94 movj(): vmo[5]: 0.000000
[error] ICartesianControlImpl.cpp:100 movj(): getRefSpeeds (for storing) failed.
[INFO]device active in background...
yarp: Removing input from /cartesianControlExample/rpc:c to /teoSim/rightArm/CartesianControl/rpc_transform:s
yarp: Removing input from /cartesianControlExample/command:o to /teoSim/rightArm/CartesianControl/command:i
yarp: Removing output from /teoSim/rightArm/CartesianControl/state:o to /cartesianControlExample/state:i
```
and no movement of the robot

PeterBowman, 2018-03-05:
Have you pulled, compiled and installed latest [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins)? What's the terminal output in the simulator?

jgvictores, 2018-03-05:
Mmm the lines above to not look good at all...
```
< ccnc [0.675174970134, -0.34692, 0.25095698028, 5.82132264516e-16, 1.0, 2.55418106437e-16, 19.9999571131]
> inv [0, -0.346927436108, -0.221801094416, 0, 1, 0, 90]
< [2880.19297392, -2160.00129458, 0.146176036394, -360.507285228, -2160.14618177, 5040.31430965]
> movj [0, -0.346927436108, -0.221801094416, 0, 1, 0, 90]
< [fail]
```
Where `2880.19297392` looks way out of limits, and `[fail]` says that performing no movement makes sense.

PeterBowman, 2018-03-05:
And...

> [warning] ICartesianControlImpl.cpp:150 stat(): FK stream timeout, sending RPC request.

Note to self: review this, shouldn't time out on the first try.

rsantos88, 2018-03-05:
Ok. I've pulled openrave-yarp-plugins and I've changed the values in the example (837970fbe10a39915510b5b9c49f2f0fcf55dbf3)
The elbow is positioned 90 degrees and the TCP is turned 20 degrees around the X axis.
Now is working perfectly! You can do PR (https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/149) if you are agree.

PeterBowman, 2018-03-05:
> Where 2880.19297392 looks way out of limits, and [fail] says that performing no movement makes sense.

@jgvictores, @rsantos88 same happened to me on my local, outdated installation of teo-openrave-models. That is, the controlboard opened by teoSim reported wierd joint limits, which later propagated to the controller/solver pair (#145 FTW!). Please `git pull` TEO's OR environment files, install them and try again, should be fine.

For the record, roboticslab-uc3m/teo-openrave-models@65b7ee6 works, roboticslab-uc3m/teo-openrave-models@767ba9a doesn't.

PeterBowman, 2018-03-06:
> Note to self: review this, shouldn't time out on the first try.

Solved at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/04fd34f5ad1c4de3a87363be952309f460fc3a74.

jgvictores, 2018-03-06:
@rsantos88 Ok to close this issue!

closed 2018-03-06 by rsantos88